nRF24/RF24 #935: Is there a known ESP32 Deep Sleep issue with radio.powerDown()?  [Question]

Tech500, 2024-01-15:
Measured just the nRF24L01 radio.powerUp() current of 26.9 mA; after radio.powerDown(), very briefly 3.1 mA --then back up to 26.9 mA, after going into ESP32 Deep Sleep.  

Using a pair of nRF24L01 supplied 5 Volts to piggyback voltage adapter.

ESP32 DEVKIT V1 is the development board.

Functions configuring RTC_GPIO  pins for external 0 wakeup and going to ESP32 Deep Sleep:

``void setup_rtc_gpio() {
  rtc_gpio_init(GPIO_NUM_25);
  rtc_gpio_set_direction(GPIO_NUM_25, RTC_GPIO_MODE_OUTPUT_ONLY);
  rtc_gpio_set_level(GPIO_NUM_25, 0);
}

//setup_rtc_gpio();  is the last line in sketch setup.

void goToDeepSleep() {

  // Enable RTC peripherals for deep sleep
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  rtc_gpio_pullup_dis(GPIO_NUM_25);
  rtc_gpio_pulldown_en(GPIO_NUM_25);

  radio.powerDown();

  delay(5000);  //Current metering delay after calling radio.powerDown() = 3.1 mA.  nRF24L01+ current after going to Deep Sleep
                       // 26.9 mA.   Current readings are from only the nRF24L01+.

  esp_deep_sleep_start();
  Serial.println("Testing --Not Deep Sleeping");  //This line will not print (Does not print).

Project code:  [https://github.com/Tech500/OneWay_nRF24L01_Battery_Switch](url)

Tech500, 2024-01-15:
Pair of nRF24L01+ are connected to a separate 5 volt power supply.  ESP32 DEVKIT V1 is powered by USB connection.

2bndy5, 2024-01-16:
Are the 5V power supplies being controlled by the ESP32? We're just trying to make sure that the power to the nRF24L is _not_ interrupted.

This doesn't sound a code issue. The nRF24L is hard wired to go into Standby-I power mode (not to be confused with power-down mode) after power-on-reset (ie when power supply is interrupted/unstable). The RF24 lib code does not care/know about the ESP32's deep sleep implementation.

Tech500, 2024-01-16:
Only connection between 5 Volt supply and ESP32 is a common ground.

2bndy5, 2024-01-16:
What about the capacitors used on the nRF24L radios?

Tech500, 2024-01-16:
One 100 uF, 16 V.  My code calls Start deep sleep from a function and not in setup.  Could this influence what I have been observering with the current, deep sleep behavior?  I linked to my project code.

2bndy5, 2024-01-16:
>  My code calls Start deep sleep from a function and not in setup. Could this influence what I have been observering with the current, deep sleep behavior? I linked to my project code.

Nope. This isn't a code issue. This is a power issue.

Tech500, 2024-01-16:
Will add capacitors to each nRF24L01.  100 uF, 16 V tantalum good choice?

2bndy5, 2024-01-16:
yes

2bndy5, 2024-01-16:
yes

Tech500, 2024-01-16:
Will have to order, none on hand in my parts box.

Tech500, 2024-01-18:
Have a working Webserver with a nRF24L01 controlled remote battery switch!

Prepared a under two minute video of processing web request for “Camera View.”  Asyncwebserver request turns on switch by triggering a 60 second countdown timer and send data = 1 to the first nRF24L01.  Second nRF24L01 recevies data = 1 and turns on switch, Wakes ESP32 from Deep Sleep (using External 0, RTC_GPIO wakeup pin.), and powers down the 2nd NRF24L01.

[nRF24L01 Webserver with Remote Switch](https://drive.google.com/file/d/1TIXK7ZPLIWeh1BAcg_uAaSn8aYGxNmo6/view?usp=sharing)

2bndy5, 2024-01-18:
Does that also mean that we can close this and #937?

Tech500, 2024-01-18:
Even with the 100 uf 16 Volt Tantalum Capacitors to both nRF24L01+ transceivers; when radio.powerDown function is called before going to Deep Sleep, deep sleep current is the same as before power down the radio.  Are external resistors need on any of the SPI pins?

This my first experience using nRF24L01; at seventy-five years of age, I am still learning.

2bndy5, 2024-01-19:
> when radio.powerDown function is called before going to Deep Sleep, deep sleep current is the same as before power down the radio.

Are able to verify that voltage to the radio's VCC post n never drops below 3v?

> Are external resistors need on any of the SPI pins?

Not usually. That's more common for I2C.

TMRh20, 2024-01-19:
> This my first experience using nRF24L01; at seventy-five years of age, I am still learning.

Thats awesome, something we should all aspire to. I'm kind of stumped on this one as well, unless there is either a problem with sleep mode or the power is being interrupted I'm not sure what would cause this issue.

TMRh20, 2024-01-19:
I'm not sure if you are powering the radio down, but IRQs won't trigger if the radio is in `powerDown()` mode.

Either way, if you've got the radio powering down properly now, its a matter of understanding the workings of the RF24 modules and the related code. We typically just deal with software issues here, so you may have better luck with that in the Arduino forums etc. where people help with coding and how-to issues. You might want to offer up more details about exactly what is going on and the exact code that you are using too, its more likely to get a response there.

Is it OK to close this issue, as I think we've resolved the main problem?

closed 2024-01-23 by TMRh20

Tech500, 2024-02-03:
Have project using RF24 library to provide an interrupt for controlling a remote switch.  Project uses RTC-IO to acess RTC-GPIO pins while in Deep Sleep.  Did not find much searching the net for reference or tutorials on RTC_IO.  Google's Bard came best resource for RTC-GPIO with some prompting.

[ESP32_nRF24L01_Remote_Relay](https://github.com/Tech500/ESP32_nRF24L01_Remote_Relay/blob/main/README.md)