nRF24/RF24 #936: Can RF24 library be used with WiFi?

Tech500, 2024-01-17:
Please read [about common issues](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md) first. It addresses the most common problems that people have (whether they know it or not).

2bndy5, 2024-01-17:
Nope.

RF24Ethernet and RF24Gateway libs can be used to build a bridge between RF24Network and a (W)LAN.

2bndy5, 2024-01-17:
I suppose this answer could go into the common_issue doc, but asking this question really indicates a misunderstanding of what WiFi ([802.11 compliant](https://en.wikipedia.org/wiki/IEEE_802.11)) is vs proprietary protocols. The nRF24L isn't even well suited for [802.15.4 compliance](https://en.wikipedia.org/wiki/IEEE_802.15#IEEE_802.15.4:_Low_Rate_WPAN).

closed 2024-01-17 by 2bndy5