nRF24/RF24 #938: [Question] setChannel & setDataRate not read correctly by printPrettyDetails ?

matou78, 2024-01-25:
Hello everyone,
i made myself a little tester so i can check all my nrf24 if they are working or not.
wasn't able to find anything similar here or on google.
it is working by detecting the connexion and try to read registers from the nrf24 modules so i can see if something is wrong or all is good

below is the full code, if someone needs, i can give you the entire file for VScode+platformio
```
#include <Arduino.h>
#include "RF24.h"
#include "printf.h"

const byte adresse[5] = {0xAA,0x0A,0x0A,0x0A,0xAA};  

#define CE_PIN PIN_PD5
#define CSN_PIN PIN_PD6

RF24 radio(CE_PIN, CSN_PIN);

void setup(void) {

  Serial.begin(9600);
  printf_begin();
  delay(1000);

  Serial.println("------------------------------------------Start------------------------------------------------------");
  Serial.println("Test connexion NRF24L01 version SMD");

  radio.begin();
  radio.setAddressWidth(5);
  radio.setChannel(5);
  radio.enableDynamicPayloads();
  radio.openWritingPipe(adresse);
  radio.enableAckPayload();
  radio.setDataRate(RF24_250KBPS);
  radio.setPALevel(RF24_PA_LOW);
  radio.openReadingPipe(1,adresse);

  delay(1000);

  if(radio.begin()== 0)
  {
    Serial.println("erreur connexion !");
  }else{
    Serial.println("connection ok");
  }

  
  Serial.println("----------------------------------------RX mode------------------------------------------------------");
  radio.startListening();
  radio.printPrettyDetails();
  Serial.println("----------------------------------------TX mode-----------------------------------------------------");
  radio.stopListening();
  radio.printPrettyDetails();
  Serial.println("------------------------------------------END--------------------------------------------------------");
}

void loop(void) {
  
}
```

it reads and shows me everything as intended but 2 informations are wrong.
the RF Data Rate = 1 MBPS and Channel = 76 (~ 2476 MHz) are read wrongly as i set the datatrate to 250kbs and channel to 5

am i doing something wrong or reading it wrong ?
Thank you

2bndy5, 2024-01-25:
Have you read the common_issues doc? There's a [section about if `printDetails()` outputs incorrectly](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md#payloads-receivedsent-are-inaccurate-or-printdetails-outputs-the-unexpected-values).

2bndy5, 2024-01-25:
> ```cpp
> if(radio.begin()== 0)
> ```

This sets up the radio with default values. All configurations should be done _after_ calling `radio.begin()`.

matou78, 2024-01-25:
Hello @2bndy5 , yes i tried it, but didn't push it further on this side as on the output it is not changing too
this is my output
```
------------------------------------------Start------------------------------------------------------
Test connexion NRF24L01 version SMD
connection ok
----------------------------------------RX mode------------------------------------------------------
SPI Frequency		= 10 Mhz
Channel			= 76 (~ 2476 MHz)
Model			= nRF24L01+
RF Data Rate		= 1 MBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 1500 microseconds
Auto Retry Attempts	= 15 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 0
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Disabled
Auto Acknowledgment	= Enabled
Primary Mode		= RX
TX address		= 0xaa0a0a0aaa
pipe 0 (closed) bound	= 0xaa0a0a0aaa
pipe 1 ( open ) bound	= 0xaa0a0a0aaa
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
----------------------------------------TX mode-----------------------------------------------------
SPI Frequency		= 10 Mhz
Channel			= 76 (~ 2476 MHz)
Model			= nRF24L01+
RF Data Rate		= 1 MBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 1500 microseconds
Auto Retry Attempts	= 15 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 0
Multicast		= Disabled
Custom ACK Payload	= Disabled
Dynamic Payloads	= Disabled
Auto Acknowledgment	= Enabled
Primary Mode		= TX
TX address		= 0xaa0a0a0aaa
pipe 0 ( open ) bound	= 0xaa0a0a0aaa
pipe 1 ( open ) bound	= 0xaa0a0a0aaa
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
------------------------------------------END--------------------------------------------------------

```

matou78, 2024-01-25:
> > ```c++
> > if(radio.begin()== 0)
> > ```
> 
> @matou78 This sets up the radio with default values. All configurations should be done _after_ calling `radio.begin()`.

Mate you are a genius , I am just retarded xD it's working now lol
Thank you a lot

in case anyone come here for the same issue, here is the correct code 

```
#include <Arduino.h>
#include "RF24.h"
#include "printf.h"

const byte adresse[5] = {0xAA,0x0A,0x0A,0x0A,0xAA};  

#define CE_PIN PIN_PD5
#define CSN_PIN PIN_PD6

RF24 radio(CE_PIN, CSN_PIN, 40000000);

void setup(void) {

  Serial.begin(9600);
  printf_begin();
  delay(1000);

  Serial.println("------------------------------------------Start------------------------------------------------------");
  Serial.println("Test connexion NRF24L01 version SMD");

  if(radio.begin()== 0)
  {
    Serial.println("erreur connexion !");
  }else{
    Serial.println("connection ok");
  }

  radio.setAddressWidth(5);
  radio.setChannel(0x05);
  radio.enableDynamicPayloads();
  radio.openWritingPipe(adresse);
  radio.enableAckPayload();
  radio.setDataRate(RF24_250KBPS);
  radio.setPALevel(RF24_PA_LOW);
  radio.openReadingPipe(1,adresse);

  delay(1000);

  
  Serial.println("----------------------------------------RX mode------------------------------------------------------");
  radio.startListening();
  radio.printPrettyDetails();
  Serial.println("----------------------------------------TX mode-----------------------------------------------------");
  radio.stopListening();
  radio.printPrettyDetails();
  Serial.println("------------------------------------------END--------------------------------------------------------");
}

void loop(void) {
  
}
```