nRF24/RF24 #939: ESP32 Core Panic on radio 0; No panic on Radio number 1, same wiring.  How to troubleshoot?[Question]

Tech500, 2024-01-28:
Radio 0   after selecting radio number, then pressing T; getting a core panic.

Board is a Doit ESP32 Devkit V1; using Arduino IDE 

`SPI pin defines:

MOSI       23
MISO       19
SCK          18
CE_PIN     17
CSN_PIN    5
IRQ           15    //Unable to upload code with connection to GPIO2` 

[Gist of Radio 0 Core panic](https://gist.github.com/Tech500/dd3e847e06b7bceb3664b9554fe1b5d2)

Tech500, 2024-01-28:
Swaped radio; issue is resolved. 

Should there be an immediate response from the Transmit ping test?

2bndy5, 2024-01-28:
> Should there be an immediate response from the Transmit ping test?

Yes. In testing the configure IRQ example, I always enabled the RX side before engaging the TX side.

But I'm not sure why it is panicking. There must be a buffer operation missing if response is not received.

Tech500, 2024-01-29:
I have taken the example and modified it  for use in relay project.  Even now I still have Core Panic after seeing "IRQ pin is actively LOW".

`nRF24L01+ IRQ_Receiver
Receving...

Boot number: 1
	IRQ pin is actively LOW
	data_sent: 1, data_fail: 0, data_ready: 0
   'Data Ready' eveGuru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4008ddac  PS      : 0x00060935  A0      : 0x8008cd1e  A1      : 0x3ffbf13c  
A2      : 0x3ffb897c  A3      : 0x3ffb81a4  A4      : 0x00000004  A5      : 0x00060923  
A6      : 0x00060923  A7      : 0x00000001  A8      : 0x3ffb81a4  A9      : 0x00000018  
A10     : 0x3ffb81a4  A11     : 0x00000018  A12     : 0x3ffc265c  A13     : 0x00060923  
A14     : 0x007bf398  A15     : 0x003fffff  SAR     : 0x0000001b  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40088ea9  LEND    : 0x40088eb9  LCOUNT  : 0xfffffff9  
Core  1 was running in ISR context:
EPC1    : 0x400dc63b  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x4008dda9:0x3ffbf13c |<-CORRUPTED

Core  0 register dump:
`

Have no idea of what to do with the Core register dumps.   Closer to goal than previously...

TMRh20, 2024-01-29:
```
  bool tx_ds, tx_df, rx_dr;                 // declare variables for IRQ masks
  radio.whatHappened(tx_ds, tx_df, rx_dr);
  wait_for_event = false;

```
I seem to be getting a similar reaction from my ESP32 using the interrupt example. I modified the interrupt to contain just the above, and it still does it on reception of data. If I remove everything except for the `wait_for_event=false` then it doesn't hang anymore. I suspect there is something going on with SPI and/or SPI transactions as they are supposed to be interrupt safe.
This will take some time to debug, so I would suggest not using interrupts with ESP32 for the time being, unless you can move all the functions that call SPI commands to outside of the interrupt.

TMRh20, 2024-01-29:
This would be a workaround until we can figure out what is going on here:
Just need to configure the CE,CS and IRQ pins.
```

#include <SPI.h>
#include "printf.h"
#include "RF24.h"

// We will be using the nRF24L01's IRQ pin for this example
#define IRQ_PIN 27                      // this needs to be a digital input capable pin
volatile bool wait_for_event = false;  // used to wait for an IRQ event to trigger
volatile bool gotInterrupt = false;

#define CE_PIN 12
#define CSN_PIN 14
// instantiate an object for the nRF24L01 transceiver
RF24 radio(CE_PIN, CSN_PIN);

// Let these addresses be used for the pair
uint8_t address[][6] = { "1Node", "2Node" };
// It is very helpful to think of an address as a path instead of as
// an identifying device destination

// to use different addresses on a pair of radios, we need a variable to
// uniquely identify which address this radio will use to transmit
bool radioNumber = 1;  // 0 uses address[0] to transmit, 1 uses address[1] to transmit

// Used to control whether this node is sending or receiving
bool role = false;  // true = TX node, false = RX node

// For this example, we'll be using a payload containing
// a string that changes on every transmission. (successful or not)
// Make a couple arrays of payloads & an iterator to traverse them
const uint8_t tx_pl_size = 5;
const uint8_t ack_pl_size = 4;
uint8_t pl_iterator = 0;
// The " + 1" compensates for the c-string's NULL terminating 0
char tx_payloads[][tx_pl_size + 1] = { "Ping ", "Pong ", "Radio", "1FAIL" };
char ack_payloads[][ack_pl_size + 1] = { "Yak ", "Back", " ACK" };

void interruptHandler();  // prototype to handle IRQ events
void printRxFifo();       // prototype to print RX FIFO with 1 buffer

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    // some boards need to wait to ensure access to serial over USB
  }

  // initialize the transceiver on the SPI bus
  if (!radio.begin()) {
    Serial.println(F("radio hardware is not responding!!"));
    while (1) {}  // hold in infinite loop
  }

  // print example's introductory prompt
  Serial.println(F("RF24/examples/InterruptConfigure"));

  // To set the radioNumber via the Serial monitor on startup
  Serial.println(F("Which radio is this? Enter '0' or '1'. Defaults to '0'"));
  while (!Serial.available()) {
    // wait for user input
  }
  char input = Serial.parseInt();
  radioNumber = input == 1;
  Serial.print(F("radioNumber = "));
  Serial.println((int)radioNumber);

  // role variable is hardcoded to RX behavior, inform the user of this
  Serial.println(F("*** PRESS 'T' to begin transmitting to the other node"));
[... 243 lines cut ...]
 * Print the entire RX FIFO with one buffer. This will also flush the RX FIFO.
 * Remember that the payload sizes are declared as tx_pl_size and ack_pl_size.
 */
void printRxFifo() {
  if (radio.available()) {  // if there is data in the RX FIFO
    // to flush the data from the RX FIFO, we'll fetch it all using 1 buffer

    uint8_t pl_size = !role ? tx_pl_size : ack_pl_size;
    char rx_fifo[pl_size * 3 + 1];  // RX FIFO is full & we know ACK payloads' size
    if (radio.rxFifoFull()) {
      rx_fifo[pl_size * 3] = 0;           // add a NULL terminating char to use as a c-string
      radio.read(&rx_fifo, pl_size * 3);  // this clears the RX FIFO (for this example)
    } else {
      uint8_t i = 0;
      while (radio.available()) {
        radio.read(&rx_fifo + (i * pl_size), pl_size);
        i++;
      }
      rx_fifo[i * pl_size] = 0;  // add a NULL terminating char to use as a c-string
    }
    Serial.print(F("Complete RX FIFO: "));
    Serial.println(rx_fifo);  // print the entire RX FIFO with 1 buffer
  }
}
```

2bndy5, 2024-01-29:
We should also flush the RX FIFO when switching back to RX mode. That example has been in need of some attention.

2bndy5, 2024-01-29:
> Would it be possiable to have a basic IRQ config for using interrupts

I would refer to the docs for finer detail:
- [Docs for `radio.maskIrq()`](https://nrf24.github.io/RF24/classRF24.html#abf68b9b0c9cd17179e9e144c3e7f9c45)
- [Docs for `radio.whatHappened()`](https://nrf24.github.io/RF24/classRF24.html#afb97dc4bdf4d2d84ea44060ac5b4ed89)

2bndy5, 2024-01-29:
> `Interrupt wdt timeout on CPU1`

I believe this error is trying to say something. Probably all the stuff that was going on in that ISR was taking too long for a typical ISR. There may also be some kind of mutex lock on the Serial bus.

Tech500, 2024-01-29:
Should there be Serial Prints in an isr; think I read they should not be used in isr and the time spent in isr short.

2bndy5, 2024-01-29:
> If I remove everything except for the wait_for_event=false then it doesn't hang anymore.

Reminds me of rust programming language's closures. Any members of an ISR may have to be limited to volatile memory on the ESP32. Its easier to just separate everything out.

Tech500, 2024-01-29:
Will make changes to project code.; thank you appreciate the help!

TMRh20, 2024-01-29:
From what I can find the ESP32 can’t handle SPI or Serial prints in the ISR. You can only set a variable and handle it outside the ISR. This seems like it might affect other boards also, so not really an RF24 issue, but with the OS involved.

Tech500, 2024-01-29:
Getting ready to test; have moved everything below whatHappen to a conditional flag in loop.

2bndy5, 2024-01-29:
Created a branch named update-irq-example with the changes proposed to both Linux and Arduino examples (not py example)

Tech500, 2024-01-29:
This is how I setup the isr; moving the function to loop with a conditional statement of interruptResults:

`portMUX_TYPE mux = portMUX_INITIALIZER_UNLOCKED;

volatile int interruptResults;

void isr() {

  portENTER_CRITICAL_ISR(&mux);

  interruptResults = 1;

  portEXIT_CRITICAL_ISR(&mux);
}
`

`Void loop(){
   if(interruptResults){
   .....your interruptHandler here
   no more core panic...
   received a couple of messages
   with SwitchState value of 1
   more responses all following
   were switchState = 0 and not 
   a value of expected 2.
   }`
   

SwitchState comes on, in my code; switchState off does not change to off.

Something I missed.  Do I need to clear anything before nextradio,write()?

Tech500, 2024-01-30:
Have "Switch Project" turning switch on and off using RF24 IRQ interrupt:

22:16:27.790 -> Boot number: 2
22:16:27.822 -> 	
22:16:27.822 -> IRQ pin is actively LOW
22:16:27.822 -> 	data_sent: 1, data_fail: 0, data_ready: 0
22:16:27.887 ->    'Data Ready' event test failed
22:16:27.918 -> switchState:  1
22:16:27.950 -> 
22:16:27.950 -> Battery power switched ON
22:16:27.950 -> ESP32 wake from Deep Sleep
22:16:27.993 -> 
22:16:47.670 -> 	
22:16:47.670 -> IRQ pin is actively LOW
22:16:47.702 -> 	data_sent: 1, data_fail: 0, data_ready: 0
22:16:47.735 ->    'Data Ready' event test failed
22:16:47.767 -> switchState:  2
22:16:47.799 -> 
22:16:47.799 -> Battery power switched OFF
22:16:47.831 -> ESP32 going to Deep Sleep
22:17:00.243 -> �&zy��V��$g��
22:17:00.307 -> 
22:17:00.307 -> 
22:17:00.307 -> nRF24L01+ IRQ_Receiver
22:17:00.338 -> Receving...
22:17:00.371 -> 
22:17:00.371 -> Boot number: 3
22:17:00.371 -> 	
22:17:00.371 -> IRQ pin is actively LOW
22:17:00.402 -> 	data_sent: 1, data_fail: 0, data_ready: 0
22:17:00.435 ->    'Data Ready' event test failed
22:17:00.499 -> switchState:  1
22:17:00.499 -> 
22:17:00.499 -> Battery power switched ON
22:17:00.531 -> ESP32 wake from Deep Sleep

Is it possiable to send just an IRQ signal; when nRF24L01 is powerDown.  Think I read the IRQ is available in powerdown mode.
Maybe have two instances of RF24; one for switchState and one for IRQ signal.

Would like to use powerDown() and powerUP() in project; investigating more.   Appreciate how responsive you have been with the library.

TMRh20, 2024-01-30:
> Is it possible to send just an IRQ signal; when nRF24L01 is powerDown.

Nope, the Microcontroller can be powered down, but not the radio if you want to receive data/trigger IRQ.

Tech500, 2024-01-30:
Any way to lower current comsumption?  Standby I or II?.

Measured 25.5 mA  with RF24_PA_LOW.  Will be running from battery...

2bndy5, 2024-01-30:
For hardware reasons, listening has to be done in Standby-II. The only configuration that might affect power consumption is the PA level (lower is better for consumption but worse for distance).

Remember, the PA level might be hardwired for some modules, namely ebyte modules.

TMRh20, 2024-02-16:
@Tech500 Thinking we can close this issue now? Hoping you've had success with your project(s)!

2bndy5, 2024-02-18:
I should PR my changes to the IRQ examples to help satisfy the panic problem

https://github.com/nRF24/RF24/compare/master...update-irq-example

PR #956 (2024-03-11): separate ISR handling from IRQ processing
closes #939 

This basically just moves most of the ISR logic into main loop to relieve the MCU from overperforming in IRQ events.
closed 2024-03-11 by 2bndy5