nRF24/RF24 #943: Use Linux kernel's character device API to implement IRQ capability

2bndy5, 2024-02-18:
[This guide](https://blog.lxsang.me/post/id/33) does mention how IRQ could be done using some `poll()` function (which is unknown to me), but it uses a blocking example.

The main obstacle (for me) is using a separate processor thread to poll the GPIO for IRQ edge detection.

_Originally posted by @2bndy5 in https://github.com/nRF24/RF24/issues/942#issuecomment-1950890771_

TMRh20, 2024-02-18:
FYI just tested the interrupt example and pigpio doesn't seem to work on the RPi5 yet.
```
+---------------------------------------------------------+
|Sorry, this system does not appear to be a raspberry pi. |
|aborting.                                                |
+---------------------------------------------------------+
```
Now you have me playing with pthreads again... this is going to take some time. I have a simple example triggering an IRQ using pins 22 or others that are toggled manually, but can't get it to trigger on the radio IRQ pin for some reason. Wondering if it needs to be set to an input or something first? The guide doesn't seem to do that...

Also wondering if I should continue, or if this is something you wanted to take on?

2bndy5, 2024-02-18:
I could take a stab at it, but I'll need you to verify RPi compatibility.

2bndy5, 2024-02-18:
Still researching pthread... Looks like the function passed to `pthread_create()` only executes once in a separate thread -- using default scheduling and thread attrs, I think -- then the thread terminates upon exit of the passed function.

2bndy5, 2024-02-18:
Ok, I'm booted into Ubuntu and inspecting /usr/include/linux/gpio.h. And I just found out that there is a v2 API in gpio.h. Some of the structs used in #942 are actually deprecated already 😡. For example:
```c

/**
 * struct gpiohandle_request - Information about a GPIO handle request
 * @lineoffsets: an array of desired lines, specified by offset index for the
 * associated GPIO device
 * @flags: desired flags for the desired GPIO lines, such as
 * %GPIOHANDLE_REQUEST_OUTPUT, %GPIOHANDLE_REQUEST_ACTIVE_LOW etc, added
 * together. Note that even if multiple lines are requested, the same flags
 * must be applicable to all of them, if you want lines with individual
 * flags set, request them one by one. It is possible to select
 * a batch of input or output lines, but they must all have the same
 * characteristics, i.e. all inputs or all outputs, all active low etc
 * @default_values: if the %GPIOHANDLE_REQUEST_OUTPUT is set for a requested
 * line, this specifies the default output value, should be 0 (low) or
 * 1 (high), anything else than 0 or 1 will be interpreted as 1 (high)
 * @consumer_label: a desired consumer label for the selected GPIO line(s)
 * such as "my-bitbanged-relay"
 * @lines: number of lines requested in this request, i.e. the number of
 * valid fields in the above arrays, set to 1 to request a single line
 * @fd: if successful this field will contain a valid anonymous file handle
 * after a %GPIO_GET_LINEHANDLE_IOCTL operation, zero or negative value
 * means error
 *
 * Note: This struct is part of ABI v1 and is deprecated.
 * Use &struct gpio_v2_line_request instead.
 */
struct gpiohandle_request {
	__u32 lineoffsets[GPIOHANDLES_MAX];
	__u32 flags;
	__u8 default_values[GPIOHANDLES_MAX];
	char consumer_label[GPIO_MAX_NAME_SIZE];
	__u32 lines;
	int fd;
};
```

2bndy5, 2024-02-18:
Yeah, I have plans for later today. But I'll keep at it throughout the coming weeks. I think I'll finish the clang-format updates first.

2bndy5, 2024-02-19:
I've also been looking at the libgpiod code, and they cache everything, possibly because nVidia Jetson/TX2/etc (& probably some RPi clones) can have multiple character devices (`dev/gpiomemX`) for GPIO ports.

2bndy5, 2024-02-19:
Apparently, [MRAA can detect RPi5 hardware](https://github.com/eclipse/mraa/commit/302280363f0e544eb70403e18163d7701028fe11), but I don't know if that means MRAA will completely work on RPi5. They have been supporting character device API for some time...

Pigpio (https://github.com/joan2937/pigpio/issues/586) and BCM2835 ([google groups discussion](https://groups.google.com/g/bcm2835/c/e44ttskxDv8)) libs are still assessing how to implement compatibility. I doubt BCM2835 will go on supporting the new hardware paradigm (which ultimately requires interfacing with PCI express). Pigpio might move forward since it already uses the Linux kernel for some stuff.

Remember, WiringPi is dead and littleWire has been broken since before I joined nRF24 org. It is starting to look like our SPIDEV driver might be the only way to support Linux in the future. I'm not saddened by this from a maintenance point of view, but user projects _will_ suffer this new RPi hardware paradigm.

TMRh20, 2024-02-19:
Yeah, I had a feeling the GPIO changes would be slower than previous implementations, but never did any performance testing. Now that we have some working code for v2, caching everything shouldn't be a big challenge.

TMRh20, 2024-02-19:
Hey if it works it works. We should be able to get the same or better performance out of it than prior versions I would hope.

I agree with keeping the GPIO and IRQ stuff separate. The caching I don't know about, I'd have to look at GPIO code more in depth to form a relevant opinion.

2bndy5, 2024-02-25:
So I think I'm caching the FDs properly (in char-dev-irq branch).
- I have a `gpio_cache` struct closing the FDs in it's destructor, so there shouldn't be any need to modify RF24's destructor.
- I'm only using 1 cached `gpio_v2_line_request` obj for all pins used by RF24; uses `request.config.attrs[offset].attr.flags` instead of `request.config.flags`.
- I also cache 1 `gpio_v2_line_values` obj for all pins' input/output ops.

It compiles and executes without errors, but running the scanner example shows no signal gets detected. 😞 

On a side note, I think our old sysfs approach does not properly free the pins upon app exit. I have to reboot my RPi to allow the char-dev API access to the GPIO22.

TMRh20, 2024-02-25:
Nice work! 

Mostly tests fine for me, but I am seeing differences in the scanner example on RPi5, RPi4 and RPi3. 
The RPi2 seems to not care. 

Teh RPI4 comes up with long lines like this sometimes tho:
`666666655555555555554444444444433333333333333333333333333333333333333333333333333333333333333333333333333334444444444445556666`

TMRh20, 2024-02-25:
Hmm, taking a look at RF24.cpp, I found something unusual that I haven't thought about in a while, but see the lines [here](https://github.com/nRF24/RF24/blob/efdc40aaefffd5001a86724328a1f1e31dc6e280/RF24.cpp#L93C1-L99C2)

Essentially, on faster devices, I put in a delay of 5us when toggling the CS pin. If I modify the code to the following, it seems to work much better on RPi with the scanner example.

```
#if !defined(RF24_LINUX)
    digitalWrite(csn_pin, mode);
#else
    static_cast<void>(mode); // ignore -Wunused-parameter
#endif // !defined(RF24_LINUX)
  delayMicroseconds(csDelay);  // Delay for all devices
}
```

Maybe we could do something like the following for the CS pin?:
```
#if !defined(F_CPU) || F_CPU > 20000000
    delayMicroseconds(5);
#endif
```

The whole point of the delay was double:
a: Ensure the pin gets toggled for at least 5us
b: Slow down the polling via SPI, leaving the radio to process radio data instead of being hammered by SPI requests

2bndy5, 2024-02-25:
I was going to play with char-dev debouncing too. I think with caching, we've hit the too fast problem.

2bndy5, 2024-02-25:
>  its a nice behavior to have

I also set the consumer string in the `request` obj. So, now if you run `gpioinfo gpiochip0` while a RF24 app is running, you'll should see which pins are consumed by the "RF24 lib".

TMRh20, 2024-02-25:
> I think with caching, we've hit the too fast problem.

Yup, there needs to be some sort of GPIO/CSN delay for Linux now along with the faster MCUs. Will you include this in your branch then?

2bndy5, 2024-02-25:
Yeah, coding the debouncing now. There's a limit of 10 attrs that we can configure for each `request.config`, so I'm going with 3: 1 attr to declare which pins are inputs, 1 attr to declare which pins are outputs, and 1 attr to declare the debouncing period of 5 microseconds on each output pin.

2bndy5, 2024-02-25:
I think only 1 type of attr (input flag, output flag, debounce period) can be associated with a pin. Meaning, setting an attr for output and another attr for debouncing on 1 pin doesn't take.
```c
/**
 * struct gpio_v2_line_config - Configuration for GPIO lines
 * @flags: flags for the GPIO lines, with values from &enum
 * gpio_v2_line_flag, such as %GPIO_V2_LINE_FLAG_ACTIVE_LOW,
 * %GPIO_V2_LINE_FLAG_OUTPUT etc, added together.  This is the default for
 * all requested lines but may be overridden for particular lines using
 * @attrs.
 * @num_attrs: the number of attributes in @attrs
 * @padding: reserved for future use and must be zero filled
 * @attrs: the configuration attributes associated with the requested
 * lines.  Any attribute should only be associated with a particular line
 * once.  If an attribute is associated with a line multiple times then the
 * first occurrence (i.e. lowest index) has precedence.
 */
struct gpio_v2_line_config {
	__aligned_u64 flags;
	__u32 num_attrs;
	/* Pad to fill implicit padding and reserve space for future use. */
	__u32 padding[5];
	struct gpio_v2_line_config_attribute attrs[GPIO_V2_LINE_NUM_ATTRS_MAX];
};
```
My initial (local) test has not shown any difference. I pushed what I have in case you can test with it. I also tried using 1 attr to define both output flag and debouncing period (for all output pins), alas I still don't see a difference.

I might have to switch to a `std::map` of `port` -> `request` key/value pairs for each pin used. Alternatively, I guess we could try altering RF24.cpp, but I was hoping to avoid more `ifdef` soup.

2bndy5, 2024-02-25:
My `std::map` attempt yielded same result. Its a pain when Linux kernel doc strings are so terse and online tutorials are scarce or outdated.

I think the debounce period is more for input pins. https://www.kernel.org/doc/html/v4.17/driver-api/gpio/driver.html#gpios-with-debounce-support

I added the `delayMicroseconds()` call to `RF24::ce()`. 😞 https://github.com/nRF24/RF24/commit/18045c7d4e111de1c6bc6d977e6ab3a07fa93d3c
We should probably have a `RF24_SPIDEV` defined so we only delay 5us on CE toggle when using the SPIDEV driver. Other Linux drivers are slow enough to not have this problem.

2bndy5, 2024-02-25:
Seems to be working now. I'm going to take a break before tackling the IRQ implementation...

TMRh20, 2024-02-25:
I'm still getting the same erroneous results with the scanner examples with your current code.

The CE pin is only toggled on transmit, and delays are added in the lib already if needed. re: startWrite()
Adding a delay to CS will affect reception as well as how often available() can be called etc.

2bndy5, 2024-02-25:
But the gpio stuff in SPIDEV isn't managing the CS. The asserted CE used for entering RX is how I perceived the problem with the scanner example. Works fine on my RPi3.

2bndy5, 2024-02-25:
ok, I'm getting periodic failures in gettingstarted on RPi3 followed by very high transmission times. I'm not getting any errors on my RPi4 with gettingstarted. Is this similar to what you're seeing?

2bndy5, 2024-02-25:
I applied your idea locally
```diff
diff --git a/RF24.cpp b/RF24.cpp
index d4a7b33..7d7836e 100644
--- a/RF24.cpp
+++ b/RF24.cpp
@@ -92,10 +92,11 @@ void RF24::csn(bool mode)
 
 #if !defined(RF24_LINUX)
     digitalWrite(csn_pin, mode);
-    delayMicroseconds(csDelay);
+    //delayMicroseconds(csDelay);
 #else
     static_cast<void>(mode); // ignore -Wunused-parameter
 #endif // !defined(RF24_LINUX)
+    delayMicroseconds(csDelay);
 }
 
 /****************************************************************************/
@@ -108,7 +109,7 @@ void RF24::ce(bool level)
 #endif
         digitalWrite(ce_pin, level);
 #ifdef RF24_LINUX
-        delayMicroseconds(5);
+        //delayMicroseconds(5);
 #endif
 #ifndef RF24_LINUX
     }
``` 

It yielded the same result but much slower scanner sweeps. The gettingstarted transmission times were considerably slower but more reliable on my RPi3
```
Transmission successful! Time to transmit = 1018 us. Sent: 0
Transmission successful! Time to transmit = 1023 us. Sent: 0.01
Transmission successful! Time to transmit = 4547 us. Sent: 0.02
Transmission successful! Time to transmit = 1033 us. Sent: 0.03
Transmission successful! Time to transmit = 1029 us. Sent: 0.04
Transmission successful! Time to transmit = 1009 us. Sent: 0.05
Transmission successful! Time to transmit = 1069 us. Sent: 0.06
Transmission successful! Time to transmit = 1033 us. Sent: 0.07
Transmission successful! Time to transmit = 2708 us. Sent: 0.08
Transmission successful! Time to transmit = 1056 us. Sent: 0.09
Transmission successful! Time to transmit = 1022 us. Sent: 0.1
Transmission successful! Time to transmit = 1025 us. Sent: 0.11
```

TMRh20, 2024-02-25:
You are correct I figure, forgot to add the CE delay to RF24.cpp when testing. Now the scanner examples seem to work fine!

My gettingstarted examples seem to work fine with or without the CE delay.

It seems I was getting better high-speed transfers at the RF24Gateway layer. Testing with data over SSH and IPerf3. Now I don't know what the answer is. :p

2bndy5, 2024-02-25:
I'm at a loss as well. This is why I can't have nice things.

TMRh20, 2024-02-25:
I think the CE delay makes more logical sense, so lets go with that for now. We can always adjust later if needed.

2bndy5, 2024-02-26:
Further tests without CE delay seem to be fine in gettingstarted. The scanner example on my RPi4 and RPi3 still require at least a 1us delay. I think the scanner example alone just needs an added delay instead of every call to `ce()`.

TMRh20, 2024-02-28:
I found only one issue so far after a bunch of testing on your char-dev-irq branch:
a: If the radio experiences an error or the radio is simply removed from the Raspberry Pi, it will crash if using RF24Mesh or RF24Gateway.
b: This is due to `radio.begin()` being called a second time and the subsequent call to `pinMode(ce_pin,OUTPUT)`

I think either we could add something in RF24.cpp `if(!pinOpened){ pinMode(ce_pin,OUTPUT); }` or else do something similar in GPIO.cpp, but you may have better ideas on how to handle this.

2bndy5, 2024-02-28:
hmm. I thought I had some code in `GPIO::open()` that would check if the pin specified is already in cache. perhaps that isn't working?
https://github.com/nRF24/RF24/blob/4f434635f36da244f7448827bd0b3677f0c0de91/utility/SPIDEV/gpio.cpp#L95-L100
https://github.com/nRF24/RF24/blob/4f434635f36da244f7448827bd0b3677f0c0de91/utility/SPIDEV/gpio.cpp#L29-L39

2bndy5, 2024-02-28:
> `__u32 i = 0`

maybe use `int 1 = 0;`? Because `offset` is signed int (negative means not in cached).

I think the max lines per request obj is 64, so if we need to use uint32, then the sentinel -1 could instead be anything > 64.

TMRh20, 2024-02-28:
Changing to int just gives a warning when compiling.
How about 
```
    int offset = gpio_cache.getPortOffset(port);
    if (offset < 0) { // pin not in use; add it to cached request
        offset = request.num_lines;
        request.offsets[offset] = port;
        request.num_lines += 1;
    }else{
      return;    
    }
```

2bndy5, 2024-02-28:
IDK, my gut says no because it just ignores the problem and lets `begin()` carry on without proper cache validation. I originally added that code because: What if the user ended up changing the pin's direction (for whatever reason)?

TMRh20, 2024-02-28:
Changing the CE pins direction while the RF24 driver is running? lol 
You might be overthinking again, but I'll leave this one with you to figure out a proper solution.

2bndy5, 2024-02-28:
> You might be overthinking again

Oh, I definitely am. I wouldn't have even worried about it if the code was namespaced. I'm also hoping to code it in a way that can be abstracted as a submodule (separate lib). I would love to be able to have this much Linux kernel wrappers (+ my work on wrapping I2C) for other Arduino libs.

All my work on that branch is in my Ubuntu partition (easier to look at the gpio.h and libgpiod code that way). I'll take a look at it tomorrow. Thanks for the steps to reproduce!

2bndy5, 2024-02-29:
Ok I looked into it, and it looks like the request.fd can't be re-opened once it is already opened.

<details><summary>this patch worked</summary>
<p>

```diff
diff --git a/examples_linux/gettingstarted.cpp b/examples_linux/gettingstarted.cpp
index 78e244a..a1d1187 100644
--- a/examples_linux/gettingstarted.cpp
+++ b/examples_linux/gettingstarted.cpp
@@ -57,6 +57,7 @@ int main(int argc, char** argv)
         cout << "radio hardware is not responding!!" << endl;
         return 0; // quit now
     }
+    radio.begin();
 
     // to use different addresses on a pair of radios, we need a variable to
     // uniquely identify which address this radio will use to transmit
diff --git a/utility/SPIDEV/gpio.cpp b/utility/SPIDEV/gpio.cpp
index aab3dbc..340ce1a 100644
--- a/utility/SPIDEV/gpio.cpp
+++ b/utility/SPIDEV/gpio.cpp
@@ -107,12 +107,15 @@ void GPIO::open(rf24_gpio_pin_t port, int DDR)
         attr_input.mask |= (1LL << offset);
     }
 
-    int ret = ioctl(gpio_cache.fd, GPIO_V2_GET_LINE_IOCTL, &request);
-    if (ret == -1) {
-        std::string msg = "[GPIO::open] Can't get line handle from IOCTL; ";
-        msg += strerror(errno);
-        throw GPIOException(msg);
-        return;
+    int ret;
+    if (request.fd <= 0) {
+        ret = ioctl(gpio_cache.fd, GPIO_V2_GET_LINE_IOCTL, &request);
+        if (ret == -1) {
+            std::string msg = "[GPIO::open] Can't get line handle from IOCTL; ";
+            msg += strerror(errno);
+            throw GPIOException(msg);
+            return;
+        }
     }
     ret = ioctl(request.fd, GPIO_V2_LINE_SET_CONFIG_IOCTL, &request.config);
     if (ret == -1) {
```

</p>
</details>

2bndy5, 2024-02-29:
Now I'm having trouble instantiating another RF24 object. Kinda need to support multiple radio objects if there's more than 1 SPI bus (with multiple CS options for each).

TMRh20, 2024-02-29:
Hmm that might be difficult if keeping the FDs open etc.
Might need to make that optional

2bndy5, 2024-02-29:
I can `pinMode(x, OUTPUT)` multiple pins, but for some reason I can't run `begin()` on multiple RF24 objects.

TMRh20, 2024-02-29:
Hmm, I can't seem to get it working on any pin except 22 even with 1 radio, every other pin I try results in ` [GPIO::write] Can't set line value from IOCTL; Operation not permitted`

2bndy5, 2024-03-01:
I think I need to switch to using a separate request object for each pin. This would involve `std::map`...

2bndy5, 2024-03-01:
Using `std::map` to cache request objects seems to be slower. But at least I can init a second, different pin and another different RF24 object (in the same program).

2bndy5, 2024-03-01:
I figured out that we can use the same `request` object to configure different pins. All we have to cache is the `request.fd` after configuring with `ioctl()`. There's a bit more "[code golfing](https://en.wikipedia.org/wiki/Code_golf)" in f184a2fd9c851bb70fde4cb4c3858b0bf2e420a8, but I like the results (on my RPi3).

TMRh20, 2024-03-02:
Tested the latest on my RPi5, 4 & 3. No problems! I haven't actually looked at the code yet, but it works.

2bndy5, 2024-03-02:
I see the [wiringPi mirror on github just releases v3.0](https://github.com/WiringPi/WiringPi/releases/tag/3.0) with RPi5 support. Looking at the code, they're still using deprecated sysfs calls though, not the recommended char-dev API. I'm guessing to support RPi5, they went with a table to map BCM pin numbers to the newer sysfs pin numbers (which are inexplicably high like 4xx).

But [how they do interrupts](https://github.com/WiringPi/WiringPi/blob/2af87b2c3ecea9fb5e86b405ef2ece4e749eba04/wiringPi/wiringPi.c#L2235-L2394) is interesting because it involves both poll() and pthread (with callbacks to [cached function pointers](https://github.com/WiringPi/WiringPi/blob/2af87b2c3ecea9fb5e86b405ef2ece4e749eba04/wiringPi/wiringPi.c#L406) provided by user code).

TMRh20, 2024-03-02:
Good deal, wiringPi is pretty handy, so I'm glad they are still supporting it. Just tested with our RF24 libs on RPi5 and latest wiringPi and it works well too!

I'm guessing we will need to do something pretty similar for interrupts in SPIDEV.

2bndy5, 2024-03-02:
Yep. Clearly, I'm still researching how to use `poll()` together with pthread... I plan on tackling that very soon.

2bndy5, 2024-03-02:
I just rebased char-dev-irq branch to pull in updates from #952. Transmission times (on my RPi3 and RPi4) are just as good as (or better than) before with using sysfs API. 🚀 And the scanner example works with no additional delay in `ce()`. 🎉

2bndy5, 2024-03-13:
> [!WARNING]
> I have re-based the char-dev-irq branch since merging #959. But I have not tested it at all. The changes in that branch are just what I had stashed locally (mostly theoretical code).

TMRh20, 2024-03-13:
Its on my RPi5 running RPi OS Light as well. Another thing that needs changing although this I wouldn't say has much urgency to change because we have SPI dev pretty much working at full capacity.

2bndy5, 2024-03-13:
I'm not sure if we can keep the cached FDs separate for IRQ and GPIO. The convention is
```cpp
pinMode(24, INTPUT); // caches the request.fd in gpio.cpp

// attachInterrupt() cannot claim pin 24 because
// pinMode() cached an open FD in gpio.cpp
attachInterrupt(24, INT_EDGE_FALLING, &myISR);
```

The real problem is that `GPIO::open()` does configure the pin for polling it as an input (requires additional `request.config.flags` be set, not to mention the debounce time attribute).

I could call `GPIO::close()` from `attachInterrupt()` to deallocate the request.fd corresponding to the specified pin. This would require an actual definition for `GPIO::close()` (which is currently blank).

2bndy5, 2024-03-13:
lgpio repo has an [example using the nRF24L01](https://github.com/joan2937/lg/blob/master/EXAMPLES/lgpio/NRF24.c)! I guess that could be a reference for anyone that wants a nRF24L driver written in pure C (in 1000+ lines). 😬

TMRh20, 2024-03-13:
Heh, I guess that just speaks to the general popularity of these radios. I wonder how long it will take the community in general to adapt to nrf52x and how long it takes to get some high powered versions of the 52x? 
The nrf24 still holds its own pretty well, seeing as we have no radio IRQs for the 52x.

2bndy5, 2024-03-13:
> I wonder how long it will take

too long. probably years after nordic deprecates the nrf5 family. I think the price difference is really significant ([nRF24L01+: <$4/unit](https://www.mouser.com/ProductDetail/Nordic-Semiconductor/NRF24L01P?qs=yqaQSyyJnNiKlgjT8AqoOw%3D%3D) & [nRF52840: >$6/unit](https://www.mouser.com/ProductDetail/Nordic-Semiconductor/NRF52840-QIAA-F-R?qs=XAiT9M5g4x8tGs2UCi1rcg%3D%3D) when not buying massive reels).

I've been waiting for someone to create a board based on the [nRF5340](https://www.mouser.com/ProductDetail/Nordic-Semiconductor/nRF5340-CLAA-R?qs=81r%252BiQLm7BTUbHQUWWib8A%3D%3D), which released during the pandemic and was poorly received due to the chip shortage. The main advantage of the nRF5240 is BLE audio capability, which implies dual-core processor to de/encode audio codec in realtime. Imagine a DIY Star Trek TNG communicator...

2bndy5, 2024-03-13:
I've been playing around on the char-dev-irq branch... The interruptConfigure example hangs when it gets to `pthread_create()`. IDK if I'm configuring the line request properly. I'm also not sure if I'm using `poll()` properly with the [char-dev API](https://docs.kernel.org/next/userspace-api/gpio/chardev.html).

TMRh20, 2024-03-13:
Don't ask me I just work here... and the pay is not too hot either. 
I can take a closer look soon hopefully, been a long time since I did anything with pthreads.

Avamander, 2024-03-14:
> how long it takes to get some high powered versions of the 52x?

Depends on what you mean by high-powered version, there's the nRF21540 and nRF54H20 for two types of power by Nordic, there are a few designs with the former out there last I saw.

2bndy5, 2024-03-14:
I thought he meant "better PA/LNA". The [nRF21540](https://www.nordicsemi.com/products/nrf21540) looks like it could get the job done but only as a complement to the nRF52 or nRF53 series chips.

TMRh20, 2024-03-14:
Yeah I meant PA + LNA / Long range, but it looks like there are already variants with these features. Nice! Thanks for the info.

TMRh20, 2024-03-14:
> The interruptConfigure example hangs when it gets to pthread_create()...

Played around for a bit. I don't know whats going on so far. Even if I empty the poll_irq function and just leave a loop the main thread hangs. It returns fine if I remove the loop. Seems to be something specific to pthreads though I think. Out of time for today, but will try again.

TMRh20, 2024-03-14:
https://github.com/nRF24/RF24/blob/char-dev-irq/utility/SPIDEV/interrupt.cpp#L127 needs to be declared outside this function, since the function returns as soon as the thread is created. That seems to be the cause of the initial hang.

Then yes, I think there is something wrong with the polling as well. Its constantly returning > 0

[47 more replies]

PR #961 (2024-03-19): IRQ support using Linux kernel Character Device + Posix threads
resolves #943

I think its ready. Given the fact that nRF24/RF24Gateway#32 relies on these changes, I'm ready to move forward and patch things up as (or if) they occur.

# Interrupt changes
- 💯 Use Linux kernel's character device API to implement `attachInterrupt()` and `detachInterrupt()` for both SPIDEV and RPi drivers. Combined with Posix threads, this gives a very quick and reliable behavior that one would expect from microcontroller platforms.
- wraps wiringPi's `wiringPiISR()` into `attachInterrupt()` for better compatibilit [... cut]
closed 2024-03-19 by 2bndy5