nRF24/RF24 #946: [Question] R24.h, testRPD(void) "Strong signal > 64dBm" : "Weak signal < 64dBm"

hrolofs, 2024-02-24:
Hi,

Just a question about the testRPD function. I am a little confused about the description of testRDP() in RF24.h

```
    /**
     * Test whether a signal (carrier or otherwise) greater than
     * or equal to -64dBm is present on the channel. Valid only
     * on nRF24L01P (+) hardware. On nRF24L01, use testCarrier().
     *
     * Useful to check for interference on the current channel and
     * channel hopping strategies.
     *
     * @code
     * bool goodSignal = radio.testRPD();
     * if(radio.available()){
     *    Serial.println(goodSignal ? "Strong signal > 64dBm" : "Weak signal < 64dBm" );
     *    radio.read(0,0);
     * }
     * @endcode
     * @return true if a signal less than or equal to -64dBm was detected,
     * false if not.
     */
    bool testRPD(void);

```

The mixing of positive and negative values confuses me. It would be easier if all values in this description were negative.

Please correct me if I am wrong.

-50dBm is better than -64dbm (-50 > -64)
-70dBm is worse than -64dBm (-70 < -64)
(all values are negative)

Shouldn't it then read:
Strong signal > -64dBm, Weak signal < -64dBm

This is also misleading for me.

```
     * Test whether a signal (carrier or otherwise) greater than
     * or equal to -64dBm is present on the channel.

     * @return true if a signal less than or equal to -64dBm was detected,
```

mixing "greater than or equal" and "less than or equal"
   
```

2bndy5, 2024-02-24:
We have also found from using reverse engineering tactics (for sniffing packets from a non-nRF24L01 radios) that general listening can be more sensitive (> -80 dBm IIRC) than `testRPD()` reports.

The main discrepancy is that the RPD flag in the radio's registers was added for compliance testing mandated by government regulations. Meaning `testRPD()` satisfies the required carrier wave [hardware] tests, but it does not fully describe the actual performance during RX operation. We have a [promiscuous-scanner](https://github.com/nRF24/RF24/tree/promiscuous-scanners) branch to experiment with this (and improve the scanner examples).

TMRh20, 2024-02-24:
> ...a "Weak signal" could also mean "no signal". We have no way of knowing given the nRF24L01's SPI intrerface. 

Hmm, I think one could call the `available();` function immediately after calling `testRPD();` or `testCarrier();`

```
        radio.startListening();
        delayMicroseconds(128);
        radio.stopListening();

        // Did we get a carrier?
        if (radio.testCarrier()) {
          ++values[i];
        }else
        if(radio.available()){
          ++weakValues[i];
        }

```

Not tested, but maybe it could work?

Incoming commit and PR, please verify I got it right this time lol. @2bndy5 @hrolofs

2bndy5, 2024-02-24:
I think it would work with a non-plus variant. The carrier wave test setup on non-plus variants was very much like sending a payload full of `0xFF` with no ACKs.

hrolofs, 2024-02-24:
Many thanks to everyone for the quick response to my questions and the fix in the documentation. :-)

closed 2024-02-25 by TMRh20