nRF24/RF24 #960: reorganize examples

2bndy5, 2024-03-17:
I want to reorganize the examples into 1 folder:
```
examples/
  linux/
  pico/
  python/
  <all Arduino examples remain as is>
```

<details><summary>Note, all Arduino examples already have their own folder in the examples folder</summary>

```
  AcknowledgementPayloads/
  GettingStarted/
  ManualAcknowledgements/
  old_backups/
  scanner/
  StreamingData
  encodeRadioDetails/
  InterruptConfigure/
  MulticeiverDemo/
  rf24_ATTiny/
  scannerGraphic/
```

</details> 

It is just getting annoying to navigate. And the shell's tab-completion would work better this way.

Of course, all docs about building/running examples would need updating.

2bndy5, 2024-03-17:
Furthermore the datasheets/ and images/ folder can get put into the docs/ folder.

The only thing stopping me from suggesting that we put all lib sources into a src/ folder is the fact that the Arduino IDE will automatically try to compile everything found in a lib's src/ folder (if it exists). That would require all non-arduino utility/ sources to have an extra `#ifndef ARDUINO` wrapping the non-Arduino source code (including the copied bcm2835 lib).

TMRh20, 2024-03-17:
Not sure if this is a good idea, this will change all links to examples online also, but I guess I won't object. :p

2bndy5, 2024-03-17:
I'm not worried about links from third-party tutorials. Any incorrect link to github sources will direct users toward the repo root anyway. An incorrect link: https://github.com/nRF24/RF24/tree/master/examples/linux

Plus, it is kind of annoying that RF24Network and RF24Mesh use examples_RPi (note capitol letters) even though they are really general linux examples. I had to use [an array of examples' paths for the different RF24* libs in the installer script](https://github.com/nRF24/.github/blob/3cd9423b3667d0bf02875c20d1a20a2178231f80/installer/install.sh#L14). And, the [reusable CI workflow has to take a special input for the examples path](https://github.com/nRF24/.github/blob/3cd9423b3667d0bf02875c20d1a20a2178231f80/.github/workflows/build_linux_cmake.yaml#L18-L20) (not to mention the inconsistent [py wrapper locations which had to be another input](https://github.com/nRF24/.github/blob/3cd9423b3667d0bf02875c20d1a20a2178231f80/.github/workflows/build_linux_cmake.yaml#L24-L26)). All this because the repo structures aren't completely in sync, but I understand the different repos' history has caused this diversion.

2bndy5, 2024-03-17:
Less clutter in repo root means more chances that people will actually see/read the README when browsing.

I think al the links in the Doxygen docs change every time we publish a new docs build too.

2bndy5, 2024-03-18:
I created a github private project called "v2.0 ideas" and added this issue and #926 to it.

Closing this until we're ready to make significant changes toward v2.0.