nRF24/RF24 #972: [Question] Address names used in examples problematic?

MaartenJB, 2024-04-02:
Hi,

I've been reading the rf24 class documentation from here, and found out that not all address names are feasible to use.

If I read the supplied link from the docs:
http://maniacalbits.blogspot.com/2013/04/rf24-addressing-nrf24l01-radios-require.html

Am I right to think that '1Node' should not be used? The first and last byte seem to contain an unwanted nibble.

NodeId str: 1Node
NodeId hex: $31 $4E $6F $64 $65

Avoid these nibbles hex: $A, $5, $2, $1
Avoid these nibbles bin: 1010, 0101, 0010, 0001

Best regards,

Maarten

2bndy5, 2024-04-02:
There's a little more to it. Arrays in C++, like an array of characters (aka string), are stored in memory using little endian. Fortunately, the nRF24L01 expects the address (& payloads) to be transfered over SPI in little endian... Essentially, `"1Node"` is stored is actually used as `0x65646F4E31`. This is why `printDetails()` (and `printPrettyDetails()`) outputs the addresses bytes backwards:
```python
>>> from pyrf24 import RF24
>>> radio = RF24(22, 0)
>>> radio.begin()
True
>>> radio.openReadingPipe(1, b"1Node")
>>> radio.printPrettyDetails()
================ SPI Configuration ================
CSN Pin                 = /dev/spidev0.0
CE Pin                  = Custom GPIO22
SPI Frequency           = 10 Mhz
================ NRF Configuration ================
Channel                 = 76 (~ 2476 MHz)
Model                   = nRF24L01+
RF Data Rate            = 1 MBPS
RF Power Amplifier      = PA_LOW
RF Low Noise Amplifier  = Enabled
CRC Length              = 16 bits
Address Length          = 5 bytes
Static Payload Length   = 32 bytes
Auto Retry Delay        = 1500 microseconds
Auto Retry Attempts     = 15 maximum
Packets lost on
    current channel     = 0
Retry attempts made for
    last transmission   = 0
Multicast               = Disabled
Custom ACK Payload      = Disabled
Dynamic Payloads        = Disabled
Auto Acknowledgment     = Enabled
Primary Mode            = TX
TX address              = 0x65646f4e31
pipe 0 ( open ) bound   = 0x65646f5555
pipe 1 ( open ) bound   = 0x65646f4e31
pipe 2 (closed) bound   = 0x0a
pipe 3 (closed) bound   = 0xa0
pipe 4 (closed) bound   = 0x00
pipe 5 (closed) bound   = 0xab
```
to see it binary (again using python for brevity)
```python
>>> address = b'edoN1'
>>> [bin(x) for x in address]
['0b1100101', '0b1100100', '0b1101111', '0b1001110', '0b110001']
```
so
| char | binary |
|-----:|:-------|
| `e` | `01100101` |
| `d` | `01100100` |
| `o` | `01101111` |
| `N` | `01001110` |
| `1` | `00110001` |

Given that `1` in binary starts with a `0011` is not as problematic as anything that starts with a `1010` or `0101`.

I would be more concerned with addresses that start with `T`
```python
>>> bin(b'T'[0])
'0b1010100'
```

2bndy5, 2024-04-02:
Also, note the scanner examples use the worst possible addresses (with an unofficially supported length of 2 bytes) on purpose:
https://github.com/nRF24/RF24/blob/5d645ae382df94f279adfd842a7318fd4cdbc635/examples_linux/scanner.py#L30-L40

MaartenJB, 2024-04-02:
wow, fast response :-) thx.

Is the letter 'e' not as problematic because only the last nibble is 0101 ? The article only mentioned to not use these four nibbles.

2bndy5, 2024-04-02:
> Is the letter 'e' not as problematic because only the last nibble is 0101 ?

No, because it starts with a `0110`. In general, anything that starts with a repeating and alternating pattern of 1 and 0. should be avoided. Recall what the datasheet says:

> 7.3.1 Preamble
> The preamble is a bit sequence used to synchronize the receivers demodulator to the incoming bit stream. The preamble is one byte long and is either 01010101 or 10101010. If the first bit in the address is 1 the preamble is automatically set to 10101010 and if the first bit is 0 the preamble is automatically set to 01010101. This is done to ensure there are enough transitions in the preamble to stabilize the receiver.

The article that the docs link to is a bit more empirical as it is written from observations/experiments.

2bndy5, 2024-04-02:
Other radio's may actually use a 2-byte (or more) preamble. IIRC, the RFM69 uses 2-4 byte preambles, but they are sub-GHz transceivers (so they might require longer time to demodulate signals seen from the antenna).

MaartenJB, 2024-04-02:
Aah ok, what I read from that is the more the bit pattern differs from the preamble, the easier it is for the receiver to calibrate to the signal. And if the first part of the byte is already different, that should be sufficient for the reader to do that.

closed 2024-04-02 by 2bndy5

MaartenJB, 2024-04-02:
Sure, thanks for the fast detailed explanation!