nRF24/RF24 #975: Unable to use SPI1 on the raspberry pi pico

s07dd, 2024-05-05:
I want to use the NRF module hooked up to the SPI1 pins on the RPi Pico... 
I learnt from the previous issues and came up this this configuration for the SPI1 pin setup: 
MISO = 8 
MOSI = 11 
SCK = 10
CSN = 9

The issue is that just cant get it to work.
Code I'm using to assign the SPI1 Pins in setup:
```
   SPI1.setRX(8);
   SPI1.setTX(11);
   SPI1.setSCK(10);
   SPI1.begin();
```
I do have the information that the default SPI pins of the pico are being used inspite of assigning them like its mentioned in the code above.

I tried to assign the pins to SPI0 and it did work!! Pins 0, 1, 2, 3 to be specific using the same code mentioned above (using just "SPI" instead of "SPI1" and the proper pin numbers ofcourse). And I made zero changes to any of the library files.

But SPI1 isn't working.
Am I doing something wrong or is it required to make and changes to the library config files.

Please help me out, thanks.

2bndy5, 2024-05-05:
The [PicoSDK docs have a table outlining the GPIO pins' capabilities](https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#autotoc_md0). According to that table, the pins you selected seem correct.

Did you pass the `SPI1` object to `radio.begin()`?
```cpp
   SPI1.setRX(8);
   SPI1.setTX(11);
   SPI1.setSCK(10);
   SPI1.begin();
   if (radio.begin(&SPI1)) {
      return 1; // radio hardware not responding
   }
```
Did you review our doc about [using the Arduino platform with a secondary SPI bus](https://nrf24.github.io/RF24/md_docs_2arduino.html#autotoc_md16)?

2bndy5, 2024-05-05:
From the Arduino-Pico docs:
> the following RP2040-specific changes:
> 
> - SPI.begin(bool hwCS) can take an options hwCS parameter.
> 
> By passing in true for hwCS the sketch does not need to worry about asserting and deasserting the CS pin between transactions. The default is false and requires the sketch to handle the CS pin itself, as is the standard way in Arduino.

So passing `false` would be preferable since the RF24 library takes care of the CSN pin independently from the other SPI pins.
```cpp
    // let RF24 lib manage the CSN pin

    SPI1.begin(false);
    // or equivalently 
    SPI1.begin();
```

s07dd, 2024-05-08:
My bad, I hadn't taken a look in the documentation that you mentioned above. 
&SPI1 inside the bracket is all that was required. 😂 I'm a complete noob.
Everything works flawlessly now! 
Always grateful to you guys, you do so much for the community
Your work cannot be appreciated enough
Thanks a ton for your help!!!!

closed 2024-05-08 by s07dd