nRF24/RF24 #976: NRF24L01 does not show all characters in the receiver

saeidbina, 2024-05-07:
Hello everyone. I am using Arduin mega (with RTC and DH22) and I wanna receive this values in receiver (UNO R4). The main problem is that in the receiver serial monitor does show all characters. When I increase the size of the payload or text in both, does not make change. Here are my code: The only way to include humidity and the rest of the characters is to delete some of the previous characters for example deleting Data or time or some of the empty characters in between.

```
[#include](tg://search_hashtag?hashtag=include) <SPI.h>
[#include](tg://search_hashtag?hashtag=include) <nRF24L01.h>
[#include](tg://search_hashtag?hashtag=include) <RF24.h>
[#include](tg://search_hashtag?hashtag=include) <DHT.h>
[#include](tg://search_hashtag?hashtag=include) <RtcDS1302.h>

RF24 radio(33, 32); // CE, CSN

const byte address[6] = "00001";

[#define](tg://search_hashtag?hashtag=define) DHTPIN 35     // Digital pin connected to the DHT sensor 
[#define](tg://search_hashtag?hashtag=define) DHTTYPE    DHT22     // DHT 22 (AM2302)

DHT dht(DHTPIN, DHTTYPE);

ThreeWire myWire(39, 41, 37); // IO, SCLK, CE
RtcDS1302<ThreeWire> Rtc(myWire);

void setup() {
  Serial.begin(9600);

  radio.begin();
  radio.openWritingPipe(address);
  radio.setPALevel(RF24_PA_MAX);
  radio.stopListening();

  dht.begin();

  Rtc.Begin();

  Serial.print("compiled: ");
  Serial.print(DATE);
  Serial.println(TIME);

  RtcDateTime compiled = RtcDateTime(DATE, TIME);
  printDateTime(compiled);
  Serial.println();

  if (!Rtc.IsDateTimeValid()) {
    Serial.println("RTC lost confidence in the DateTime!");
    Rtc.SetDateTime(compiled);
  }

  if (Rtc.GetIsWriteProtected()) {
    Serial.println("RTC was write protected, enabling writing now");
    Rtc.SetIsWriteProtected(false);
  }

  if (!Rtc.GetIsRunning()) {
    Serial.println("RTC was not actively running, starting now");
    Rtc.SetIsRunning(true);
  }

  RtcDateTime now = Rtc.GetDateTime();
  if (now < compiled) {
    Serial.println("RTC is older than compile time!  (Updating DateTime)");
    Rtc.SetDateTime(compiled);
  } else if (now > compiled) {
    Serial.println("RTC is newer than compile time. (this is expected)");
  } else if (now == compiled) {
    Serial.println("RTC is the same as compile time! (not expected but all is fine)");
  }
}

void loop() {
  float temperature = dht.readTemperature();
  float humidity = dht.readHumidity();

  RtcDateTime now = Rtc.GetDateTime();

  // Convert temperature, humidity, date, and time to strings
  char temperatureStr[6];
  dtostrf(temperature, 4, 2, temperatureStr); // Format: "%.2f"

  char humidityStr[6];
  dtostrf(humidity, 4, 2, humidityStr); // Format: "%.2f"

  char dateString[12];
  sprintf(dateString, "%04u/%02u/%02u", now.Year(), now.Month(), [now.Day](https://now.day/)());

  char timeString[9];
  sprintf(timeString, "%02u:%02u:%02u", now.Hour(), now.Minute(), now.Second());

  // Print date, time, temperature, and humidity in one line
  Serial.print(dateString);
  Serial.print(" ");
  Serial.print(timeString);
  Serial.print(" T:");
  Serial.print(temperatureStr);
  Serial.print("°C H:");
  Serial.print(humidityStr);
  Serial.println("%");

  // Construct message payload
  char payload[500];
  sprintf(payload, "%s %s T:%s°C H:%s%%", dateString, timeString, temperatureStr, humidityStr);

  // Send payload via NRF24L01
  radio.write(&payload, sizeof(payload));

  delay(1000); // Adjust delay as needed
}

void printDateTime(const RtcDateTime& dt) {
  char datestring[50];
  snprintf_P(datestring, 
            sizeof(datestring),
            PSTR("%04u/%02u/%02u %02u:%02u:%02u"),
            dt.Year(),
            dt.Month(),
            [dt.Day](https://dt.day/)(),
            dt.Hour(),
            dt.Minute(),
            dt.Second() );
  Serial.print(datestring);
}
```

My Reciever:

[#include](tg://search_hashtag?hashtag=include) <SPI.h>
[#include](tg://search_hashtag?hashtag=include) <nRF24L01.h>
[#include](tg://search_hashtag?hashtag=include) <RF24.h>

RF24 radio(7, 8); // CE, CSN

const byte address[6] = "00001";

void setup() {
  Serial.begin(9600);
  radio.begin();
  radio.openReadingPipe(0, address);
  radio.setPALevel(RF24_PA_MAX);
  radio.startListening();
}

void loop() {
  if (radio.available()) {
    char text[500] = ""; // Increase the buffer size to ensure it can hold the entire payload
    [radio.read](https://radio.read/)(&text, sizeof(text));

    
    // Print the received payload
    Serial.println(text);

  }
}

Serial monitor: 
Sender:
2024/05/07 15:02:07 T:25.90°C H:48.90%

reciever:
2024/05/07 15:02:12 T:25.90°C H

2bndy5, 2024-05-07:
This is not a RF24 lib problem, rather it is a combination of misusing c-str data and misunderstanding the hardware limitations of hte nRF24L01 radio.

## Lets talk c-str data
```cpp
  char humidityStr[6];
```
Here, you allocate 6 bytes for string to represent a percentage (usually ranging [0, 100]) with 2 decimal places. So, max is `100.00`% (6 characters) and min is `0.00`%. At this point, it is important to note that the `hunidityStr` does not have enough capacity for the c-str's required null terminating character (`'\0'`) when the float value is at max (`100.00`).

## Lets talk RF24 payload size
Using RF24, payloads can only be 32 bytes maximum.
```shell
# this is 32 characters (32 bytes of UTF-8 encoded data)
2024/05/07 15:02:07 T:25.90°C H:

# this is 38 characters (38 bytes of UTF-8 encoded data)
2024/05/07 15:02:07 T:25.90°C H:48.90%
```
Increasing the buffer beyond 32 bytes is pointless because the nRF24L01 will only handle payloads with a maximum of 32-bytes. BTW, increasing the size of a buffer of uninitialized bytes does not guarantee that the c-str data will end in a `'\0'` because not all compilers use implicit initialization values (that's up the Arduino core you are using for the Uno R4).

### Some other hints
- use `if (!radio.begin())` to detect bad radios or incorrect wiring.
- using `sprintf()` instead of `dtostrf()` can tell you how many bytes were stored to the c-str. The byte count in turn can be used as the payload length...

saeidbina, 2024-05-07:
thank you much for your reply. Could you please help me? the final project will have 12 sensors in a MEGA board that will be sent to UNO using nrf24. I am begginer.

2bndy5, 2024-05-07:
Sorry, but no. We try to focus on actual library problems here. You could try your luck with [Arduino forums](https://forum.arduino.cc/).

> project will have 12 sensors in a MEGA board that will be sent to UNO using nrf24

Sounds like a good scenario for using [RF24Network](https://github.com/nRF24/RF24Network). It allows for payloads greater than 32 bytes and the network can have hundreds of nodes instead of just 7 nodes. The only obstacle for beginners is understanding the [network topology](https://nrf24.github.io/RF24Network/md_docs_2tuning.html)