nRF24/RF24 #977: Problem using stopListening() and startListening()

V1LL4pro, 2024-05-13:
Good morning guys, I’ve been having problems with a code I’m working on in Arduino, which involves bidirectional communication between a TX and RX.

My problem is the following, when pressing a button the transmitter should send a byte with 00000000, this works well the first time I do it, however when pressing the button a second time or more the message is not sent and failureDetected detects an error, I have tried to remove radio.startListening() and the code works correctly, however this should not happen or have any problem.

Below, I summarize a fragment of the code. Does anyone have an idea of the problem?

```
const uint8_t PROGMEM Pipes[6][5] = { { "1Pipe" }, { "2Pipe" }, { "3Pipe" }, { "4Pipe" }, { "5Pipe" }, { "dPipe" } };

bool EnvioTransmisor() {  // Envio de mensajes por Transceptor (lado Transmisor)
  radio.stopListening();          // Transceptor modo para transmitir
  Dato[0] = 0b00000000;          // Identificador
  bool var2 = radio.write(&Dato, 1);  // Solo enviar el primer byte del vector Data
  radio.startListening();  // Transceptor modo para recibir
  return var2;
}

void setup() {
  radio.begin();

  radio.setChannel(Canal);          // Canal que trabajara el Transceptor
  radio.setPALevel(RF24_PA_MIN);    // Potencia de trabajo Maxima 0dbm
  radio.setDataRate(RF24_250KBPS);  // Velocidad de 250Kbps

  radio.openReadingPipe(0, pgm_read_word_near(&Pipes[5]));          // Asigna pipe de lecura por defecto
  radio.openWritingPipe(pgm_read_word_near(&Pipes[5]));             // Asigna pipe de escritura por defecto
}
```

Note: I am using an interrupt to detect the button press, however this does not directly call the function, it only changes a variable so that later in the void loop() the EnvioTransmisor() function is called.

2bndy5, 2024-05-13:
Is there any point in the receiver code that effectively stops listening after the first payload is received?

V1LL4pro, 2024-05-13:
It turns out that it was all **my** mistake, I was shortening my code to check the error in its main parts, when I noticed that I was using `RF24 radio(9, 10);` instead of `RF24 radio(8, 10);` as I have in my assemblies.

I can’t believe I wasted 12 hours of my time reviewing all my 700-line code for such a small error coming from a single number, anyway thank you for your help, everything is now solved.

closed 2024-05-13 by V1LL4pro

2bndy5, 2024-05-13:
> I can’t believe I wasted 12 hours of my time reviewing ...

Welcome to the club. We've all been there and still run the risk of doing it again. 😉