nRF24/RF24 #978: [Question] Some modules take way longer to write()

Luddi96, 2024-06-04:
My code snipped running in loop():
´´´
float payload = 1.0;
unsigned long microsbef = micros();
radio.write(&payload, sizeof(float));
Serial.println(micros() - microsbef);
  ´´´
With one module, the write() takes 2700us, like expected, with another it also took 2700us before but now suddenly takes almost 100**m**s
Any ideas? Module defect?

Here is the dump of both modules:
[image]
[image]

TMRh20, 2024-06-04:
This is likely not an issue with the library, but with the modules you have. I would suggest that the retry periods aren't working as intended. I would assume that the radios are just having trouble communicating, in case of the PA+LNA modules, you can try shielding if none is present per https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md

What kind of modules are you using and which work, which don't?

Luddi96, 2024-06-04:
_What kind of modules are you using and which work, which don't?_
It's always the same module, this basic one: https://de.aliexpress.com/item/1005005642753224.html
At the moment they are sending without Rx present, so I want them to not reach any destination. In this case I expect to have the timeout after 2.7us, as per retries and time defined. But one module, that previously did 2.7us now does 98000us... So probably a hardware defect?

_2.7 ms for a single 8-byte transmission seems rather long._
It's 2.7 microseconds, not milliseconds

2bndy5, 2024-06-04:
I've never considered aliexpress to be a good source for these radios, rather I think of aliexpress amazon but much much worse. The first review on that store page calls them clones. FYI, its not a good sign if there is no writing printed on the chip itself. Since the nRF24L01 chip goes for $2 or more on mouser.com when purchased in mass quantities, I think its safe to say that a $1 radio is not using an authentic nRF24L01 chip.

The 100 ms timeout used (exclusively) in `RF24::write()` often indicates a wiring problem. But it is not specific to wiring; it simply means the radio is not behaving as the library expects it to. Also, you can check [`RF24::failureDetected`](https://nrf24.github.io/RF24/classRF24.html#a2e40fe66d1231a333aa2534e8491f828) to after calling `write()` because the 100 ms timeout will assert that boolean attribute.

Luddi96, 2024-06-04:
Yeah, its not the best source for sure. Out of 100 modules I had 5 with a defect from production, thats ok for me. None of them use a genuine IC for sure, but all work properly so far (except this one which died during operation)
So I will book this module as "death in service" and close this case.
Thanks for the hints. My hope was that I just overlooked some register which now magically changed the content, producin this behavior :D

closed 2024-06-04 by Luddi96

Avamander, 2024-06-04:
There's a Hackaday article about identifying counterfeit nRF24L01+ chips, it might help identifying them in the future.