nRF24/RF24 #979: can't find RF24 method inside RF24 module

FalconCode10, 2024-06-10:
after installing RF24 module on raspberry pi, i successfully import module
also could import all methods using the following command:

from RF24 import *

till now there is no problem.
when i tried to use a receiver code, first command was

from RF24 import RF24

by running code, there is an error: unable to locate RF24    
any help please

2bndy5, 2024-06-10:
I need a little more information to help you.
1. How exactly did you install the python module?
2. Are you using `sudo`?

FalconCode10, 2024-06-10:
I used virtual environment because raspberry refused to install RF24 and many other modules like numpy

2bndy5, 2024-06-10:
You still haven't answered my questions. 

Are you following [our documented instructions about installing the python wrapper](https://nrf24.github.io/RF24/md_docs_2python__wrapper.html)?

FalconCode10, 2024-06-14:
your document talk about Linux not raspbian

FalconCode10, 2024-06-14:
also i tried more than method to install your module, but every time after i finished this message appeared to me: "unable to locate xyz"

xyz is the module or in the best case it's the RF24 method inside RF24 module

2bndy5, 2024-06-15:
I can't help you if you don't tell me the exact commands that you are typing.

It would be better if you posted the exact error message instead of telling me what _you think_ the error message means. I doubt that it has trouble locating the RF24 _class_ (not method) if it can locate the RF24 _module_. Usually, this is because
- you used `sudo` to install the module but not to execute the example
- you used `sudo` to execute the example but not to install the module

Either way don't use `sudo`! This is talked about in the note at the end of the install instructions (especially regarding the use of virtual environments or venv).

Raspbian (which has been renamed to "RPi OS") is a flavor of Linux. The install instructions are the same for any flavor of Linux (even Ubuntu).

I'm also sensing that English is not your primary language. You can write here in your preferred language, and I'll just use Google translate on my end. Hopefully that helps.

FalconCode10, 2024-06-15:
this is the last step i reached
"
>> import RF24
>>> from RF24 import RF24
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'RF24' from 'RF24' (unknown location)
"
you can see that i can import RF24 without any problems, but once i tried to import RF24 method from RF24 module, it can't be done
dear, you are right, English is not my native language, but if i used google translate it will be more difficult that's because google translate, translate literal translation

2bndy5, 2024-06-15:
By following both the
1. [instructions to build and install the C++ RF24 library](https://nrf24.github.io/RF24/md_docs_2using__cmake.html).
2. [instructions to build and install the python wrapper](https://nrf24.github.io/RF24/md_docs_2python__wrapper.html)

I'm still not able to reproduce this error message.

## I REALLY NEED TO KNOW THE COMMANDS YOU USE TO INSTALL THE RF24 MODULE!

FalconCode10, 2024-06-15:
sudo apt update
sudo apt upgrade -y
sudo apt install -y build-essential cmake python3-dev python3-pip python3-venv git

cd ~
python3 -m venv rf24_env
source rf24_env/bin/activate

git clone https://github.com/nRF24/RF24.git

cd ~/RF24
mkdir build
cd build
cmake ..
make
sudo make install

cd ~/RF24/pyRF24
python3 setup.py build
python3 setup.py install

2bndy5, 2024-06-15:
## before installing updates
It is important to delete old build artifacts and uninstall any previous pyRF24 module.
```
cd ~/RF24
sudo rm -r build/
rm Makefile.inc
cd pyRF24
rm -r build/ __pycache__/ RF24.egg-info/
cd ../
sudo rm /usr/local/lib/librf24.*
```
Then repeat all the ommands you posted.

> [!TIP]
> Please explicitly use the SPIDEV driver. Replace the command
> ```
> cmake ..
> ```
> with
> ```
> cmake .. -DRF24_DRIVER=SPIDEV
> ```

> [!IMPORTANT]
> The following commands are no longer encouraged. These commands:
> ```
> python3 setup.py build
> python3 setup.py install
> ```
> should be replaced with
> ```
> python3 -m pip install .
> ```

2bndy5, 2024-06-15:
If you are seeing a build error, then you should report that as well.

The following is an error:
```
      pyRF24.cpp:1:10: fatal error: boost/python.hpp: No such file or directory
          1 | #include <boost/python.hpp>
            |          ^~~~~~~~~~~~~~~~~~
      compilation terminated.
```
This means you need to install boost.python library first:
```
sudo apt install libboost-python-dev
sudo ln -s $(ls /usr/lib/$(ls /usr/lib/gcc | tail -1)/libboost_python3*.so | tail -1) /usr/lib/$(ls /usr/lib/gcc | tail -1)/libboost_python3.so
```
Then try `python -m pip install .` again.

FalconCode10, 2024-06-15:
last update dear

sudo apt install libboost-python-dev
sudo ln -s $(ls /usr/lib/$(ls /usr/lib/gcc | tail -1)/libboost_python3*.so | tail -1) /usr/lib/$(ls /usr/lib/gcc | tail -1)/libboost_python3.so
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libboost-python-dev is already the newest version (1.74.0.3).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
ln: failed to create symbolic link '/usr/lib/aarch64-linux-gnu/libboost_python3.so': File exists

2bndy5, 2024-06-15:
Ok, the "File exists" output is good. It means that pyRF24/setup.py can build as expected.

Has any of this helped?
Are you still getting the `ImportError`?

I am also curious, what does your output for the following command?
```
pip list
```

FalconCode10, 2024-06-16:
moshtohor@raspberrypi:~ $ pip list
Package                            Version
---------------------------------- ----------
arandr                             0.1.11
asgiref                            3.6.0
astroid                            2.14.2
asttokens                          2.2.1
av                                 10.0.0
Babel                              2.10.3
beautifulsoup4                     4.11.2
blinker                            1.5
certifi                            2022.9.24
chardet                            5.1.0
charset-normalizer                 3.0.1
click                              8.1.3
colorama                           0.4.6
colorzero                          2.0
cryptography                       38.0.4
cupshelpers                        1.0
dbus-python                        1.3.2
dill                               0.3.6
distlib                            0.3.6
distro                             1.8.0
docutils                           0.19
filelock                           3.9.0
Flask                              2.2.2
gpiozero                           2.0
html5lib                           1.1
idna                               3.3
importlib-metadata                 4.12.0
isort                              5.6.4
itsdangerous                       2.1.2
jedi                               0.18.2
Jinja2                             3.1.2
lazy-object-proxy                  1.9.0
lgpio                              0.2.2.0
libevdev                           0.5
logilab-common                     1.9.8
lxml                               4.9.2
MarkupSafe                         2.1.2
mccabe                             0.7.0
meson                              1.0.1
more-itertools                     8.10.0
mypy                               1.0.1
mypy-extensions                    0.4.3
numpy                              1.24.2
oauthlib                           3.2.2
olefile                            0.46
parso                              0.8.3
pexpect                            4.8.0
pgzero                             1.2
picamera2                          0.3.18
pidng                              4.0.9
piexif                             1.1.3
pigpio                             1.78
Pillow                             9.4.0
pip                                23.0.1
platformdirs                       2.6.0
psutil                             5.9.4
ptyprocess                         0.7.0
pycairo                            1.20.1
pycryptodomex                      3.11.0
pycups                             2.0.1
pygame                             2.1.2
Pygments                           2.14.0
PyGObject                          3.42.2
pyinotify                          0.9.6
PyJWT                              2.6.0
pylint                             2.16.2
PyOpenGL                           3.1.6
[... 158 lines cut ...]
types-urllib3                      1.26
types-vobject                      0.9
types-waitress                     2.1
types-whatthepatch                 1.0
types-xmltodict                    0.13
types-xxhash                       3.0
types-zxcvbn                       4.4
typing_extensions                  4.4.0
urllib3                            1.26.12
v4l2-python3                       0.3.3
virtualenv                         20.17.1+ds
webencodings                       0.5.1
Werkzeug                           2.2.2
wheel                              0.38.4
wrapt                              1.14.1
zipp                               1.0.0
moshtohor@raspberrypi:~ $ python3
Python 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RF24
>>> from RF24 import RF24
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'RF24' from 'RF24' (unknown location)
>>>

2bndy5, 2024-06-16:
I don't see RF24 in the `pip list` output. This means that it is not installed. There must be some build error that you are not aware of.

This spoken language barrier between us makes helping you too difficult for me.

I only have 1 last recommendation: Switch to using [our newer pyrf24 package](https://pypi.org/project/pyrf24). You only need to [change the import statements](https://nrf24.github.io/pyRF24/#migrating-to-pyrf24).
```
python3 -m pip install pyrf24
python3
>>> from pyrf24 import RF24
>>> radio = RF24(22, 0)
>>> radio.begin()
True
```
```

2bndy5, 2024-06-16:
I'm sensing that you don't know how to use a python virtual environment. It is not my job to teach people how to use python. You'll need to learn this from somewhere else.

FalconCode10, 2024-06-16:
thanks bro, you are very helpful

2bndy5, 2024-06-16:
Now that I have nothing left to recommend or suggest, I'm going to close this. You can still post comments here if you want.

closed 2024-06-16 by 2bndy5