nRF24/RF24 #980: Change `SERIAL_DEBUG` into `RF24_DEBUG`

2bndy5, 2024-06-15:
Basically, we need a way to differentiate between enabling RF24 debug output and enabling RF24Network output when compiling the RF24 stack into a single binary.

**Describe the solution you'd like**
Just as the title says: Change `SERIAL_DEBUG` into `RF24_DEBUG`. We should also update the macro name from `IF_SERIAL_DEBUG()` into `IF_RF24_DEBUG()`.

**Describe alternatives you've considered**
This problem is not affected in typical usage where the RF24 stack is compiled into separate binary drivers.

**Additional context**
See https://github.com/nRF24/pyRF24/pull/64#issuecomment-2169452047. While this problem was identified in the pyRF24 project, I think it would also affect any project that involves RF24 and RF24Network libs using debug output for the Arduino and PicoSDK platforms.

2bndy5, 2024-06-15:
In researching why some portability uses https://github.com/nRF24/RF24/blob/29231f22d5739b8fb4d3a4d43c9a8ae07a7f1ce4/utility/ATXMegaD3/RF24_arch_config.h#L33
I found [a comment way back in 2016](https://github.com/nRF24/RF24/pull/238#issuecomment-211220781) that might be trying to reference this situation. 😄

TMRh20, 2024-06-15:
Hehe, I think the RF24Network define should probably be changed to `RF24NETWORK_DEBUG` too.

2bndy5, 2024-06-15:
Yeah, we can do that too. It is just a simple search and replace solution.

2bndy5, 2024-06-15:
BTW, the debug toggle in RF24Mesh is `MESH_DEBUG`. Do you want to prefix that to be `RF24MESH_DEBUG`?

TMRh20, 2024-06-15:
Sure - Nice and consistent and very unlikely to be repeated in other code.

2bndy5, 2024-06-15:
RF24Ethernet seems to be already using some lib-specific debug toggle names.

RF24Gateway seems to use a `DEBUG_LEVEL` in which the integer value dictates the verbosity of debug output. I'm guessing `DEBUG_LEVEL` should probably get renamed to `RF24GATEWAY_DEBUG_LEVEL` as well?

TMRh20, 2024-06-15:
Yeah lol

PR #981 (2024-06-15): Change SERIAL_DEBUG into RF24_DEBUG
resolves #980
commit 0aac3d58 (2024-06-15): Change SERIAL_DEBUG into RF24_DEBUG (#981)

resolves #980
closed 2024-06-15 by 2bndy5