nRF24/RF24 #989: NRF24s not Working with Arduino UNO or Pro Mini but do on ESP8266

Rodazak, 2024-07-20:
### What radio module do you use?

nRF20L01+

### What driver board(s) do you use?

Arduino UNO, Arduino Pro Mini, ESP8266

### If using Linux, what RF24 driver did you select?

None

### Describe your problem

Hello,
Im using the Getting Started Script,

08:58:32.412 -> SPI Speedz	= 10 Mhz
08:58:32.412 -> STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
08:58:32.412 -> RX_ADDR_P0-1	= 0x65646f4e31 0x65646f4e32
08:58:32.412 -> RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
08:58:32.412 -> TX_ADDR		= 0x65646f4e31
08:58:32.412 -> RX_PW_P0-6	= 0x04 0x04 0x04 0x04 0x04 0x04
08:58:32.412 -> EN_AA		= 0x3f
08:58:32.412 -> EN_RXADDR	= 0x02
08:58:32.412 -> RF_CH		= 0x4c
08:58:32.412 -> RF_SETUP	= 0x03
08:58:32.412 -> CONFIG		= 0x0f
08:58:32.412 -> DYNPD/FEATURE	= 0x00 0x00
08:58:32.412 -> Data Rate	= 1 MBPS
08:58:32.444 -> Model		= nRF24L01+
08:58:32.444 -> CRC Length	= 16 bits
08:58:32.444 -> PA Power	= PA_LOW
08:58:32.444 -> ARC		= 0

All modules print the radio.printDetails(); like that wich i think is ok.
I have 3 NRF24 modules with antenas and 2 without, they all work while connected to te ESP8266s but i cant make any of them work with either my Arduino UNO or any of my 4 Arduino Pro Mini, all the modules are probably clones do to its price cuz im from Argentina. 
The modules work perfect con the ESPs without extra PSU or capacitor or anything i just connect them to 3.3 of the board and they work like a charm but i have tried Externals PSUs and Capacitor on each of the othersd with the arduinos and i couldnt make any single one of them work.

08:58:32.444 -> SPI Frequency		= 10 Mhz
08:58:32.444 -> Channel			= 76 (~ 2476 MHz)
08:58:32.444 -> Model			= nRF24L01+
08:58:32.444 -> RF Data Rate		= 1 MBPS
08:58:32.444 -> RF Power Amplifier	= PA_LOW
08:58:32.444 -> RF Low Noise Amplifier	= Enabled
08:58:32.444 -> CRC Length		= 16 bits
08:58:32.444 -> Address Length		= 5 bytes
08:58:32.444 -> Static Payload Length	= 4 bytes
08:58:32.444 -> Auto Retry Delay	= 1500 microseconds
08:58:32.476 -> Auto Retry Attempts	= 15 maximum
08:58:32.476 -> Packets lost on
08:58:32.476 ->     current channel	= 0
08:58:32.476 -> Retry attempts made for
08:58:32.476 ->     last transmission	= 0
08:58:32.476 -> Multicast		= Disabled
08:58:32.476 -> Custom ACK Payload	= Disabled
08:58:32.476 -> Dynamic Payloads	= Disabled
08:58:32.476 -> Auto Acknowledgment	= Enabled
08:58:32.476 -> Primary Mode		= RX
08:58:32.476 -> TX address		= 0x65646f4e31
08:58:32.476 -> pipe 0 (closed) bound	= 0x65646f4e31
08:58:32.476 -> pipe 1 ( open ) bound	= 0x65646f4e32
08:58:32.518 -> pipe 2 (closed) bound	= 0xc3
08:58:32.518 -> pipe 3 (closed) bound	= 0xc4
08:58:32.518 -> pipe 4 (closed) bound	= 0xc5
08:58:32.518 -> pipe 5 (closed) bound	= 0xc6

This is the prety details just in case.
Ive tried every possible  Data Rate Speed also Min to Max Power and it made no diference.
I configured the ESP8266 with the exact same code on the Arduino IDE . 

Im using Arduino IDE 2.3.2 and the RF24 Library 1.4.9, i already reinstalled the library 4 times including downloading the ZIP from GitHub and intalling it that way.

Im completly out of ideas.

Thanks for anyone that ocn offer advice.

### What is the RX code?

```cpp
/*
 * See documentation at https://nRF24.github.io/RF24
 * See License information at root directory of this library
 * Author: Brendan Doherty (2bndy5)
 */

/**
 * A simple example of sending data from 1 nRF24L01 transceiver to another.
 *
 * This example was written to be used on 2 devices acting as "nodes".
 * Use the Serial Monitor to change each node's behavior.
 */
#include <SPI.h>
#include "printf.h"
#include "RF24.h"

#define CE_PIN 9
#define CSN_PIN 10
// instantiate an object for the nRF24L01 transceiver
RF24 radio(CE_PIN, CSN_PIN);

// Let these addresses be used for the pair
uint8_t address[][6] = { "1Node", "2Node" };
// It is very helpful to think of an address as a path instead of as
// an identifying device destination

// to use different addresses on a pair of radios, we need a variable to
// uniquely identify which address this radio will use to transmit
bool radioNumber = 1;  // 0 uses address[0] to transmit, 1 uses address[1] to transmit

// Used to control whether this node is sending or receiving
bool role = false;  // true = TX role, false = RX role

// For this example, we'll be using a payload containing
// a single float number that will be incremented
// on every successful transmission
float payload = 0.0;

void setup() {

  Serial.begin(115200);
  while (!Serial) {
    // some boards need to wait to ensure access to serial over USB
  }

  // initialize the transceiver on the SPI bus
  if (!radio.begin()) {
    Serial.println(F("radio hardware is not responding!!"));
    while (1) {}  // hold in infinite loop
  }

  // print example's introductory prompt
  Serial.println(F("RF24/examples/GettingStarted"));

  // To set the radioNumber via the Serial monitor on startup
  Serial.println(F("Which radio is this? Enter '0' or '1'. Defaults to '0'"));
  while (!Serial.available()) {
    // wait for user input
  }
  char input = Serial.parseInt();
  radioNumber = input == 1;
  Serial.print(F("radioNumber = "));
  Serial.println((int)radioNumber);

  // role variable is hardcoded to RX behavior, inform the user of this
  Serial.println(F("*** PRESS 'T' to begin transmitting to the other node"));

  // Set the PA Level low to try preventing power supply related problems
  // because these examples are likely run with nodes in close proximity to
  // each other.
  //radio.setPALevel(RF24_PA_MAX);  // RF24_PA_MAX is default.
  //radio.setAutoAck(false);
  //radio.setDataRate (RF24_250KBPS);
  radio.setPALevel(0);

  // save on transmission time by setting the radio to only transmit the
  // number of bytes we need to transmit a float
   radio.setPayloadSize(sizeof(payload));  // float datatype occupies 4 bytes

  // set the TX address of the RX node into the TX pipe
  radio.openWritingPipe(address[radioNumber]);  // always uses pipe 0

  // set the RX address of the TX node into a RX pipe
  radio.openReadingPipe(1, address[!radioNumber]);  // using pipe 1

  // additional setup specific to the node's role
  if (role) {
    radio.stopListening();  // put radio in TX mode
  } else {
    radio.startListening();  // put radio in RX mode
  }

  // For debugging info
   printf_begin();             // needed only once for printing details
   radio.printDetails();       // (smaller) function that prints raw register values
   radio.printPrettyDetails(); // (larger) function that prints human readable data

}  // setup

void loop() {

  if (role) {
    // This device is a TX node

    unsigned long start_timer = micros();                // start the timer
    bool report = radio.write(&payload, sizeof(float));  // transmit & save the report
    unsigned long end_timer = micros();                  // end the timer

    if (report) {
      Serial.print(F("Transmission successful! "));  // payload was delivered
      Serial.print(F("Time to transmit = "));
      Serial.print(end_timer - start_timer);  // print the timer result
      Serial.print(F(" us. Sent: "));
      Serial.println(payload);  // print payload sent
      payload += 0.01;          // increment float payload
    } else {
      Serial.println(F("Transmission failed or timed out"));  // payload was not delivered
    }

    // to make this example readable in the serial monitor
    delay(1000);  // slow transmissions down by 1 second

  } else {
    // This device is a RX node

    uint8_t pipe;
    if (radio.available(&pipe)) {              // is there a payload? get the pipe number that recieved it
      uint8_t bytes = radio.getPayloadSize();  // get the size of the payload
      radio.read(&payload, bytes);             // fetch payload from FIFO
      Serial.print(F("Received "));
      Serial.print(bytes);  // print the size of the payload
      Serial.print(F(" bytes on pipe "));
      Serial.print(pipe);  // print the pipe number
      Serial.print(F(": "));
      Serial.println(payload);  // print the payload's value
    }
  }  // role

  if (Serial.available()) {
    // change the role via the serial monitor

    char c = toupper(Serial.read());
    if (c == 'T' && !role) {
      // Become the TX node

      role = true;
      Serial.println(F("*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK"));
      radio.stopListening();

    } else if (c == 'R' && role) {
      // Become the RX node

      role = false;
      Serial.println(F("*** CHANGING TO RECEIVE ROLE -- PRESS 'T' TO SWITCH BACK"));
      radio.startListening();
    }
  }

}  // loop
```

### What is the TX code?

```cpp
// The code for RX is same for TX.
// See above code for RX
```

2bndy5, 2024-07-20:
> all the modules are probably clones do to its price cuz im from Argentina

Its probably not anything to do with the software. You have pretty much eliminated that probability by using our basic example.

1. You may have to play with higher capacitance and/or mix capacitors (low capacitance in parallel with high capacitance).
2. I doubt lowering the SPI speed will help since the `printDetails()` output looks fine.
3. Radios with a detachable antenna (usually labeled "PA/LNA") tend to test better when the PA level is set to `RF24_PA_LOW` or `RF24_PA_MIN` because it takes less power to transmit (including the auto-ack packets sent during RX role).

When you say, "they don't work", I assume you are seeing "Transmission failed or timed out" in the Serial output during TX role. This usually indicates a power problem in which the auto-ack packets aren't getting sent/received.

> The modules work perfect con the ESPs without extra PSU or capacitor or anything i just connect them to 3.3 of the board and they work like a charm

Since the ESP chips have a radio builtin, they would also have builtin power stabilizing capacitors/regulator. Its unlikely that the Arduino boards (without a builtin radio) have these extra capacitors. We typically recommend 100uF capacitance, but this can vary based on the power supply and nRF24L module.

2bndy5, 2024-07-20:
"Power" does not mean "only Voltage". Remember, $`P = I * V`$. The current ($`I`$) must remain consistent as well.

Rodazak, 2024-07-20:
> "Power" does not mean "only Voltage". Remember, P=I∗V. The current (I) must remain consistent as well.

You are totally right, im actually not good at electronics i just started as a hobby like 2 or 3 months ago, i actually have no idea on how to stabilize current idk if a capacitor can actually stabilize both.
I honestly have started hating my arduinos more and more each day im in this XD the esps cost about the same for me and give me so many less issues, the only reason im using arduinos ir cuz im trying to build an universal rc controller and and i need more analog imputs than just he 1 the esp boards have

2bndy5, 2024-07-20:
Capacitors' behavior and application are not a beginner-friendly discussion. Just think of capacitors like a momentary battery that only fails when power drops for longer than a moment (an extended period of time).

If you want a cheap board without a builtin radio, I'd recommend something with the ATSAMD21 or the RP2040 (a newer dual-core offshoot of the ATSAMD21). Analog inputs are only a problem when using a Raspberry Pi computer (which didn't have any until the recent RPi5).

> [!note]
> The ESP8266 is no longer recommended for new designs. But then again so is the nRF24L01. 😆
> 
> ESP32 boards can be surprisingly cheap also. However, they tend to consume a lot of power with the WiFi enabled.

TMRh20, 2024-07-20:
At this point its either the CE pin is wired wrong or power supply issues I would say. Also when connecting an external power source to the radio, the GND needs to be connected to the radio AND Microcontroller GND. It should be really straight forward at this point so its gotta be either a: wiring b: user error c: power supply

Rodazak, 2024-07-20:
So im kinda ashamed but im gonna kinda explain what happened, ie been fighting with tihis for about 4 days, ive started from code from a proyect for the rc on youtube and couldnt make it work, from that point i tryid many many alternatives to the code and couldnt make any of them work,,,,,,BUT,,,, this work actually works fine, my brain was not acting brainy and i dindt read carefully enought so when the soft ask for the radio either 1 or 0 y thought it ment adress so i was setting them bothe to the same everytime. So nothing... Im really Sorry, Still the code i had probably doesnt work or any of the thousand things i tried actually fixed my inital problems and i was bugging myselt into issues at this point, so yea the getting started code actually works on the arduinos.
Regardless im sorry to have wated your brainpower and time. I actully learned a lot about electronics on the process of all this so for me it was a win i guess.
Thanks for the rapid response btw.

2bndy5, 2024-07-20:
> so yea the getting started code actually works on the arduinos

That's good to hear 🎉 

> i dindt read carefully enought so when the soft ask for the radio either 1 or 0

Its unfortunate that we can't use a translation proxy (at compile time) for the examples' English output...

closed 2024-07-20 by 2bndy5

2bndy5, 2024-07-20:
> code from a proyect for the rc on youtube 

Yeah, one of our biggest problems with troubleshooting is this library's popularity. There's a lot of third-party tutorials out there, and they almost always neglect the nuanced behaviors of this radio.

I would advise you pay attention to the examples' comments as well, especially for our "MulticeiverDemo" and "AcknowledgementPayloads" examples..