nRF24/RF24 #990: [Question]: PICO's not communicating

AndyVotava, 2024-07-30:
### What would you like to know?

Hi there,

I’ve built and uploaded the "Getting Started" example for the Pico, but I’m encountering transmission failures or timeouts. The Pico seems to communicate with the RF24 module correctly, as `begin` returns true. I have CE connected to GPIO 7 and CSN connected to GPIO 8 (as per `defaultPins.h`), and I’m getting the expected responses on my oscilloscope.

When running the program, I’ve set one Pico as node 0 and the other as node 1. I initiate transmission by sending ‘T’ to one of them.

I’ve tried different pin configurations, and I’m confident that the pins are connected correctly. I also added a 100nF capacitor between 3.3V and GND, but that didn’t resolve the issue.

Any suggestions or ideas would be greatly appreciated!

2bndy5, 2024-07-30:
What type of radio module are you using? Is it a PA/LNA type module?

Typically, if `radio.begin()` returns `true`, but you still see transmission timeouts, then it is likely a power ($`P = I * V`$) problem.

Last I checked, the RPi Pico board uses a switching regulator. It is good that you added a capacitor to filter out the noise on the 3v line because that is practically required when using a switching regulator. However, the PA/LNA modules require a significant amount of peak current to transmit (including auto-ACK packets in RX mode).

AndyVotava, 2024-07-30:
Hello,

I was just using the base nRF module does the pico 3v3 pin provide enough output current to power the normal ones?

AndyVotava, 2024-07-30:
I just tried powering the modules with my power supply and it worked! 

Thank you so much!

closed 2024-07-30 by AndyVotava