nRF24/RF24 #994: [Question]: Pico gettingstarted.cpp

ParrSt, 2024-08-25:
### What would you like to know?

I have used the NRF24 transceivers for a while on Arduino.  I am now trying to use on a Raspberry Pico .  Using the GettingStarted example from the examples I was able to successfully compile and install.  However I only get a VERY small number of packets received on the reverencing unit and NONE show as successful on the transmitting unit.  I have modified the code to increment payload by 1 for unsuccessful and left it to increment by .01 for successful transmission.  The below shows you about how frequently packets are received by the receiving unit.

`*** CHANGING TO RECEIVE ROLE -- PRESS 'T' TO SWITCH BACK
Received 4 bytes on pipe 1: 64.000000
Received 4 bytes on pipe 1: 100.000000
Received 4 bytes on pipe 1: 122.000000
Received 4 bytes on pipe 1: 153.000000
Received 4 bytes on pipe 1: 155.000000
Received 4 bytes on pipe 1: 219.000000
Received 4 bytes on pipe 1: 308.000000
Received 4 bytes on pipe 1: 368.000000
Received 4 bytes on pipe 1: 414.000000
Received 4 bytes on pipe 1: 426.000000
Received 4 bytes on pipe 1: 525.000000
Received 4 bytes on pipe 1: 568.000000
Received 4 bytes on pipe 1: 569.000000
Received 4 bytes on pipe 1: 607.000000
Received 4 bytes on pipe 1: 944.000000
Received 4 bytes on pipe 1: 1202.000000
`
I have had the transeivers as close as an inch or less and as far as 2-3 feet.  Both are plugged into the same computer but different ports to try to ensure good voltage and current needs.

Also I thought the addresses on two NRF chips had to be the same.  But if I am reading the code right, the first thing you do when the Pico comes up is select 0 or 1 which sets up the read/write pipes using node1 or node2 as the address depending on if that Pico was set at 0 or 1.

Bob

2bndy5, 2024-08-25:
> NONE show as successful on the transmitting unit

This is likely because the RPi Pico board uses a switching regulator to supply power ($`P = I * V`$) on the 3V line. By nature, switching regulators are *very* noisy (electronically speaking), thus they produce a rather unstable power source for sensitive peripherals (like nRF24L01 modules).

We keep getting users with the same power problem using the Pico board. The solution is to either use a external power source (if using a PA/LNA module) or add more capacitance in parallel to the radio's VCC/GND lines.

I think I need to add this to the COMMON_ISSUES.md doc This is the second time I've said this answer in the last few months.

2bndy5, 2024-08-25:
> Also I thought the addresses on two NRF chips had to be the same. But if I am reading the code right, the first thing you do when the Pico comes up is select 0 or 1 which sets up the read/write pipes using node1 or node2 as the address depending on if that Pico was set at 0 or 1.

For me, t helps to think of addresses as paths instead of destinations.

```mermaid
flowchart LR
    subgraph "unit 0"
    mcu0tx[TX pipe]
    mcu0rx[RX pipe]
    end
    subgraph "unit 1"
    mcu1tx[TX pipe]
    mcu1rx[RX pipe]
    end
    mcu0tx -- "1Node" --> mcu1rx;
    mcu1tx -- "2Node" --> mcu0rx;
```

ParrSt, 2024-08-25:
Okay, I will put a 10uF electrolytic on the NFR. I do remember being given that advise a long time ago but on the Arduino I found I really didn't need it and forgot.

As for the pipe address.  I think I was confusing that with the channel.  I did check some other documentation they they seem to use the same pipe address for both the sending pipe and the receiving pipe.  So if I understand what you are saying, the two pipes communicating don't need to be the same address, but the sender will need to know the receiver's pipe so it can send to the right place.   

Bob

2bndy5, 2024-08-25:
If you want to use a RX pipe on one unit to get data from the TX pipe on a different unit, then both of those pipes need to use the same address. It gets a bit more complicated when talking about auto-ack behavior, but the library takes care of that for you.

ParrSt, 2024-08-25:
Okay, added a 4.7uF to one and a 10uF to the other.  the receiving unit is not getting 30 - 80% of packets.  Much better than 1% or less.  I am still only getting a very rare ACK from the transmitter.  This is why I bumped the cap to a 10uF.  This may not be good enough.  Your other suggestion was to use a cleaner power for the nrf24.  So the Pico is powered by USB, but the nrf24 by separate power?  Would the grounds of the two need to be connected?

Also are there any other FR transceivers that might work better?

Bob

ParrSt, 2024-08-25:
`If you want to use a RX pipe on one unit to get data from the TX pipe on a different unit, then both of those pipes need to use the same address (a wireless "path"). It gets a bit more complicated when talking about auto-ack behavior, but the library takes care of that for you.`

I totally missed that the openreadpipe uses a ! on the address array.  So yes both radios are using the same pipe address.  Sometime I just need to be hit with the info like a 2x4.  lol

Bob

2bndy5, 2024-08-25:
>  Would the grounds of the two need to be connected?

Yes. This is a basic DC electronics answer. GND is a reference to 0V, so different circuits with different 0V references yield inconsistent behavior.

> Also are there any other FR transceivers that might work better?

This seems like a question better suited for [Arduino Forums](https://forum.arduino.cc/). Here, we only focus on nRF24L01 radios.

ParrSt, 2024-08-25:
Unless there is anything else you can suggest, then I guess we can close this out.

Thank you for your help,
Bob

closed 2024-08-25 by 2bndy5