nRF24/RF24 #997: Get Signal Strength

justcallmekoko, 2024-09-05:
### Describe the behavior you would like

I would like to have a public method available which would allow the user to get the signal strength from the most recently received data. Perhaps whatever is received during a period of listening. In addition to having methods like testCarrier or testRPD, there could be a separate method to call to retrieve the signal strength of the transmission which would be related to testCarrier etc. i.e. if testCarrier returns true, the user can call `carrierDPM` or something and it returns the signal strength value.

### Describe alternatives you have considered

I do not personally understand the registers of the nRF24L01+ enough to attempt an alternative solution but since there exists a testRPD method which has a dpm threshold, I imagine there is already a way signal strength is being measured.

### Additional context

I am creating a frequency analyzer which will display a graph representing the number of transmissions received within a certain amount of time before plotting the next point on the graph. The graph is to be color coded based on the average strength of the signal received during that time.

2bndy5, 2024-09-05:
This isn't possible with the nRF24L01. It is possible with the nRF5x radios, but the SPI interface for the nRF24L01 only allows reading the RPD flag. The signal strength is mostly implied from facts stated in the nRF24L01 datasheet.
- `available()` -> `true` means signal (for latest payload received) was at least 
   - -82dBm sensitivity at 2Mbps
   - -85dBm sensitivity at 1Mbps
   - -94dBm sensitivity at 250kbps
- `testRPD()` -> `true` means signal in the channel was present and detected with at least -64dBm or better

See our [scanner example](https://github.com/nRF24/RF24/blob/master/examples/scanner/scanner.ino) for the best-guess attempt. We also have [scannerGraphic][graph-scan] example (requires a display) that tracks the the history on every channel and uses that to calculate peak decays. The [scannerGraphic][graph-scan] could be better, but we're ultimately limited by the nRF24L01 interface.

[graph-scan]: https://github.com/nRF24/RF24/blob/master/examples/scannerGraphic/scannerGraphic.ino

2bndy5, 2024-09-05:
And certain PA/LNA variants obviously have different sensitivities as well. There's just no way we can read the signal strength given the SPI interface for the nRF24L01 chip.

justcallmekoko, 2024-09-07:
Thank you for the information. I was not aware of the limitations of the IC itself.

closed 2024-09-07 by justcallmekoko