nRF24/RF24 #1002: When using payload ack, the ack replied by the receiving end lags 4 times

camerakang, 2024-09-19:
### What radio module do you use?

nRF24L01+

### What driver board(s) do you use?

esp32-s3-devkitc-1

### If using Linux, what RF24 driver did you select?

None

### Describe your problem

I want the payload ack on the receiving end to reply to the last received data at any time other than the first communication, but now I'm finding that I can't receive the ack four frames ago until after my sender has sent it multiple times

### What is the RX code?

```cpp
https://github.com/camerakang/Nrf24_demo_send
```

### What is the TX code?

```cpp
https://github.com/camerakang/Nrf24_demo_rev
```

camerakang, 2024-09-19:
[image]

2bndy5, 2024-09-19:
`RF24::stopListening()` will flush the TX FIFO when ack payloads are enabled.

2bndy5, 2024-09-19:
If you're sending ACK payloads to multiple devices/pipes, then (I think) the RX device will only transmit the ACK payload in the TX FIFO's top level (first ACK payload queued) when the payload is received on the pipe intended for the ACK payload.

2bndy5, 2024-09-19:
Its kind of hard to follow everything your code is doing.

2bndy5, 2024-09-19:
> [!tip]
> `RF24::writeAckPayload()` returns a bool to tell you if the payload was uploaded into the TX FIFO.

2bndy5, 2024-09-19:
Have you run our AcknowledgementPayload example? It should demonstrate expected behavior to verify your radio is working correctly.

2bndy5, 2024-09-19:
```c++
        if (ackPayload != NULL && ackSize > 0)
``` 
Are you sure `ackPayload` is never null? This seems pretty clear that the ACK payload will not be uploaded to TX FIFO if this condition is not satisfied.

2bndy5, 2024-09-19:
Honestly, this seems like question for [Arduino Forums](https://forum.arduino.cc/) (or [PlatformIO forums](https://community.platformio.org/)). I don't think this is a library bug.

camerakang, 2024-09-19:
> ```c++
>         if (ackPayload != NULL && ackSize > 0)
>         {
>             // ackPayload->counter = result.data[2];
>             Serial.print(ackPayload->message);
>             Serial.println(ackPayload->counter);
> 
>             radio.writeAckPayload(result.pipe, ackPayload, ackSize);
>         }
> ```
> 
> 你确定`ackPayload`永远不会为空吗？这似乎很清楚，如果不满足此条件，ACK 有效负载将不会上传到 TX FIFO。

Thank you very much for your reply, I re-read the example about paylod ack and the problem is solved. But I have a problem with setting the channel, I'm modifying the channel using the " radio.setChannel()" function and after the modification I'm printing the current channel via radio.getChannel(), I'm sure that the sender and the receiver are modifying to the same channel but I can't send successfully after the modification. Looking forward to your reply

2bndy5, 2024-09-19:
Its best to call `setChannel()` while not listening. If the antenna is being used, the radio has trouble tuning it to the specified frequency.
```cpp
radio.stopListening(); // this will flush TX FIFO if ACK payloads enabled
// radio is now in an inactive TX mode.
radio.setChannel(87);
radio.startListening(); // resume RX behavior
```

closed 2025-01-29 by 2bndy5