nRF24/RF24 #1004: [Question]: ESP32 (using esp-idf-mirf) can not communicate with Arduino (using RF24)

ntdkhoa0410, 2024-10-06:
### What would you like to know?

Hello, as the title said, ESP32 as a transmitter using library  [nopnop2002/esp-idf-mirf](https://github.com/nopnop2002/esp-idf-mirf ) can not communicate with Arduino Mega using   [nRF24/RF24](https://github.com/nRF24/RF24) 
I have tried to print all the details to the monitor and everything seems to match, but on the receiver side, no message is receive.
Can you guys have a look at my code, suppose the hardware set up is correct as if both use RF24 libs, they work fine.

## Arduino receiver code (RF24 lib)
```
// Include neccessary library
#include <string.h>
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

//Define pins
#define CE_PIN   31
#define CSN_PIN 53
#define LED_PIN 13

// Variable for delay functions
unsigned long previousMillis = 0; // Stores the last time the LED was updated
const long interval = 1000; // Interval for the delay (in milliseconds)

// Radio variables
char text[10] = "s";
char text_old[10] ;
const byte address[6] = "00002";
uint8_t payLoad = 16;

// State to be changed when receive messaged
bool led_state = 0;

// Create a Radio
RF24 radio(CE_PIN, CSN_PIN);

// Prototype
void radio_ini();
void print_debug();
void radio_handling();

void setup() {
  Serial.begin(115200);
  strcpy(text_old,text);
  pinMode(LED_PIN, OUTPUT);
  radio_ini();
}
void loop() {
  if (radio.available()) {
    radio_handling();
  }
  print_debug();
}

void radio_ini(){
  radio.begin();
  radio.openReadingPipe(0, address);
  radio.setPALevel(RF24_PA_MAX);
  radio.setPayloadSize(payLoad);
  radio.setChannel(120);
  radio.startListening();
}

void radio_handling(){
  radio.read(&text, sizeof(text)); // Read incoming data
  //Serial.println(text);
  // Compare original with temp, if message received -> LED changes states
  if (strcmp(text, text_old) != 0) {
    led_state = !led_state;
    digitalWrite(LED_PIN,led_state);
    strcpy(text_old, text);
  }
}

void print_debug(){
  unsigned long currentMillis = millis(); // Get the current time
  if (currentMillis - previousMillis >= interval) {
    uint8_t chn = radio.getChannel();
    previousMillis = currentMillis;

    Serial.print("Current channel: ");
    Serial.print(chn);
    Serial.print("\n");
    bool connected = radio.isChipConnected();
    if (connected) {
    Serial.println("Connected");
    }
    else {
    Serial.println("Not connected");
    }
    char buffer[870] = {'\0'};
    uint16_t used_chars = radio.sprintfPrettyDetails(buffer);
    Serial.println(buffer);
  }
}
```
## ESP32 transmitter code (ESP-IDF-MIRF lib)
# main.c
```
#include <stdio.h>
#include "defines.h"
#include "components_list.h"
#include "esp_timer.h"

// Variable for delay functions
static unsigned long previousMillis = 0; // Stores the last time the LED was updated
static const long interval = 1000;       // Interval for the delay (in milliseconds)

void ini(void)
{
    led_init();

    // Init for mirf
    uint8_t payload = 16;
    uint8_t channel = 120;
    char *address = "20000";
nrf24l01_init(payload, channel, address);
}

void app_main(void)
{
    ini();
    while (1)
    {
        unsigned long currentMillis = esp_timer_get_time() / 1000;
        if (currentMillis - previousMillis >= interval) // Update and send message every 1 second
        {
            previousMillis = currentMillis;
            sender();
        }
    }
}
```
**nrf24l01_init function:**
```
void nrf24l01_init(uint8_t payload, uint8_t channel, char *address)
{
    ESP_LOGI(pcTaskGetName(NULL), "Start");
    // NRF24_t dev;
    Nrf24_init(&dev);

    Nrf24_config(&dev, channel, payload);

    // Set destination address using 5 characters
    esp_err_t ret = Nrf24_setTADDR(&dev, (uint8_t *)address);
    if (ret != ESP_OK)
    {
        ESP_LOGE(pcTaskGetName(NULL), "nrf24l01 not installed");
        while (1)
        {
            vTaskDelay(1);
        }
    }

    AdvancedSettings(&dev);
    // Print settings
    Nrf24_printDetails(&dev);
}
```
**AdvancedSettings function:** 
```
void AdvancedSettings(NRF24_t *dev)
{

    ESP_LOGW(pcTaskGetName(NULL), "Set RF Data Ratio to 1MBps");
    Nrf24_SetSpeedDataRates(dev, 0);

    ESP_LOGW(pcTaskGetName(NULL), "CONFIG_RETRANSMIT_DELAY=%d", CONFIG_RETRANSMIT_DELAY);
    Nrf24_setRetransmitDelay(dev, CONFIG_RETRANSMIT_DELAY);
}
```
Sender function:
```
void sender(void)
{
    uint8_t buf[16];
    uint8_t var_Num = updateMessage();
    sprintf((char *)buf, "Message %" PRIu8, var_Num);
    Nrf24_send(&dev, buf);
    // vTaskDelay(1);
    ESP_LOGI(pcTaskGetName(NULL), "The data to send: %s", buf);
    ESP_LOGI(pcTaskGetName(NULL), "Wait for sending.....");
    if (Nrf24_isSend(&dev, 1000))
    {
        ESP_LOGI(pcTaskGetName(NULL), "Send success:%s", buf);
    }
    else
    {
        ESP_LOGW(pcTaskGetName(NULL), "Send fail:");
    }
    blink_led();
}
```
##The monitor of Arduino:
[image]

##The monitor of ESP32:
[image]

TMRh20, 2024-10-06:
First things, you may want to use the older `printDetails()` function on the Mega, as it is more similar to the output you are getting with the other library.

Then you want to look directly at the CONFIG register etc and make sure they match.

I'm noticing the ESP is using 8-bit CRC right now (CONFIG ix 0x2B or 00101011) while the Mega is using 16-bit CRC.

Your best bet is to compare registers and look at the datasheet: https://github.com/nRF24/RF24/blob/master/datasheets/nRF24L01_datasheet_v2.pdf

Feel free to post the output of `printDetails();` here as well if you still can't get it working.

2bndy5, 2024-10-06:
This seems like a question for the Arduino forums or platformio forums.

At first glance, I think your address is wrong.

In ESP32 transmitter code (ESP-IDF-MIRF lib)
```c
char *address = "20000";
```
In Arduino receiver code (RF24 lib)
```c
const byte address[6] = "00002";
```

If you are looking for assistance in writing C++, then this is not the place for that.

If you are looking for assistance with the ESP_IDF_MIRF library, then this is not the place that either. We didn't author that lib. BTW, we are working on ESP-IDF support, but it isn't ready to publish yet.

2bndy5, 2024-10-06:
It seems the author of the [ESP-IDF-MIRF lib](https://github.com/nopnop2002/esp-idf-mirf) also wrote an [Arduino lib](https://github.com/nopnop2002/Arduino-STM32-nRF24L01) which claims to have been tested on Arduino UNO, MEGA, and DUE (along with various STM32 boards).

ntdkhoa0410, 2024-10-06:
> You probably want to use `"20000"` in RF24 lib side. The `byte` type is just an alias to `unsigned char`, and a `[]` operator is similar to `*` in variable declaration.

I understand your point, I have first used the same address for both side. First i used "00002" in the esp-idf-mirf side, however, when the address is printed out, it is 0x30303030**32** - which is in reverse with the arduino side (which is 0x**32**30303030). Hence the difference you saw
>If you are looking for assistance with the ESP_IDF_MIRF library, then this is not the place that either. We didn't author that lib. BTW, we are working on ESP-IDF support, but it isn't ready to publish yet.
I understand this point also, i just want to seek for all possible help.

2bndy5, 2024-10-06:
It is reversed in `printDetails()` because that's how it is stored in memory ([Endianness](https://en.wikipedia.org/wiki/Endianness)).

ntdkhoa0410, 2024-10-06:
> First things, you may want to use the older `printDetails()` function on the Mega, as it is more similar to the output you are getting with the other library.

> Feel free to post the output of `printDetails();` here as well if you still can't get it working.
Thank you, I will try this tmr

2bndy5, 2024-10-06:
> ```c
> radio.openReadingPipe(0, address);
> ```

Also, it is highly recommended to use pipe 1 for receiving because pipe 0 is used for transmitting (including auto-ack packets during RX). This is shown in all of our examples.

ntdkhoa0410, 2024-10-06:
> It is reversed in `printDetails()` because that's how it is stored in memory ([Endianness](https://en.wikipedia.org/wiki/Endianness)).

Thank you, maybe the memory storage is different in 2 libs and the same on the nrf24l01 module

2bndy5, 2024-10-06:
That is unlikely if you are using C strings. That would be more likely if you are using a data structure with different data types.

2bndy5, 2024-10-06:
The address in ESP-IDF-MIRF lib is not properly reversed in `Nrf24_printDetails()` (which uses [`Nrf24_print_address_register()`](https://github.com/nopnop2002/esp-idf-mirf/blob/3276ffd4cff627a933289363362972f0433e4b0b/components/mirf/mirf.c#L505) under the hood).

ntdkhoa0410, 2024-10-07:
@2bndy5  oh i see, so the address stored in nrf24l01 module is the same, but the way ESP-IDF-MIRF lib prints is in reversed order with Arduino RF24 lib? in the code both address should be "00002"?

I will follow you suggestions when I get back home from work

2bndy5, 2024-10-07:
Use `"20000"`. The unique byte of the address should be the first character.

stefan123t, 2024-10-07:
@ntdkhoa0410 please consider to try it first with PA_LOW or PA_MIN, the NRF24L01+ modules with a power amplifier (usually those sporting an external Bender-like antenna) are prone to just receive too little power for transmitting **and** the Power amplifier stages on the board. A 100uF capacitor between VCC and GND may help at times, but it is probably better to start with LOW / MIN settings for a bench setup where the sender and receiver are next to each other.

Also did you get it to work with two times the same code / library on each side, i.e. 2x ESP32 TX & RX side and 2x Arduino Mega on TX & RX side each ?

2bndy5, 2024-10-07:
> is it possible to communicate esp32 and arduino with two different libs though

It should be. But adequate troubleshooting will require some familiarity with each lib's code and the radio's [datasheet]. We have a [COMMON_ISSUES] doc that tries to detail the most common problems without looking at the code or [datasheet].

> but on the receiver side, no message is receive.

This does not give us much to go on. What is your `sender()`'s log output? Most of our suggestions so far are already documented in the [COMMON_ISSUES] doc.

[COMMON_ISSUES]: https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md
[datasheet]: https://github.com/nRF24/RF24/blob/master/datasheets/nRF24L01P_datasheet_v1.pdf

stefan123t, 2024-10-07:
> @stefan123t I managed to use RF24 lib on ESP32 and Arduino and they worked with PA_MAX. The capacitor was added in the first place.

So you are good to go. Thanks for the confirmation. 

> Still figuring out is it possible to communicate esp32 and arduino with two different libs though

You are trying ESP-IDF-MIRF -> RF24 lib now.
What about the converse RF24 -> ESP-IDF-MIRF ? Does that work ?

And yes, it should be possible as the libraries both use the NRF24's own ESB Enhanced Shockburst protocol for transmission.

Though I am still struggling a bit to find all the details / settings that you see on the RF24 library side in `printPrettyDetails()`/`printDetails()`.

We had a couple of users that analysed the data going over the TX/RX lines of the SPI bus for analysing a third party binary protocol, maybe this could be useful to see if the data gets sent or the IRQ flag gets raised on the receiving end.
Other users used an HackRF to catch the signal on the air, but I do not have such a board myself.

2bndy5, 2024-10-07:
> ```c
>     Nrf24_setRetransmitDelay(dev, CONFIG_RETRANSMIT_DELAY);
> ```

@ntdkhoa0410 You probably should not change this `0` unless you know what it actually does. It is probably equivalent to our [`setRetries()`](https://nrf24.github.io/RF24/classRF24.html#a4c6d3959c8320e64568395f4ef507aef)'s delay parameter (which defaults to `5`) I think the ESP-IDF-MIRF default (`0`) is too low for general stability.

That lib also seems to prefer idling in RX 👎🏼. This makes that lib rather power hungry, which is a bit surprising for a lib that is dedicated to the ESP32.

2bndy5, 2024-10-07:
The knowledge you seek is in the datasheet (Appendix A). You are asking about how the auto-ack feature works.

ntdkhoa0410, 2024-10-09:
Hello @2bndy5 @stefan123t @TMRh20 ,
I managed to make it work with [Arduino-STM32-nRF24L01](https://github.com/nopnop2002/Arduino-STM32-nRF24L01)
Thanks for your support a lot!

closed 2024-10-09 by ntdkhoa0410

stefan123t, 2024-10-09:
@ntdkhoa0410 great and good to hear.

So you are now able to send from 
* from ESP TX -> Arduino RX or
* from Arduino TX -> ESP RX or 
* even in both directions ?

And which library are you using now on which side ?

Arduino: Arduino-STM32-nRF24L01
Arduino: RF24
ESP32: RF24 
ESP32: ESP-IDF-MIRF

Especially interesting is why it did not work in the first place and which combination:

You were trying ESP-IDF-MIRF TX -> RF24 RX lib.
What about the reverse RF24 TX -> ESP-IDF-MIRF RX, did that work ?

2bndy5, 2024-10-10:
See duplicate issue with similar responses: https://github.com/nopnop2002/esp-idf-mirf/issues/13. There, they noticed the CRC length was not the same.

ntdkhoa0410, 2024-10-11:
@stefan123t I am able to send from:
1. ESP TX (using ESP-IDF-MIRF) -> Arduino RX ([Arduino-STM32-nRF24L01](https://github.com/nopnop2002/Arduino-STM32-nRF24L01)) and
2. ESP TX (using RF24) -> Arduino Rx (using RF24)

Other cases, I haven't had time to test it yet, and was not planning to due to the differences in register values as explained in the duplicate issue