nRF24/RF24 #1006: ESP32 dev kit1 + TFTLCD + nRF24L01 setup working for years stopped working after late revisions

rodd01, 2024-10-21:
### What radio module do you use?

nRF24L01

### What driver board(s) do you use?

ESP32 dev kit rev1

### If using Linux, what RF24 driver did you select?

None

### Describe your problem

I have a setup where two Arduinos ProMicro (2 Tx's)  send sensor data to a receiver (one RX).
The RX uses an ESP32 dev ket rev1 board, nRF24L01 and a TFT display (ILI9341 driver) and TFT_eSPI library.
The code has been working without problems for more than 2 years.
Recently I decided to make some improvements on the code and I stated getting the error below:
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32_DEV
  Arduino Variant   : doitESP32devkitV1
  Arduino FQBN      : esp32:esp32:esp32doit-devkit-v1:FlashFreq=80,UploadSpeed=921600,DebugLevel=debug,EraseFlash=none
============ Before Setup End ============
[   800][E][esp32-hal-cpu.c:121] addApbChangeCallback(): duplicate func=0x400d9810 arg=0x3ffbdc40
[   809][D][esp32-hal-spi.c:185] spiDetachBus(): Stopping SPI bus
radio hardware is not responding!!

Then I tried the ORIGINAL, WORKING CODE and got the SAME ERROR.
Can you help me?

I won't post the codes at this moment since it seems they are not the cause of the problem, once I have working versions uploaded in the past.
Thank you

### What is the RX code?

```cpp
#include <RF24.h>
// ...
```

### What is the TX code?

```cpp
// The code for RX is same for TX.
// See above code for RX
```

2bndy5, 2024-10-21:
> Recently I decided to make some improvements on the code and I stated getting the error below

Please give us more details! Have you tried testing your setup (without the TFT display) using our gettingStarted example?

> ```
> [   800][E][esp32-hal-cpu.c:121] addApbChangeCallback(): duplicate func=0x400d9810 arg=0x3ffbdc40
> [   809][D][esp32-hal-spi.c:185] spiDetachBus(): Stopping SPI bus
> ```

This error (`[E]`) and debug ([`D`]) message looks like it isn't something caused by the RF24 library.

---
> stopped working after late revisions 

Please provide exact version numbers of the libraries used. It would also help if we know the previous (working) version numbers, so we can do a comparison of changes.

2bndy5, 2024-10-21:
> Then I tried the ORIGINAL, WORKING CODE and got the SAME ERROR.
Can you help me?

You should probably remove the bad build artifacts in the .pio/ folder first.

rodd01, 2024-10-22:
Thank you for your reply Brendan,
The versions in use today are as follows:
RF24: 1.4.10
TFT_eSPI: 2.5.34
esp32 board 3.0.5 (Expressif)
Arduino IDE 1.8.57.0
SPI  (?)
nRF24L01 (?)

Unfortunately I do not know how to retrieve the versions of the LAST
working libraries, but they were the latest available when the last update
to the code was uploaded, which was in January 2024.
Earlier versions of code using the same hardware started in June 2021 and
have been improved/updated at least twice a year.

Regards,
Roger

2bndy5, 2024-10-22:
I'm still not convinced this is a RF24 library problem. The "radio not responding" message appears because something else stopped the SPI bus.

1. Have you checked with whatever library you are using to drive the TFT displays? My guess is that the TFT display library is not sharing the SPI bus adequately.
2. Have you tried running our gettingStarted example to ensure the RF24 library works properly?

We are not liable for whatever users do with their code. We are only here to maintain the RF24 library.

rodd01, 2024-10-22:
I understand that there are 3 possible suspects; the board file, the tft drive and the rf24 drive.The display drive is working properly and also the compilation error/debug message disappeared when I changed the CSN pin connection to GPIO5, but the radio is still not available.Also, there are known issues of how RF24 library “define” the spi bus parameters even when the library is not in use.That is why I decided to ask you for help, since I have NO training on C++.Enviado do meu iPhone

2bndy5, 2024-10-22:
> Also, there are known issues of how RF24 library “define” the spi bus parameters even when the library is not in use

Huh? This is the first I'm hearing about it. Please explain.

rodd01, 2024-10-22:
dear Brendan,
I don't intend to be impolite, but I found some references to it while
googling...
SD card readers seem to have problems with the RF24 library as well as
ADXL345.
The explanation provided is that the RF24 library "changes" the properties
of the SPI bus not only during its use, (the changes are "permanent") so
that other devices with different SPI settings become "incompatible".
One solution offered by some sources is to add the SPI.beginTransaction()
/ SPI.endTransaction() at the beginning and end of each call to the
specific devices.
As I said, I do not the enough knowledge to understand in depth what they
are talking about, but maybe you could use this opportunity to post some
clarification on RF24 library common problems section or even consider
changing the code in case their observations are correct.
Regards,
Roger

Avamander, 2024-10-22:
Some devices do not work well on a shared SPI bus, nRF24L01+ is luckily not one of those. This library utilizes transactions and leaves the handling of the bus to the Arduino HAL. Very standard way of operating.

If any of your other devices on the same bus require specific bus configuration or aren't properly using transactions then those indeed have to be manually implemented in your code, in locations where you're interacting with those devices. Such workarounds might also interfere with the operation of this library.

Please try to get a minimal example working. Connect other devices to your bus and include relevant libraries one-by-one. This will help you narrow down the incompatibility. You should also make sure things like the TFT library isn't setting the bus speed too high for the nRF24L01+.

2bndy5, 2024-10-22:
> I found some references to it while googling...

Then you might be reading about outdated problems. The RF24 library has been improved in the last 2 years to be more flexible when used with other SPI devices that don't have a CS pin (effectively not allowed to share a SPI bus with another SPI device).

We have found that some cheap TFT displays and SD card modules (especially those purchased from amazon, aliexpress, alibaba, and others) don't expose a CS pin which makes it impossible for the nRF24L01 radio to be used on the same bus/pins (MOSI, MISO, SCLK). That is a hardware issue, meaning it is out of our control as software developers.

> One solution offered by some sources is to add the SPI.beginTransaction()
/ SPI.endTransaction() at the beginning and end of each call to the
specific devices.

That has been implemented in RF24 library for over 10 years. I think you are misunderstanding the information you found from google.

> The explanation provided is that the RF24 library "changes" the properties
of the SPI bus not only during its use, (the changes are "permanent") so
that other devices with different SPI settings become "incompatible".

1. The changes are not permanent. There is no API that can even do that permanently.
2. The SPI properties are changed _because_ we use `SPI.beginTransaction()`/`SPI.endTransaction()`. This would only be a problem if some other library uses the same default SPI pins (MOSI, MISO, SCLK) but does not use `SPI.beginTransaction()`/`SPI.endTransaction()`.

> [!NOTE]
> Not all Arduino cores provide `SPI.beginTransaction()`/`SPI.endTransaction()` as part of the SPI.h library. The [ESP32 Arduino-core does provide these methods](https://github.com/espressif/arduino-esp32/blob/4285912a46d6b5283a55d2342a7ca5ed1b926e15/libraries/SPI/src/SPI.h#L33).

> maybe you could use this opportunity to post some
clarification on RF24 library common problems section 

This problem isn't specific to the nRF24L01. Rather, this is more about understanding how SPI works. We are not here to tutor people on such topics.

rodd01, 2024-10-26:
Coming back to inform that the issue was caused by a hardware failure.
My apologies for the trouble.

2bndy5, 2024-10-26:
Ok. Since there's nothing to be done in the library sources, I'm closing this.

closed 2024-10-26 by 2bndy5