nRF24/RF24 #1012: [Question]: Can I use openWritePipe to modify the sending address before each shipment?

camerakang, 2024-12-19:
### What would you like to know?

`size_t rf24_send(uint8_t *send_buffer, int send_len, uint8_t *recv_buffer)
{
    radio.setAutoAck(true);
    unsigned long start_timer = micros(); // start the timer
    Serial.print(F("Sending data: "));
    for (size_t i = 0; i < send_len; i++)
    {
        Serial.print(send_buffer[i], HEX);
        Serial.print(" ");
    }
    Serial.println();

    bool report = radio.writeFast(send_buffer, send_len); // transmit & save the report
    unsigned long end_timer = micros();                   // end the timer

    if (report)
    {
        // Serial.print(F("Transmission successful! ")); // payload was delivered
        // Serial.print(F("Time to transmit = "));
        // Serial.print(end_timer - start_timer); // print the timer result
        // Serial.print(F(" us. Sent: "));

        uint8_t pipe;
        if (radio.available(&pipe))
        { // is there an ACK payload? grab the pipe number that received it
            size_t bytes = radio.getDynamicPayloadSize();
            radio.read(recv_buffer, bytes); // get incoming ACK payload
            // Serial.print(F(" Received "));
            // Serial.print(bytes); // print incoming payload size
            // Serial.print(F(" bytes on pipe "));
            // Serial.print(pipe); // print pipe number that received the ACK
            // Serial.print(F(": "));
            // for (uint8_t i = 0; i < radio.getDynamicPayloadSize(); i++)
            // {
            //     Serial.print(recv_buffer[i], HEX);
            //     Serial.print(" ");
            // }
            // Serial.println();
            return bytes;
        }
        else
        {
            Serial.println(F(" Received: an empty ACK packet")); // empty ACK packet received
            return 0;
        }
    }
    else
    {
        Serial.println(F("Transmission failed or timed out")); // payload was not delivered
        return 0;
    }
}
void change_address_send(uint8_t *address, uint8_t *send_buffer, int send_len, uint8_t *recv_buffer)
{
    // 打印address
    radio.openWritingPipe(address); // always uses pipe 0
    radio.stopListening(); // this also discards any unused ACK payloads
    radio.printDetails();
    auto recv_len = rf24_send(send_buffer, send_len, recv_buffer);
    if (recv_len > 0)
    {
        // Serial.print(F("Received "));
        // for (int i = 0; i < recv_len; i++)
        // {
        //   Serial.print(dataToSend[i], HEX);
        //   Serial.print(" ");
        // }
    }
}`
I have encapsulated two functions, and I want to modify the specified address before sending, but no matter what, I cannot receive the data on the receiving end.Here is the print I received
`SPI Speedz	= 10 Mhz
STATUS		= 0x3e RX_DR=0 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0x3165646f4e 0x000000004e
RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
TX_ADDR		= 0x3165646f4e
RX_PW_P0-6	= 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		= 0x3f
EN_RXADDR	= 0x03
RF_CH		= 0x4c
RF_SETUP	= 0x03
CONFIG		= 0x0e
DYNPD/FEATURE	= 0x3f 0x06
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_LOW
ARC		= 15
Sending data: A5 A5 D 1 1 2 3 4 5 6 7 8 9 A E9 4E 
 Received: an empty ACK packet
SPI Speedz	= 10 Mhz
STATUS		= 0x3e RX_DR=0 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0x3265646f4e 0x000000004e
RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
TX_ADDR		= 0x3265646f4e
RX_PW_P0-6	= 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		= 0x3f
EN_RXADDR	= 0x03
RF_CH		= 0x4c
RF_SETUP	= 0x03
CONFIG		= 0x0e
DYNPD/FEATURE	= 0x3f 0x06
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_LOW
ARC		= 15
Sending data: A5 A5 D 1 1 2 3 4 5 6 7 8 9 A E9 4E 
 Received: an empty ACK packet
SPI Speedz	= 10 Mhz
STATUS		= 0x3e RX_DR=0 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0x3365646f4e 0x000000004e
RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
TX_ADDR		= 0x3365646f4e
RX_PW_P0-6	= 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		= 0x3f
EN_RXADDR	= 0x03
RF_CH		= 0x4c
RF_SETUP	= 0x03
CONFIG		= 0x0e
DYNPD/FEATURE	= 0x3f 0x06
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_LOW
ARC		= 15
Sending data: A5 A5 D 1 1 2 3 4 5 6 7 8 9 A E9 4E 
 Received: an empty ACK packet
SPI Speedz	= 10 Mhz
STATUS		= 0x3e RX_DR=0 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0x3465646f4e 0x000000004e
RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
TX_ADDR		= 0x3465646f4e
RX_PW_P0-6	= 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		= 0x3f
EN_RXADDR	= 0x03
RF_CH		= 0x4c
RF_SETUP	= 0x03
CONFIG		= 0x0e
DYNPD/FEATURE	= 0x3f 0x06
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_LOW
ARC		= 15
Sending data: A5 A5 D 1 1 2 3 4 5 6 7 8 9 A E9 4E 
 Received: an empty ACK packet
`
Here is my receiver
`
[15:13:34.893]收←◆__cplusplus: 202100

[15:13:34.920]收←◆SPI Speedz	= 10 Mhz
STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P
[15:13:34.926]收←◆_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0x65646f4e31 0x3165646f4e
RX_ADDR_P2-5	= 0xa0 0xab 0xac 0xad
TX_ADDR		= 0x65646f4e31
RX_PW_P0
[15:13:34.938]收←◆-6	= 0x20 0x20 0x20 0x20 0x20 0x20
EN_AA		= 0x3f
EN_RXADDR	= 0x02
RF_CH		= 0x4c
RF_SETUP	= 0x03
CONFIG		= 0x0f
DYNPD/FEATURE	= 0x3f 0x06
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Leng
[15:13:34.954]收←◆th	= 16 bits
PA Power	= PA_LOW
ARC		= 0
`
We look forward to your reply,Thank you.

2bndy5, 2024-12-19:
I don't see anything in the code that cause the receiver to not work, but you only submitted the sender side. Looking at the `printDetails()` output from your receiver side, the settings seem correct.

1. What radio module are you using?
2. Did you try running [our `AcknowledgementPayloads` example](https://github.com/nRF24/RF24/blob/master/examples/AcknowledgementPayloads/AcknowledgementPayloads.ino) to verify the radio's are working correctly?

camerakang, 2024-12-19:
> 我没有在代码中看到任何导致接收器无法工作的内容，但您只提交了发送方。查看`printDetails()`接收器端的输出，设置似乎正确。
> 
> 1. 您使用什么无线电模块？
> 2. 您是否尝试运行[我们的`AcknowledgementPayloads`示例](https://github.com/nRF24/RF24/blob/master/examples/AcknowledgementPayloads/AcknowledgementPayloads.ino)来验证无线电是否正常工作？

Thank you for your reply. Please add the radio. openWriting Pipe (address) in the changeaddresses_send function; radio.stopListening();  It is possible to send and receive ACKs normally by commenting them out. I encountered an issue when modifying the address before sending after activating the sending and receiving functions,
`    change_address_send(send_address[0], frame().data(), frame().size(), recv_buffer);
    change_address_send(send_address[1], frame().data(), frame().size(), recv_buffer);
    change_address_send(send_address[2], frame().data(), frame().size(), recv_buffer);
    change_address_send(send_address[3], frame().data(), frame().size(), recv_buffer);
    delay(100);`
    When I call in a loop like this, the receiver cannot receive the message，But when I only call `change_address_send(send_address[0], frame().data(), frame().size(), recv_buffer);`I can complete the communication.
    I am doing this in the hope of communicating with four receiving ends simultaneously with ACK

camerakang, 2024-12-19:
> 我没有在代码中看到任何导致接收器无法工作的内容，但您只提交了发送方。查看`printDetails()`接收器端的输出，设置似乎正确。
> 
> 1. 您使用什么无线电模块？
> 2. 您是否尝试运行[我们的`AcknowledgementPayloads`示例](https://github.com/nRF24/RF24/blob/master/examples/AcknowledgementPayloads/AcknowledgementPayloads.ino)来验证无线电是否正常工作？

I am using the nRF24L01+module

2bndy5, 2024-12-19:
> I am using the nRF24L01+module

I mean, what exact radio module are you using? Where did you buy it? Is it actually an Si24R1 (a bad Chinese clone of nRF24L01+)?

camerakang, 2024-12-19:
> > 我正在使用 nRF24L01+模块
> 
> 我的意思是，您使用的是什么_无线电_模块？您在哪里买的？它实际上是 Si24R1（nRF24L01+ 的劣质中国仿制品）吗？

No, he is the NORDIC original nRF24L01P RF chip

TMRh20, 2024-12-19:
Can you post more complete examples that we can view/test with?

camerakang, 2024-12-24:
Thank you for your patient response. I have been delayed by other things...
I changed the address, and the phenomenon is the same
This is my sender code [https://github.com/camerakang/Nrf24_demo_send](url)
This is my receiver code[https://github.com/camerakang/Nrf24_demo_rev](url)
When in use, transmit the desired data to the sender through the serial port
When in use, transmit the desired data to the sender through the serial port

TMRh20, 2024-12-24:
Sender:

```
#include <SPI.h>
#include "RF24.h"
RF24 radio(7,8);

size_t send_len;
bool send_flag = true;

uint8_t dataToSend[32] = {};
uint8_t dataToReceive[32];
uint8_t recv_buffer[32]{"recv buffer is empty"};
uint8_t send_buffer[32]{"send buffer is empty"};

uint8_t address[][6] = {"1Node", "2Node"};
uint8_t send_address[][6] = {"3Node", "4Node", "5Node", "6Node"};

void change_address_send(uint8_t *address, uint8_t *send_buffer, int send_len, uint8_t *recv_buffer);
size_t rf24_send(uint8_t *send_buffer, int send_len, uint8_t *recv_buffer);

void setup() {
  Serial.begin(115200);
  while (!Serial)
  {
  }
  //Serial.setRxTimeout(1);
  //Serial.onReceive(onReceive);
    if (!radio.begin())
    {
        Serial.println(F("radio hardware is not responding!!"));
        while (1)
        {
        } // hold in infinite loop
    }

    // Set the PA Level low to try preventing power supply related problems
    // because these examples are likely run with nodes in close proximity to
    // each other.
    radio.setPALevel(RF24_PA_LOW); // RF24_PA_MAX is default.

    // to use ACK payloads, we need to enable dynamic payload lengths (for all nodes)
    radio.enableDynamicPayloads(); // ACK payloads are dynamically sized

    // Acknowledgement packets have no payloads by default. We need to enable
    // this feature for all nodes (TX & RX) to use ACK payloads.
    radio.enableAckPayload();

    // set the TX address of the RX node into the TX pipe
    radio.openWritingPipe(send_address[1]); // always uses pipe 0

    // set the RX address of the TX node into a RX pipe
    radio.openReadingPipe(1, address[0]); // using pipe 1

    radio.stopListening(); // this also discards any unused ACK payloads
    // For debugging info
    // printf_begin(); // needed only once for printing details
    radio.printDetails(); // (smaller) function that prints raw register values

}

uint8_t myData[5] = {1,2,3,4,5};

void loop() {
  if (send_flag)
  {
    //send_flag = false;
    //auto frame{sprotocol_send->make_packer(1, 32)};
    //frame.push_back(send_buffer, send_len).end_pack();
    
[... 144 lines cut ...]
        Serial.print(F(" bytes on pipe "));
        Serial.print(pipe); // print the pipe number
        Serial.println(F(": "));
        // for (uint8_t i = 0; i < bytes; i++)
        // {
        //     Serial.print(recv_buffer[i], HEX);
        // }
        // Serial.println();
        // Serial.print("Sending ACK payload...");
        // for (uint8_t i = 0; i < bytes; i++)
        // {
        //     Serial.print(send_buffer[i], HEX);
        // }
        // Serial.println();
        radio.writeAckPayload(1, send_buffer, send_len);
        return bytes;
    }
    return 0;
}

```

The main things I changed:
1. `bool report = radio.write(send_buffer, send_len);1  - use the standard write() instead of writeFast(). Otherwise the radio will not wait for the ack packet, it will just write the data to the FIFO and continue.
2. `myData[0]++;`` - You need to ensure the data in each packet is different. You can do this by adding an 8 or 16-bit header to the data, this will ensure the CRC is different each time so the radios will not discard the packet.

camerakang, 2024-12-25:
> 发件人：
> 
> ```
> #include <SPI.h>
> #include "RF24.h"
> RF24 radio(7,8);
> 
> size_t send_len;
> bool send_flag = true;
> 
> uint8_t dataToSend[32] = {};
> uint8_t dataToReceive[32];
> uint8_t recv_buffer[32]{"recv buffer is empty"};
> uint8_t send_buffer[32]{"send buffer is empty"};
> 
> uint8_t address[][6] = {"1Node", "2Node"};
> uint8_t send_address[][6] = {"3Node", "4Node", "5Node", "6Node"};
> 
> void change_address_send(uint8_t *address, uint8_t *send_buffer, int send_len, uint8_t *recv_buffer);
> size_t rf24_send(uint8_t *send_buffer, int send_len, uint8_t *recv_buffer);
> 
> 
> void setup() {
>   Serial.begin(115200);
>   while (!Serial)
>   {
>   }
>   //Serial.setRxTimeout(1);
>   //Serial.onReceive(onReceive);
>     if (!radio.begin())
>     {
>         Serial.println(F("radio hardware is not responding!!"));
>         while (1)
>         {
>         } // hold in infinite loop
>     }
> 
>     // Set the PA Level low to try preventing power supply related problems
>     // because these examples are likely run with nodes in close proximity to
>     // each other.
>     radio.setPALevel(RF24_PA_LOW); // RF24_PA_MAX is default.
> 
>     // to use ACK payloads, we need to enable dynamic payload lengths (for all nodes)
>     radio.enableDynamicPayloads(); // ACK payloads are dynamically sized
> 
>     // Acknowledgement packets have no payloads by default. We need to enable
>     // this feature for all nodes (TX & RX) to use ACK payloads.
>     radio.enableAckPayload();
> 
>     // set the TX address of the RX node into the TX pipe
>     radio.openWritingPipe(send_address[1]); // always uses pipe 0
> 
>     // set the RX address of the TX node into a RX pipe
>     radio.openReadingPipe(1, address[0]); // using pipe 1
> 
>     radio.stopListening(); // this also discards any unused ACK payloads
>     // For debugging info
>     // printf_begin(); // needed only once for printing details
>     radio.printDetails(); // (smaller) function that prints raw register values
> 
> }
> 
> uint8_t myData[5] = {1,2,3,4,5};
> 
> void loop() {
>   if (send_flag)
>   {
>     //send_flag = false;
>     //auto frame{sprotocol_send->make_packer(1, 32)};
>     //frame.push_back(send_buffer, send_len).end_pack();
[... 156 lines cut ...]
>         Serial.println(F(": "));
>         // for (uint8_t i = 0; i < bytes; i++)
>         // {
>         //     Serial.print(recv_buffer[i], HEX);
>         // }
>         // Serial.println();
>         // Serial.print("Sending ACK payload...");
>         // for (uint8_t i = 0; i < bytes; i++)
>         // {
>         //     Serial.print(send_buffer[i], HEX);
>         // }
>         // Serial.println();
>         radio.writeAckPayload(1, send_buffer, send_len);
>         return bytes;
>     }
>     return 0;
> }
> ```
> 
> 我改变的主要内容：
> 
> 1. `bool report = radio.write(send_buffer, send_len);`  - 使用标准 write() 而不是 writeFast()。否则无线电不会等待 ack 数据包，它只会将数据写入 FIFO 并继续。
> 2. `myData[0]++;`- 您需要确保每个数据包中的数据都不同。您可以通过向数据添加 8 位或 16 位标头来实现这一点，这将确保 CRC 每次都不同，因此无线电不会丢弃数据包。

Thank you very much!!!!!!!! So the main reason is that the same data was sent and the radio dropped the packet right? Now it's working.

camerakang, 2024-12-25:
But what if I just want to send the same packet, do I need to add a timestamp inside the protocol?

TMRh20, 2024-12-25:
Here is what I know, from the datasheet.

[image]

A timestamp would work, but an 8-bit header would be the simplest solution:

Packet format:
( HEADER,  DATA )
Where header is a value between 0 and 255 that increments each send, followed by your data.

See RF24Network for examples of this:
https://nrf24.github.io/RF24Network/structRF24NetworkHeader.html#ad05daaac28b7d1e7a7ef749f7f0ce317

camerakang, 2024-12-25:
[image]
[image]
`    /**
     * Set the number of retry attempts and delay between retry attempts when
     * transmitting a payload. The radio is waiting for an acknowledgement
     * (ACK) packet during the delay between retry attempts.
     *
     * @param delay How long to wait between each retry, in multiples of
     * 250 us. The minimum of 0 means 250 us, and the maximum of 15 means
     * 4000 us. The default value of 5 means 1500us (5 * 250 + 250).
     * @param count How many retries before giving up. The default/maximum is 15. Use
     * 0 to disable the auto-retry feature all together.
     *
     * @note Disable the auto-retry feature on a transmitter still uses the
     * auto-ack feature (if enabled), except it will not retry to transmit if
     * the payload was not acknowledged on the first attempt.
     */
    void setRetries(uint8_t delay, uint8_t count);`
    Auto-retry is the default，Even if three addresses don't reply with ACK,that should only waste 1500us*3 right? The switching address communication time should be 10 (my send delay) + 1500us*3 (auto retry time), so why does it take closer to 100ms?
    Here's the function I sent the call to
    
[image]

camerakang, 2024-12-25:
I tested the change_address_send function without an ACK reply it took 30ms 。。。。。

camerakang, 2024-12-25:
Wow!  I set `radio.setRetries(0, 0);`, It's works great. But I haven't figured out why the send function takes 30ms when there's no receiver to receive it.
Thank you both for your patience!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

2bndy5, 2024-12-25:
> Wow! I set radio.setRetries(0, 0);, It's works great. But I haven't figured out why the send function takes 30ms when there's no receiver to receive it.

`setRetries(0, 0)` tells the radio to wait at least 250 microseconds for an ACK packet before returning failure.

camerakang, 2024-12-25:
> > 哇！我设置了 radio.setRetries(0, 0);，效果很好。但我还没搞明白为什么在没有接收器接收时，发送函数需要 30 毫秒。
> 
> [`setRetries(0, 0)`](https://nrf24.github.io/RF24/classRF24.html#a4c6d3959c8320e64568395f4ef507aef)告诉无线电在返回失败之前等待至少 250 微秒的 ACK 数据包。

It's working so well now that I can even communicate with more than ten receivers on a time-sharing basis and keep it under 15ms at a time!   ❤

2bndy5, 2024-12-25:
I guess we can close this now?

camerakang, 2024-12-25:
> 我想我们现在可以结束这件事了吗？

Yes, thank you very very very very very very very very very very very very much.

closed 2024-12-25 by 2bndy5