nRF24/RF24 #1013: [Question]: How to reset Packet ID on the RX side?

CSC-Olivier, 2024-12-30:
### What would you like to know?

Hi,
I have been using your library for many years, thanks!
To save more power, I cut the power to the radio on the TX side between transmissions. I send the same advertisement packets over and over.
Because they have the same payload (as well as CRC and Packet ID), the radio on the RX side automatically reject the packet as a re-transmit of the same packet. I understand that this is a feature of the ESB protocol to avoid reporting the same packet twice. But in my case, that prevents me for receiving a legitimate packet.

I cannot change my code or hardware on the TX side or my hardware on the RX side.
Is there a code-only solution to solving this on the RX side?
I tried to powerDown(), powerUp() the radio, to empty the RX and TX buffer, to change the CRC and AutoAck settings, to stopListening and transmit and then startListening again, but it seems the radio still keep the record of the last successfully acknowledged received payload and reject new identical payload even when received after all these changes and after 1minute or two.
Any help on this would be greatly appreciated and might be useful to other users of these radios.

Thanks for the help,
Olivier

TMRh20, 2024-12-30:
Well if the tx radio is being powered down each time it might be sending the same packet id. Essentially tho the only way I know to work around/test this would be to use a header in each payload with a randomly generated number inside to ensure the crc is different each time. The random() function on Arduino is not really random btw with AVR devices.

CSC-Olivier, 2024-12-31:
Thanks for your answers. I do power down the NRF24 module between each transmit. When the packets are identical, the RX don't receive them, when there are different they are received properly. I am pretty sure it is coming from the PID and CRC being identical.
The problem I have is that I can't change the code on the TX side. I can however change the code easily on the RX side. 
The CRC and PID of the last successfully received payload must be stored somewhere on the RX side registers (maybe in one of the reserved sections?).
Are you aware of any way I could access or reset these on the RX side?

TMRh20, 2024-12-31:
I should state that you have ventured into technically unsupported territory. We normally deal with library issues here, not technical issues using it. For that we generally recommend forums like the Arduino forums.

However, you can take a look at disabling auto-ack and CRC validation on the RX side. Check out our scanner example or the docs for more info. You need to play around with settings a bit more and see if this is possible.

2bndy5, 2024-12-31:
The CRC polynomial and PID counter cannot be changed because they are designed to be automatic. And no, there is no known API (software or hardware) that allows access to do so on the nRF24L01.

Your choices are
1. Alter the TX payload to include changing info.
2. Don't cut the power between similar transmissions.
3. Broadcast a garbage payload after the TX advertisement, to increment the PID counter on the RX side.

All of these ideas involve changing the TX side. I added the third one as a hypothetical idea.

TMRh20, 2025-01-01:
One more idea that just struck me is that you could also change the RX device over to an nRF52840 based device like the Feather, Clue or XIAO Sense boards, which are over-the-air compatible with nRF24, but everything is driven in software, so you could control the PID/CRC manually or simply deactivate this functionality.

https://github.com/TMRh20/nrf_to_nrf

2bndy5, 2025-01-01:
You wouldn't want to change the CRC polynomial. I think if you could alter duplicate packet detection, you would probably timestamp the original and accept a dupe after a period of time.

CSC-Olivier, 2025-01-02:
Thanks again for your answers. I ended up using the advice from #768.
I can configure the RX side in SB mode by disabling AutoAck, CRC and DynamicPayload (all 3 of these settings are needed). I can then receive all the payloads sent by TX side but they include the 9 bits control packet field (and stop 9 bits short of the whole 32 Bytes payload). So I need to shift the payload by 9 bits  (and loose the last 2 bytes of payload in the process). I can't check the CRC as well.
So it is not a perfect solution but it allows me to work with these legacy TX device by only changing the code on the RX side.
For upcoming sensors nodes, I will add a 2bit field in the payload that is incremented by the micro-controller to prevent running into this issue again.
Thanks again for taking the time to answer.
Closing this question, now.

closed 2025-01-02 by CSC-Olivier