nRF24/RF24 #1015: Per RF24Network issue https://github.com/nRF24/RF24Network/issues/242 failureDetected var not initialized at runtime

TMRh20, 2025-01-08:
### What radio module do you use?

All modules affected

### What driver board(s) do you use?

All devices affected

### If using Linux, what OS are you using?

```text
Linux 6.1.21-v8+ aarch64
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

### If using Linux, what RF24 driver did you select?

SPIDEV (linux kernel)

### Describe your problem

failureDetected var is not initilized at runtime, See https://github.com/nRF24/RF24Network/issues/242

### What is the RX code?

```cpp
if(radio.failureDetected){
  Serial.print("Failure Detected:");
  Serial.println(radio.failureDetected);
  radio.failureDetected = false;
}
```
```

### What is the TX code?

```cpp
n/a
```

2bndy5, 2025-01-08:
Thinking an initializer in c'tors should suffice, right?
https://github.com/nRF24/RF24/blob/80e7f137ae4caf0df4889b5499c22857019a328a/RF24.cpp#L564-L576
Or just add the `#ifdef` paragraph to `_init_obj()`
https://github.com/nRF24/RF24/blob/80e7f137ae4caf0df4889b5499c22857019a328a/RF24.cpp#L580-L592

TMRh20, 2025-01-08:
Was thinking c'tors because thats where everything else is initialized.

2bndy5, 2025-01-08:
```c++
#if defined FAILURE_HANDLING
    , failureDetected(0)
#endif
```
vs

```cpp
#if defined FAILURE_HANDLING
    failureDetected = 0;
#endif
```

PR #1016 (2025-01-10): init `failureDetected`
resolves #1015
closed 2025-01-10 by 2bndy5