nRF24/RF24 #1020: ACK Payloads with 3 or more pipes - FIFO filling up

TMRh20, 2025-01-28:
### What radio module do you use?

PA+LNA or without PA+LNA

### What driver board(s) do you use?

Arduino Nano, Arduino Due

### If using Linux, what OS are you using?

```text

```

### If using Linux, what RF24 driver did you select?

None

### Describe your problem

I modified the ACK Payloads example to send to individual pipes, 1, 2, 3 and sequences through the pipes and successfully receives an ACK payload. The transmitter should be device 1, receiver device 0.

My problem is that on the receiving end, if I use >2 pipes, the TX FIFO is filling up for some reason.

For example, if only using two pipes, the example still functions as required:
1. Payload is sent to recipient
2. Ack payload is received
3. Ack payload IS cleared from RX devices TX FIFO...

With 3 pipes:
1. Payload is sent to recipient
2. Ack payload is received
3. Ack payload is NOT cleared from RX devices TX FIFO

If you change the `#define NUM_PIPES` to 2 it works fine...

Having problems figuring out if this is normal radio behavior or if I'm doing something wrong here.

### What is the RX code?

```cpp
/*
 * See documentation at https://nRF24.github.io/RF24
 * See License information at root directory of this library
 * Author: Brendan Doherty (2bndy5)
 */

/**
 * A simple example of sending data from 1 nRF24L01 transceiver to another
 * with Acknowledgement (ACK) payloads attached to ACK packets.
 *
 * This example was written to be used on 2 devices acting as "nodes".
 * Use the Serial Monitor to change each node's behavior.
 */
#include <SPI.h>
#include "printf.h"
#include "RF24.h"

#define CE_PIN 7
#define CSN_PIN 8

#define NUM_PIPES 3

// instantiate an object for the nRF24L01 transceiver
RF24 radio(CE_PIN, CSN_PIN);

// an identifying device destination
// Let these addresses be used for the pair
uint8_t address[][6] = { "1Node", "2Node", "3Node","4Node" };
// It is very helpful to think of an address as a path instead of as
[... 191 lines cut ...]
      memcpy(payload.message, "Hello ", 6);  // change payload message
      radio.stopListening();                 // this also discards any unused ACK payloads

    } else if (c == 'R' && role) {
      // Become the RX node

      role = false;
      Serial.println(F("*** CHANGING TO RECEIVE ROLE -- PRESS 'T' TO SWITCH BACK"));
      memcpy(payload.message, "World ", 6);  // change payload message
      payload.pipeCounter = 1;
      // load the payload for the first received transmission on pipe 0
      radio.writeAckPayload(1, &payload, sizeof(payload));
      radio.startListening();
    }
  }
}  // loop
```

### What is the TX code?

```cpp
Same, make sure to configure transmitter as device 1, receiver as device 0

Change #define NUM_PIPES to 2 for a working example that doesn't flush the TX FIFO
```

2bndy5, 2025-01-28:
Its been a long time since I did a deep dive of the ACK payloads' behavior. IIRC, the order of the received payloads' origins must match the order of the loaded ACK payloads' destination. Remember, FIFO stands for "First In First Out".

In pseudo code (using python because its easier to ~play with~ debug at runtime):
```py
r = RF24(22, 0)
# ...

is_loaded = r.writeAckPayload(1, b"\xc1")
is_loaded = r.writeAckPayload(2, b"\xc2")
is_loaded = r.writeAckPayload(3, b"\xc3")
assert is_loaded, "failed to upload all 3 ACK payloads"

# ...

# assuming the TX FIFO is fully loaded as shown above
is_available, pipe = r.available_pipe()
if is_available and pipe > 1:
    # ACK payload in first level of TX FIFO not sent because origin was not pipe 1
    print("TX FIFO state:", r.is_fifo(True))
```

TMRh20, 2025-01-28:
>  the order of the received payloads' origins must match the order of the loaded ACK payloads' destination

Yeah, I think I've got that. The receiver writes first to pipe 1, and the payload contains the pipe number which is then incremented, so the ack payload is written to the next pipe. If no payloads are lost, theoretically this would work, however...

1. Radio A sends a payload to radio B using Auto-ack on pipe1
2. Radio B replies with an ack payload using pipe1 info
  a. Radio B has no way of knowing if radio A received the ACK payload for pipe1. It must hold on to it.
  b. I am assuming here, from testing that radio B will hold onto the ACK payload in the TX FIFO indefinitely, or until a new payload is received on Pipe1
3. Testing seems to indicate this is the behavior related to issues I've noticed in the past with only being able to use 2 ack-payloads at a time

The weird thing is, the radio seems to clear ALL the ack payloads when it does clear them. This is demonstrated by commenting out `radio.flush_tx()` in my example code. The radio will receive 3 ack payloads OK, but the 4th is blank, then it works 3 times again.

I am starting to believe this is just standard radio inner-workings.

2bndy5, 2025-01-29:
> The radio will receive 3 ack payloads OK, but the 4th is blank, then it works 3 times again.

This sounds similar to #1002 , but that issue was a problem with the user code (and was resolved).

> radio B will hold onto the ACK payload in the TX FIFO indefinitely, or until a new payload is received on Pipe1

If the TX FIFO's first level has an ACK payload for only pipe 1, then this sounds like expected behavior to me.

Are you using timed intervals or a sync'd timestamp to ensure the correct order of received payloads?

## Looking again at the datasheet

### In Appendix A (section 2 about "Received Payload" behavior):

> If auto acknowledgement is enabled, an ACK packet is transmitted back, unless the NO_ACK bit
is set in the received packet. If there is a payload in the `TX_PLD` FIFO, this payload is added to
the ACK packet.

This sounds like nordic didn't intend ACK payloads to be used for multiple pipes at the same time. Here, they don't mention the need for the received payload's pipe number to match the ACK payload's destination pipe number.

### In section 7.5.1:

> If the TX FIFO (PRX) contains more than one payload to a PTX, payloads are handled using the first in first out principle. The TX FIFO (PRX) is blocked if all pending payloads are addressed to a PTX where the link is lost. In this case, the MCU can flush the TX FIFO (PRX) by using the FLUSH_TX command.

This sounds like the what I was saying about the order of received payloads. The "where the link is lost" part usually just means "expected data was not transmitted" in the datasheet.

2bndy5, 2025-01-29:
FYI, when I was researching (long ago) about how Logitech uses nRF24LU1 (the precursor to nRF51 family), I found that they use ACK payloads. However, they

1. send a request for an ACK payload (ACK packet doesn't require a payload attached here)
2. spam the same receiver with a dummy payload (probably should be a counter) until the the receiver responds with an ACK payload.

TMRh20, 2025-01-29:
> Are you using timed intervals or a sync'd timestamp to ensure the correct order of received payloads?

Just the above example, timed intervals of 1 second to send, with the pipe number in the payload, so the receiver knows which pipe to put the ack-payload.

> This part is confusing. Are you saying the radio will flush the TX FIFO every 3 ACK payloads (even if the FIFO is not empty)?

Nope, I'm saying it will hold onto each ACK payload that has been written until it receives a new payload on a pipe that contains an ACK payload. So for example, we sequentially, over 3 seconds end up filling up pipe 1, 2 & 3 with ack payloads, when the next TX happens to pipe 1, it ends up with a blank payload, because the FIFO is full. The radio automatically seems to flush all the payloads when this happens. When there are only 2 pipes, this is not a problem, because there is always a spare space in the FIFO, but the FIFO always remains occupied. 

This leads me to believe the radio is 'hanging on' to all 3 ack payloads until a new ack payload is received on one of the pipes containing an old ack payload.

TMRh20, 2025-02-01:
Just to clarify, it does seem that Auto Ack requires the receiving radio to hold onto the ACK Payload indefinitely or until another payload is received on a pipe that contains an ACK payload. This essentially limits the radio to holding 2 ack payloads when different pipes are used, unless the user manually clears the TX FIFO after receiving 3 payloads.

This appears to be standard radio behavior, so I'm closing this issue.

closed 2025-02-01 by TMRh20