nRF24/RF24 #1022: Are there issues with ESP32-DOWD-V3 (revision v3.1) board?

OleksandrGora, 2025-03-11:
### What radio module do you use?

nRF24L01+ PA/LNA, RF24L01+ both

### What driver board(s) do you use?

ESP32-DOWD-V3 (revision v3.1) 38 pin devkit board with external and internal antenna, use external antenna (disconected internal)

### If using Linux, what OS are you using?

```text

```

### If using Linux, what RF24 driver did you select?

None

### Describe your problem

Asking if there are issues with the outlined board. Radio boards (any, with/without ext. ant.) work well with ESP32-DOWD06 (revision V1.0) - usual 38 pin devkit board.
radio.begin() returns false.

### What is the RX code?

```cpp
#include <SPI.h>
#include <nRF24L01.h>
#include <RF24.h>

#define CE_PIN 4
#define CSN_PIN 5

const uint64_t Pipe = 0xAABBCCDD11LL;
RF24 radio(CE_PIN, CSN_PIN);

void setup_nrf24()
{
  if (!radio.begin())
  {
    Serial.println("radio is not responding!!");
  }
  else
  {
    radio.setChannel(27);
    radio.setDataRate(RF24_250KBPS);
    radio.setPALevel(RF24_PA_LOW);
    radio.openReadingPipe(1, Pipe);
    radio.startListening();
    Serial.println("Radio configured");
  }
}
```

### What is the TX code?

```cpp
// Transmitter sends data, it can be received with another ESP32 board
```

2bndy5, 2025-03-11:
I don't have the same exact board to verify.

Have you tried reducing the SPI speed? Some boards' SPI lines have a high impedance which does not work well with loose or long wires.

```c++
RF24 radio(CE_PIN, CSN_PIN, 4000000); // set SPI speed to 4MHz instead of default 10MHz
```

Just to be sure, have you checked with our [COMMON_ISSUES.md](https://github.com/nRF24/RF24/blob/master/COMMON_ISSUES.md)?

OleksandrGora, 2025-03-11:
Thanks for prompt reply.
Tried reduce SPI speed, unfortunately nothing helped.
I have connected a capasitor 100 microF to power pins, checked connection, did not find error. The radio works fine on another board, simply unconnect and connect again, pretty difficult to made a mistake. Looks like the board have the usual pinout like other devkit boards.
Can not move forward (can not change parameters as radio.begin() returns false at the very beginning). I thought may be present of both antennas on the board could influence on the radio module.
ESPNOW works on the board as well (just FYI).

2bndy5, 2025-03-11:
ESPNOW uses internal radio hardware. RF24 uses external radio hardware. When `radio.begin()` fails, it usually means a hardware problem.

ESP32 boards are known to have little power left over for external hardware. It might be simple case of not enough power ($`P = I * V`$). Capacitors cannot create amperage, they only help ensure voltage doesn't drastically spike or drop.

OleksandrGora, 2025-03-11:
Thank you. Will try to deal.

closed 2025-03-11 by OleksandrGora