nRF24/RF24 #1024: 2 transmitting addresses, only one set

bionotic, 2025-04-05:
### What radio module do you use?

nRF24L01+

### What driver board(s) do you use?

Arduino nano

### If using Linux, what OS are you using?

```text

```

### If using Linux, what RF24 driver did you select?

None

### Describe your problem

Not sure why but, Writing Pipe 1 stays open as 0xc2c2c2c2c2. I am Pretty sure its in the c++ code. Its been plaguing me for a while now, I found it when I Printed the pretty details. On a note: pipe 0 stays open as the address i set.  It appears that the second Writing Pipe (0xc2c2c2c2c2) is stopping module from transmitting  correctly. This error is in TX part of the code, RX looks fine on print details ( i think ). I'm new on GitHub this is the issue i have ever reported. but i will paste  what print details says. Thank you in advance .

18:27:47.669 -> SPI Frequency		= 10 Mhz
18:27:47.669 -> Channel			= 56 (~ 2456 MHz)
18:27:47.701 -> Model			= nRF24L01+
18:27:47.768 -> RF Data Rate		= 250 KBPS
18:27:47.768 -> RF Power Amplifier	= PA_MIN
18:27:47.801 -> RF Low Noise Amplifier	= Enabled
18:27:47.868 -> CRC Length		= 16 bits
18:27:47.868 -> Address Length		= 5 bytes
18:27:47.868 -> Static Payload Length	= 32 bytes
18:27:47.934 -> Auto Retry Delay	= 1000 microseconds
18:27:47.934 -> Auto Retry Attempts	= 5 maximum
18:27:48.001 -> Packets lost on
18:27:48.001 ->     current channel	= 0
18:27:48.068 -> Retry attempts made for
18:27:48.068 ->     last transmission	= 5
18:27:48.068 -> Multicast		= Disabled
18:27:48.101 -> Custom ACK Payload	= Disabled
18:27:48.168 -> Dynamic Payloads	= Disabled
18:27:48.168 -> Auto Acknowledgment	= Enabled
18:27:48.201 -> Primary Mode		= TX
18:27:48.234 -> TX address		= 0x4165646f6e
18:27:48.234 -> pipe 0 ( open ) bound	= 0x4165646f6e
18:27:48.301 -> pipe 1 ( open ) bound	= 0xc2c2c2c2c2
18:27:48.334 -> pipe 2 (closed) bound	= 0xc3
18:27:48.367 -> pipe 3 (closed) bound	= 0xc4
18:27:48.434 -> pipe 4 (closed) bound	= 0xc5
18:27:48.434 -> pipe 5 (closed) bound	= 0xc6

### What is the RX code?

```cpp
#include <RF24.h>
// ...
```

### What is the TX code?

```cpp
#include <SPI.h>
#include <printf.h>
#include <RF24.h>
#include <string.h>
#define RF24_CE_PIN 9
#define RF24_CSN_PIN 8

RF24 radio(RF24_CE_PIN, RF24_CSN_PIN);

const byte address[5] = {'n','o','d','e','A'};
void setup() {
  Serial.begin(9600);

  if (!radio.begin()) {
    Serial.println("Radio initialization failed!");
    while (1); // Stop execution
  } else {
    Serial.println("Radio initialization successful.");
  }
  radio.setChannel(56);
  radio.setDataRate(RF24_250KBPS);
  radio.setPALevel(RF24_PA_MIN);
 
 
  radio.setRetries(3, 5);
   radio.stopListening();
  radio.openWritingPipe(address);
 

  printf_begin();

 radio.printPrettyDetails();// Print radio configuration details
}

void loop() {
 const char text[] = "hello world";
  Serial.print("Sending: ");
  Serial.println(text);
bool good = radio.write(text, strlen(text) + 1);
  if (good) {
    Serial.println("Transmission sux.");
  } else {
    Serial.println("Transmission failed!");
  }

  delay(1000);
}
```

2bndy5, 2025-04-06:
> It appears that the second Writing Pipe (0xc2c2c2c2c2) is stopping module from transmitting correctly

The nRF24L01 has only used pipe 0 to transmit. This gets even more confusing when you consider the RX requirements of the auto-ack feature... I'm guessing you didn't read the datasheet at all (see Appendix A), so I won't bother confusing you further.

---
Setting aside your mistaken observation, your TX code is missing a call to `radio.stopListening()`. It can go anywhere before calling `radio.write()` (and after `radio.openWritingPipe()`), but it only needs to be called once as long as `radio.startListening()` is never called.

bionotic, 2025-04-06:
Could you elaborate further?..... i ran the code with radio.stopListening() in both locations. It still has the second address on PrintDetails. You are right to say i didn't read the datasheet. I will read it.  The IDE says it goes where i have it though. 

[image]

bionotic, 2025-04-06:
> > It appears that the second Writing Pipe (0xc2c2c2c2c2) is stopping module from transmitting correctly
> 
> The nRF24L01 only uses pipe 0 to transmit. This gets even more confusing when you consider the RX requirements of the auto-ack feature... I'm guessing you didn't read the datasheet at all (see Appendix A), so I won't bother confusing you further.
> 
> Setting aside your mistaken observation, your TX code needs to call `radio.stopListening()` before `radio.write()` and after `radio.openWritingPipe()`.

Data sheet says the same thing. no offense 

[image]

2bndy5, 2025-04-06:
> [image]

Please update the RF24 library.

---
> Be sure to call [stopListening()](https://nrf24.github.io/RF24/classRF24.html#a6f144d73fc447c8ac2d1a4166210fd88) prior to calling this function.

My bad. I got the order mixed up. Looking at the source code, this advice makes sense.

---

The address `0xc2c2c2c2c2` is the default address for pipe 1 hard coded in the radio's firmware. This lib opens RX pipe 1 by default because it is very common to use the radio to both receive and transmit.

Best experiences recommend using pipes 1-5 to receive and leave pipe 0 for only transmitting. You can still use pipe 0 to RX because `stopListening()` will set the TX address to the RX pipe 0 for auto-acking purposes. `startListening()` will restore whatever RX address was set to pipe 0 (if any). I highly encourage reading the datasheet's Appendix A because it has a detailed explanation about how pipe 0 (and pipe 1) is meant to be used.

---

You can close RX pipe 1 using `radio.closeReadingPipe(1)`, but I doubt that will solve your problem.

I noticed you have set the radio to 250kbps data rate, but you haven't shown us your RX config. Please read our COMMON_ISSUES' ["Settings that must match" guide](https://nrf24.github.io/RF24/md_COMMON__ISSUES.html#autotoc_md183).

Other than that, I'm not sure what is actually going wrong. What is the RX side doing? You only posted the code _you think_ is problematic.

bionotic, 2025-04-06:
i found it. It was in the RX code like you suggested.... i went over the data sheet closely the startListening was in the wrong spot. stopListening goes before openwriting pipe in TX. and startListening goes after open reading pipe in RX. MY BAD! thanks so much for answering and being there for me my man. I have self learned everything i know about much tech stuff by my self... never had a teacher.  sorry if i sounded stupid... much love.

2bndy5, 2025-04-06:
Glad to hear you solved it!

Sorry I was a bit curt at first. Something else unrelated to this significantly annoyed me... You just caught me on a bad day.