nRF24/RF24 #1033: csDelay required for Linux devices

TMRh20, 2025-04-28:
### What radio module do you use?

Both PA+LNA and smaller versions

### What driver board(s) do you use?

RPi5 & RPi4

### If using Linux, what OS are you using?

```text
Linux 6.6.51+rpt-rpi-2712 aarch64
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

### If using Linux, what RF24 driver did you select?

SPIDEV (linux kernel)

### Describe your problem

I found another bug...

There has been SPI instability problems that have plagued me for a long time on Linux devices. I've attempted everything, low PA & LNA levels, different radios & adapters, etc.
Lowering the SPI speed makes radio failures happen less, but it still fails from time to time.
This is why all the RF24Gateway examples have failure handling built-in.

I finally think I've identified the problem though, and it has to do with the `csDelay` variable.

This was included for Arduino devices, because fast devices can kind of swamp the radio with SPI requests when it should be focused on receiving/transmitting data. 

With Linux devices, it seems there is not enough of a delay for CS toggling before beginning SPI transactions. At least that's what I think the root cause is.

https://github.com/nRF24/RF24/blob/0c34c0f8f36e03c27f8402b7ed2e4c97037722a8/RF24.cpp#L96C1-L98C31

needs to be changed to

```cpp
#else
    static_cast<void>(mode); // ignore -Wunused-parameter
    delayMicroseconds(csDelay);
#endif // !defined(RF24_LINUX)
```

### What is the RX code?

```cpp
RF24Gateway ncurses or ncursesInt examples

iperf3 -s
```

### What is the TX code?

```cpp
Same as RX

Run the following command to cause failures after a short time:
iperf3 -c 10.1.3.134 -4 -w1500 -t 90
```

TMRh20, 2025-04-28:
Hehe, I tried that to no avail

2bndy5, 2025-04-28:
I used to have a similar problem in my RPi4 back when I was running Ubuntu (64 bit) on it. It was almost always solved by slowing the SPI speed.

TMRh20, 2025-04-28:
The issue is much more prevalent on RPi5, especially when doing max-rate testing with iperf3.

The Gateway examples log the failure counts to a file, so I've been trying different things (SPI Speeds etc) on RPi3, RPi4, but the RPi5 finally made me look into it more. The issue is repeatable and seems completely fixed with this change.

TMRh20, 2025-04-29:
Just FYI, I've been testing this fix overnight, pushing all 4 RPis pretty good, and I cannot recreate this issue now, regardless of SPI speed, data rate, power & PA levels. This is great!

2bndy5, 2025-04-30:
A signal analyzer should prove confirm its a CSN issue but the infrequency makes it hard to spot. I'm guessing the software viewer could watch for an extra pin that would be activated when the error occurs.

TMRh20, 2025-05-01:
I'm starting to think it must be something else, unless there is a problem with the SPIDEV driver itself, which I don't think.

Now I'm testing and looking at the following code:

https://github.com/nRF24/RF24/blob/48005ce854e695108fe478c7c21669cde6287a35/RF24.cpp#L1505-L1515

I think there is an issue at the network layer:

https://github.com/nRF24/RF24Network/blob/5b24caf7ca0b0a215466a8c55e2ecaa644008a43/RF24Network.cpp#L144-L159

should be

```cpp
#if defined(ENABLE_DYNAMIC_PAYLOADS) && !defined(XMEGA_D3)
        frame_size = radio.getDynamicPayloadSize();
#else
        frame_size = MAX_FRAME_SIZE;
#endif
        if(frame_size < sizeof(RF24NetworkHeader)){continue;}

        // Fetch the payload, and see if this was the last one.
        radio.read(frame_buffer, frame_size);

        // Read the beginning of the frame as the header
        RF24NetworkHeader* header = (RF24NetworkHeader*)(&frame_buffer);

        // Throw it away if it's not a valid address or too small
        if (!is_valid_address(header->to_node) || !is_valid_address(header->from_node)) {
            continue;
        }
```

Notice there should not be any radio.reads happening if the payload is rejected.

After making this change, I'm looking at 1 hour straight of no failures on any of my RPis. I hate having to wait for no response. It would be much more helpful if I could make this issue happen on demand.

2bndy5, 2025-05-01:
Your analysis is sound, as usual.

To support your findings further, I recall that a `radio.read()` on an empty FIFO will still fill the given buffer with the last byte stored in the RX FIFO (for the specified `len`). So, its possible that the repeated byte would pass the 2 `is_valid_address()` checks.

TMRh20, 2025-05-01:
Well still nothing, I'm remaining hopeful. Going to let it ride overnight at max throughput.

TMRh20, 2025-05-01:
OK, I've narrowed down the problem to the network layer.

In `update()` there is the following code:

https://github.com/nRF24/RF24Network/blob/5b24caf7ca0b0a215466a8c55e2ecaa644008a43/RF24Network.cpp#L137-L141

Now what I did was remove failure detection from RF24.cpp and leave it in the update() function in RF24Network.
Then at the Gateway layer, I left it to log the failures, but not restart the radio.

It turns out that something is causing radio.available() to get stuck in a loop, but simply returning from the loop after 1second fixes the problem. This could be a number of things, the simplest explanation being that data gets sent faster than it can be processed by the network layer via `update()`

The simple fix is to just remove the lines in RF24Network:

```cpp
#if defined FAILURE_HANDLING
            radio.failureDetected = 1;
#endif
```

Since this can be, but in this case is not actually a radio failure I'm almost thinking RF24Network should have its own variable, instead of borrowing from the RF24 layer. This would allow further investigation as to whether this always can be fixed by simply returning from `update()` or if there needs to be additional failure handling here.

2bndy5, 2025-05-01:
There's still some space in the [returned message types](https://nrf24.github.io/RF24Network/group__DEFINED__TYPES.html). Maybe we can reserve a set of numbers to indicate a radio/network/mesh/gateway error. Not sure what that would look like though.

TMRh20, 2025-05-01:
That sounds like a really good idea, just return a system payload indicating network overrun. Should be simple enough.

2bndy5, 2025-05-01:
I've been brainstorming how to implement the network layer in rust, and error handling would imply a different error type for each layer... That's what made me think of it.

closed 2025-05-02 by TMRh20