nRF24/RF24 #1037: Community assistance sought

The-Futurist, 2025-05-05:
Hello,

I'm developing a C API for abstracting the NRF24L01+ device for use on STM32 MCUs.

Unfortunately this device has acquired a reputation that is complicating my development, mainly anecdotal claims about "clone" chips and "power issues" and so on.

I'm posting here in the hope others can shed light on this, we can discuss the device itself and no the API as such.

The only real issue I've encountered is the apparent absence of acks when trying to use the auto-ack feature, getting two stations to communicate wasn't much effort once the base code as written but the auto-ack is frustrating me and I need to determine if this is indeed faulty hardware (clones etc) or something else.

I posted my problem on the Nordic forum, so please have a look at this and either share your thought there or here in this repo's issue (if that's OK with you folks).

[Understanding the NRF24L01+ Auto Ack apparently not working](https://devzone.nordicsemi.com/f/nordic-q-a/121178/understanding-the-nrf24l01-auto-ack-apparently-not-working)

Thank you.

TMRh20, 2025-05-05:
I've yet to discover an NRF24 radio that doesn't work with auto-ack.

```
SETUP_RETR (03)
  ARD         = 0
  ARC         = 3
```

Try setting the Auto Retry duration to something higher than 0. 5,5 is the default for RF24 if I remember.

The-Futurist, 2025-05-05:
Thanks, yes zero was asking for trouble, I corrected that and the sender still gets a MAX_RT interrupt. 

On the sender I actually now have:
```
 RX_ADDR_P0   =  33  33  33  33  33
 RX_ADDR_P1   =  C2  C2  C2  C2  C2
 RX_ADDR_P2   = [C2  C2  C2  C2] C3
 RX_ADDR_P3   = [C2  C2  C2  C2] C4
 RX_ADDR_P4   = [C2  C2  C2  C2] C5
 RX_ADDR_P5   = [C2  C2  C2  C2] C6

 TX_ADDR      = 19 51 38 31 00
```

and on the receiver we have:
```
 RX_ADDR_P0   =  33  33  33  33  33
 RX_ADDR_P1   =  19  51  38  31  00
 RX_ADDR_P2   = [19  51  38  31] C3
 RX_ADDR_P3   = [19  51  38  31] C4
 RX_ADDR_P4   = [19  51  38  31] C5
 RX_ADDR_P5   = [19  51  38  31] C6

 TX_ADDR      = E7 E7 E7 E7 E7
```

Is there something there that looks wrong to you?

TMRh20, 2025-05-05:
`RX_ADDR_P0   =  33  33  33  33  33` needs to match `  TX_ADDR      = 19 51 38 31 00`

The-Futurist, 2025-05-05:
> `RX_ADDR_P0 = 33 33 33 33 33` needs to match ` TX_ADDR = 19 51 38 31 00` on the tx side

OK I adjusted that and the sender once again gets a MAX_RT interrupt, here are the complete register settings for that test:

**Sender**
```
NRF24L01+ REGISTER DUMP
 CONFIG (0E)
  RESERVED    = 0
  MASK_RX_DR  = 0
  MASK_TX_DS  = 0
  MASK_MAX_RT = 0
  EN_CRC      = 1
  CRCO        = 1
  PWR_UP      = 1
  PRIM_RX     = 0

 EN_AA (03)
  RESERVED    = 0
  ENAA_P5     = 0
  ENAA_P4     = 0
  ENAA_P3     = 0
  ENAA_P2     = 0
  ENAA_P1     = 1
  ENAA_P0     = 1

 EN_RXADDR (03)
  RESERVED    = 0
  ERX_P5      = 0
  ERX_P4      = 0
  ERX_P3      = 0
  ERX_P2      = 0
  ERX_P1      = 1
  ERX_P0      = 1

 SETUP_AW (03)
  RESERVED    = 0
  AW          = 3

 SETUP_RETR (FF)
  ARD         = 15
  ARC         = 15

 RF_CH (2D)
  RESERVED    = 0
  RF_CH       = 45

 RF_SETUP (04)
  CONT_WAVE   = 0
  RESERVED    = 0
  RF_DR_LOW   = 0
  PLL_LOCK    = 0
  RF_DR_HIGH  = 0
  RF_PWR      = 2
  OBSOLETE    = 0

 STATUS (0E)
  RESERVED    = 0
  RX_DR       = 0
  TX_DS       = 0
  MAX_RT      = 0
  RX_P_NO     = 7
  TX_FULL     = 0

 OBSERVE_TX (0F)
  PLOS_CNT    = 0
  ARC_CNT     = 15

 RPD (00)
  RESERVED    = 0
  RPD         = 0

 RX_ADDR_P0   =  19  51  38  31  00
 RX_ADDR_P1   =  C2  C2  C2  C2  C2
 RX_ADDR_P2   = [C2  C2  C2  C2] C3
 RX_ADDR_P3   = [C2  C2  C2  C2] C4
 RX_ADDR_P4   = [C2  C2  C2  C2] C5
 RX_ADDR_P5   = [C2  C2  C2  C2] C6

 TX_ADDR      = 19 51 38 31 00

 RX_PW_0 (00)
  RESERVED    = 0
  RX_PW_P0    = 0

 RX_PW_1 (00)
  RESERVED    = 0
  RX_PW_P1    = 0

 RX_PW_2 (00)
  RESERVED    = 0
  RX_PW_P2    = 0

 RX_PW_3 (00)
  RESERVED    = 0
  RX_PW_P3    = 0

 RX_PW_4 (00)
  RESERVED    = 0
  RX_PW_P4    = 0

 RX_PW_5 (00)
  RESERVED    = 0
  RX_PW_P5    = 0

 FIFO_STATUS (11)
  RESERVED    = 0
  TX_REUSE    = 0
  TX_FULL     = 0
  TX_EMPTY    = 1
  RESERVED    = 0
  RX_FULL     = 0
  RX_EMPTY    = 1

 DYNPD (00)
  RESERVED    = 0
  DPL_P5      = 0
  DPL_P4      = 0
  DPL_P3      = 0
  DPL_P2      = 0
  DPL_P1      = 0
  DPL_P0      = 0

 FEATURE (00)
  RESERVED    = 0
  EN_DPL      = 0
  EN_ACK_PAY  = 0
  EN_DYN_ACK  = 0
```

**Receiver**
```
NRF24L01+ REGISTER DUMP
 CONFIG (3F)
  RESERVED    = 0
  MASK_RX_DR  = 0
  MASK_TX_DS  = 1
  MASK_MAX_RT = 1
  EN_CRC      = 1
  CRCO        = 1
  PWR_UP      = 1
  PRIM_RX     = 1

 EN_AA (03)
  RESERVED    = 0
  ENAA_P5     = 0
  ENAA_P4     = 0
  ENAA_P3     = 0
  ENAA_P2     = 0
  ENAA_P1     = 1
  ENAA_P0     = 1

 EN_RXADDR (03)
  RESERVED    = 0
  ERX_P5      = 0
  ERX_P4      = 0
  ERX_P3      = 0
  ERX_P2      = 0
  ERX_P1      = 1
  ERX_P0      = 1

 SETUP_AW (03)
  RESERVED    = 0
  AW          = 3

 SETUP_RETR (FF)
  ARD         = 15
  ARC         = 15

 RF_CH (2D)
  RESERVED    = 0
  RF_CH       = 45

 RF_SETUP (04)
  CONT_WAVE   = 0
  RESERVED    = 0
  RF_DR_LOW   = 0
  PLL_LOCK    = 0
  RF_DR_HIGH  = 0
  RF_PWR      = 2
  OBSOLETE    = 0

 STATUS (0E)
  RESERVED    = 0
  RX_DR       = 0
  TX_DS       = 0
  MAX_RT      = 0
  RX_P_NO     = 7
  TX_FULL     = 0

 OBSERVE_TX (00)
  PLOS_CNT    = 0
  ARC_CNT     = 0

 RPD (00)
  RESERVED    = 0
  RPD         = 0

 RX_ADDR_P0   =  19  51  38  31  00
 RX_ADDR_P1   =  19  51  38  31  00
 RX_ADDR_P2   = [19  51  38  31] C3
 RX_ADDR_P3   = [19  51  38  31] C4
 RX_ADDR_P4   = [19  51  38  31] C5
 RX_ADDR_P5   = [19  51  38  31] C6

 TX_ADDR      = E7 E7 E7 E7 E7

 RX_PW_0 (00)
  RESERVED    = 0
  RX_PW_P0    = 0

 RX_PW_1 (08)
  RESERVED    = 0
  RX_PW_P1    = 8

 RX_PW_2 (00)
  RESERVED    = 0
  RX_PW_P2    = 0

 RX_PW_3 (00)
  RESERVED    = 0
  RX_PW_P3    = 0

 RX_PW_4 (00)
  RESERVED    = 0
  RX_PW_P4    = 0

 RX_PW_5 (00)
  RESERVED    = 0
  RX_PW_P5    = 0

 FIFO_STATUS (11)
  RESERVED    = 0
  TX_REUSE    = 0
  TX_FULL     = 0
  TX_EMPTY    = 1
  RESERVED    = 0
  RX_FULL     = 0
  RX_EMPTY    = 1

 DYNPD (00)
  RESERVED    = 0
  DPL_P5      = 0
  DPL_P4      = 0
  DPL_P3      = 0
  DPL_P2      = 0
  DPL_P1      = 0
  DPL_P0      = 0

 FEATURE (00)
  RESERVED    = 0
  EN_DPL      = 0
  EN_ACK_PAY  = 0
  EN_DYN_ACK  = 0
```

The-Futurist, 2025-05-05:
I'm going to zeroize the unused RX/TX registers to make this easier on the eye.

The-Futurist, 2025-05-05:
OK this is the same dump and test, but unused RX address registers are zeroized to avoid the visual confusion with the unused registers having odd non zero defaults.

Sender
```
NRF24L01+ REGISTER DUMP
 CONFIG (0E)
  RESERVED    = 0
  MASK_RX_DR  = 0
  MASK_TX_DS  = 0
  MASK_MAX_RT = 0
  EN_CRC      = 1
  CRCO        = 1
  PWR_UP      = 1
  PRIM_RX     = 0

 EN_AA (03)
  RESERVED    = 0
  ENAA_P5     = 0
  ENAA_P4     = 0
  ENAA_P3     = 0
  ENAA_P2     = 0
  ENAA_P1     = 1
  ENAA_P0     = 1

 EN_RXADDR (03)
  RESERVED    = 0
  ERX_P5      = 0
  ERX_P4      = 0
  ERX_P3      = 0
  ERX_P2      = 0
  ERX_P1      = 1
  ERX_P0      = 1

 SETUP_AW (03)
  RESERVED    = 0
  AW          = 3

 SETUP_RETR (FF)
  ARD         = 15
  ARC         = 15

 RF_CH (2D)
  RESERVED    = 0
  RF_CH       = 45

 RF_SETUP (04)
  CONT_WAVE   = 0
  RESERVED    = 0
  RF_DR_LOW   = 0
  PLL_LOCK    = 0
  RF_DR_HIGH  = 0
  RF_PWR      = 2
  OBSOLETE    = 0

 STATUS (0E)
  RESERVED    = 0
  RX_DR       = 0
  TX_DS       = 0
  MAX_RT      = 0
  RX_P_NO     = 7
  TX_FULL     = 0

 OBSERVE_TX (0F)
  PLOS_CNT    = 0
  ARC_CNT     = 15

 RPD (00)
  RESERVED    = 0
  RPD         = 0

 RX_ADDR_P0   =  19  51  38  31  00
 RX_ADDR_P1   =  00  00  00  00  00
 RX_ADDR_P2   = [00  00  00  00] 00
 RX_ADDR_P3   = [00  00  00  00] 00
 RX_ADDR_P4   = [00  00  00  00] 00
 RX_ADDR_P5   = [00  00  00  00] 00

 TX_ADDR      = 19 51 38 31 00

 RX_PW_0 (00)
  RESERVED    = 0
  RX_PW_P0    = 0

 RX_PW_1 (00)
  RESERVED    = 0
  RX_PW_P1    = 0

 RX_PW_2 (00)
  RESERVED    = 0
  RX_PW_P2    = 0

 RX_PW_3 (00)
  RESERVED    = 0
  RX_PW_P3    = 0

 RX_PW_4 (00)
  RESERVED    = 0
  RX_PW_P4    = 0

 RX_PW_5 (00)
  RESERVED    = 0
  RX_PW_P5    = 0

 FIFO_STATUS (11)
  RESERVED    = 0
  TX_REUSE    = 0
  TX_FULL     = 0
  TX_EMPTY    = 1
  RESERVED    = 0
  RX_FULL     = 0
  RX_EMPTY    = 1

 DYNPD (00)
  RESERVED    = 0
  DPL_P5      = 0
  DPL_P4      = 0
  DPL_P3      = 0
  DPL_P2      = 0
  DPL_P1      = 0
  DPL_P0      = 0

 FEATURE (00)
  RESERVED    = 0
  EN_DPL      = 0
  EN_ACK_PAY  = 0
  EN_DYN_ACK  = 0
```

Receiver
```
NRF24L01+ REGISTER DUMP
 CONFIG (3F)
  RESERVED    = 0
  MASK_RX_DR  = 0
  MASK_TX_DS  = 1
  MASK_MAX_RT = 1
  EN_CRC      = 1
  CRCO        = 1
  PWR_UP      = 1
  PRIM_RX     = 1

 EN_AA (03)
  RESERVED    = 0
  ENAA_P5     = 0
  ENAA_P4     = 0
  ENAA_P3     = 0
  ENAA_P2     = 0
  ENAA_P1     = 1
  ENAA_P0     = 1

 EN_RXADDR (03)
  RESERVED    = 0
  ERX_P5      = 0
  ERX_P4      = 0
  ERX_P3      = 0
  ERX_P2      = 0
  ERX_P1      = 1
  ERX_P0      = 1

 SETUP_AW (03)
  RESERVED    = 0
  AW          = 3

 SETUP_RETR (FF)
  ARD         = 15
  ARC         = 15

 RF_CH (2D)
  RESERVED    = 0
  RF_CH       = 45

 RF_SETUP (04)
  CONT_WAVE   = 0
  RESERVED    = 0
  RF_DR_LOW   = 0
  PLL_LOCK    = 0
  RF_DR_HIGH  = 0
  RF_PWR      = 2
  OBSOLETE    = 0

 STATUS (0E)
  RESERVED    = 0
  RX_DR       = 0
  TX_DS       = 0
  MAX_RT      = 0
  RX_P_NO     = 7
  TX_FULL     = 0

 OBSERVE_TX (00)
  PLOS_CNT    = 0
  ARC_CNT     = 0

 RPD (00)
  RESERVED    = 0
  RPD         = 0

 RX_ADDR_P0   =  19  51  38  31  00
 RX_ADDR_P1   =  19  51  38  31  00
 RX_ADDR_P2   = [19  51  38  31] 00
 RX_ADDR_P3   = [19  51  38  31] 00
 RX_ADDR_P4   = [19  51  38  31] 00
 RX_ADDR_P5   = [19  51  38  31] 00

 TX_ADDR      = E7 E7 E7 E7 E7

 RX_PW_0 (00)
  RESERVED    = 0
  RX_PW_P0    = 0

 RX_PW_1 (08)
  RESERVED    = 0
  RX_PW_P1    = 8

 RX_PW_2 (00)
  RESERVED    = 0
  RX_PW_P2    = 0

 RX_PW_3 (00)
  RESERVED    = 0
  RX_PW_P3    = 0

 RX_PW_4 (00)
  RESERVED    = 0
  RX_PW_P4    = 0

 RX_PW_5 (00)
  RESERVED    = 0
  RX_PW_P5    = 0

 FIFO_STATUS (11)
  RESERVED    = 0
  TX_REUSE    = 0
  TX_FULL     = 0
  TX_EMPTY    = 1
  RESERVED    = 0
  RX_FULL     = 0
  RX_EMPTY    = 1

 DYNPD (00)
  RESERVED    = 0
  DPL_P5      = 0
  DPL_P4      = 0
  DPL_P3      = 0
  DPL_P2      = 0
  DPL_P1      = 0
  DPL_P0      = 0

 FEATURE (00)
  RESERVED    = 0
  EN_DPL      = 0
  EN_ACK_PAY  = 0
  EN_DYN_ACK  = 0
```

TMRh20, 2025-05-05:
`RX_PW_P0 and RX_PW_P1` also need to be configured I believe. That's about the only other difference I can spot right now in register values.

What I would do is test the radios with known working devices with the RF24 gettingstarted example just to be sure they work.

2bndy5, 2025-05-05:
@The-Futurist Have you read the datasheet's Appendix A? It has a very good example of how to setup the radio and use auto-ack.

TMrH20 has already suggested everything I noticed that was wrong. 

Coincidentally, we just had some long discussions about persisting the TX and RX address correctly on pipe 0 when switching between modes. This led us to overload the `stopListening()` function to allow for setting the TX address, while deprecating the `openWritingPipe()` function all together.

2bndy5, 2025-05-05:
> ```
> TX_ADDR      = 19 51 38 31 00
> ```

I'm not sure if the Endianness is accounted for here. In our `printDetails()` output, we reverse the addresses so that the MSB shows last.

I say this because, if your posted output is also little endian, then the MSB is `0`. The datasheet clearly states that this is a bad idea due to the auto-generated preamble. If you're so keen on blog posts, then have a look at [ManiacBug's post about the MSB of addresses](http://maniacalbits.blogspot.com/2013/04/rf24-addressing-nrf24l01-radios-require.html) (written by the original author of this library).

> I'm trying to establish some settings that are known to correctly work with auto-ack. Is there any possibility someone could perhaps do a test that proves auto ack works with a pair of devices (just sending a simple single payload) and then share the sender and receive register settings here?

You can do this yourself with our [GettingStarted example](https://github.com/nRF24/RF24/blob/master/examples/GettingStarted/GettingStarted.ino). You would just need to

1. un-comment one of the `print*Details()` functions in `setup()`: https://github.com/nRF24/RF24/blob/beb8ffebc5be0f5470fb08c851f941d0c70a45e0/examples/GettingStarted/GettingStarted.ino#L88-L91
2. change the `CE_PIN` and `CSN_PIN` definitions for your board: https://github.com/nRF24/RF24/blob/beb8ffebc5be0f5470fb08c851f941d0c70a45e0/examples/GettingStarted/GettingStarted.ino#L17-L20

---
FYI, our GitHub issues are actually for developing _this_ library. They are not here to aid others in writing an entirely different driver library. I'd say we're being really generous so far.

2bndy5, 2025-05-05:
`0xAA` (`0b10101010`) and `0x55` (`0b01010101`) are the _worst_ choices for an address' MSB. Did you actually read the [blog post I linked](http://maniacalbits.blogspot.com/2013/04/rf24-addressing-nrf24l01-radios-require.html)? It builds on what is stated in the datasheet.

Just to be clear, the `MAX_RT` flag must be cleared before sending anything. Otherwise, the TX FIFO is locked when `MAX_RT` is asserted.

FWIW, I had to read the datasheet countless[^1] times when writing a driver lib from scratch (in pure python for CircuitPython ecosystem). Years later, to this day, I still have to re-read some parts to verify my brain's memory isn't faltering.

[^1]: "countless" is not an exaggeration here. I literally lost count.

2bndy5, 2025-05-05:
> some settings are superfluous like RX_PW_P1 = 8 in the sender, its unnecessary but harmless I suspect, a leftover from prior experimentation

Its not superfluous. The static payload size must match on both nodes (and be $`0 < x <= 32`$). In this lib, we pad the payload with zeroes, if the given buffer is shorter than the static payload size.

For dynamic payload lengths, the `RX_PW_*` registers do nothing.

The-Futurist, 2025-05-05:
> Its not superfluous. The static payload size must match on both nodes (and be 0 < x <= 32 ). In this lib, we pad the payload with zeroes, if the given buffer is shorter than the static payload size.

What I meant was it's superfluous for the test I did which is only using address 0, having it set for Pipe 1 should not have any impact on the outcome of my test which only xfers data over the Pipe 0 address. Ideally unused settings should be zeroized for tests like this so I should tidy that. 

> 
> For dynamic payload lengths, the `RX_PW_*` registers do nothing.

Yes, that's my understanding.

2bndy5, 2025-05-05:
> But as I said, if I were to setup NRF registers EXACTLY the same as someone who had a successful auto ack test, then I should get the same result unless something like hardware is at fault.

You are basically asking for our library defaults. This can observed from source code (see our [`_init_radio()`](https://github.com/nRF24/RF24/blob/beb8ffebc5be0f5470fb08c851f941d0c70a45e0/RF24.cpp#L1060)) or by running our examples. We are not here to develop your library. So, I will not be going out of my way to post the `printDetails()` output for each node.

Given the registers' values you posted, it looks like it should work (disregarding the `0xAA` MSB in the address). This is why I was hinting at your (unseen) code.

The-Futurist, 2025-05-05:
> You are basically asking for our library defaults. This can observed from source code (see our [...]

Thank you.

I too am suspecting a defective device (well all three, because I have swapped them around and get the same behavior). I read that some "clones" (e.g. the [Si24R1](https://www.lcsc.com/datasheet/lcsc_datasheet_2412091511_Nanjing-Zhongke-Microelectronics-Si24R1_C14436.pdf)) were suspect but found no solid conformation just anecdotal reports.

There's always a risk of misinterpretation when trying to fathom the state of a system by looking at source code, this is why a dump of device registers from a known good test case is so helpful, it goes tight to the heart of the matter - if my devices configured EXACTLY as those in a working test, then we can forget about code (to a large extent) and look elsewhere for the cause of the failure.

2bndy5, 2025-05-05:
Not _exactly_. The use of the radio's CE pin is very important, specifically the duration of the pulses. It is something that is often overlooked. In this lib, we just keep the CE pin asserted in TX mode after the first transmission; that way subsequent uploaded payloads can be transmitted immediately, as long as the `MAX_RT` flag is cleared.

---
I never bought an Si24R1. But we have had users report the `NO_ACK` flag (in `FEATURE` register) was implemented in reverse, which seems to have stemmed from a typo in the first (v1) datasheet for the nRF24L01 (non-plus variant). Consequently, auto-ack will not work correctly on Si24R1 chips. And there is no way (via software alone) to detect if it is the Si24R1 Chinese clone or a genuine nRF24L01.

The-Futurist, 2025-05-05:
This is interesting too. [Nordic NRF24L01+ – Real Vs Fake](https://hackaday.com/2015/02/23/nordic-nrf24l01-real-vs-fake/#comment-2474764)

2bndy5, 2025-05-05:
Yeah. That's the first result google gives you.

TMRh20, 2025-05-05:
I have a bunch of SI24R1s and auto-ack works fine, its the selective ack feature I'm sure is broken. I attempted to use it in RF24Network, but that didn't pan out because of SI24R1s

The-Futurist, 2025-05-05:
> I have a bunch of SI24R1s and auto-ack works fine, its the selective ack feature I'm sure is broken. I attempted to use it in RF24Network, but that didn't pan out because of SI24R1s

If you get an opportunity please run a test o that and dump the device setting right before you send data, I'd like to compare your settings with mine, if I were to copy them exactly mine should also work.

TMRh20, 2025-05-05:
> If you get an opportunity and have some time, please run a test on that and dump the device setting right before you send data, I'd like to compare your settings with mine, if I were to copy them exactly mine should also work.

That's pretty much what I've done here is compared registers. The only one that's still different is the RX_PW_0 and RX_PW_1 registers. We have them initialized at 32 bytes at startup and 4 bytes in the gettingstarted example. You have one set to 0 another set to 8... Set them all the same. 0 is inactive...

TMRh20, 2025-05-05:
In your last receiver dump also 
```
  EN_AA (01)
  RESERVED    = 0
  ENAA_P5     = 0
  ENAA_P4     = 0
  ENAA_P3     = 0
  ENAA_P2     = 0
  ENAA_P1     = 0
  ENAA_P0     = 1
```
ENAA_P1  needs to be 1 again...

Transmitter:
```
STATUS		= 0x0e RX_DR=0 TX_DS=0 TX_DF=0 RX_PIPE=7 TX_FULL=0
RX_ADDR_P0-1	= 0x65646f4e31 0x65646f4e32
RX_ADDR_P2-5	= 0x33 0xce 0x3e 0xe3
TX_ADDR		= 0x65646f4e31
RX_PW_P0-6	= 0x04 0x04 0x04 0x04 0x04 0x04
EN_AA		= 0x3f
EN_RXADDR	= 0x03
RF_CH		= 0x4c
RF_SETUP	= 0x03
CONFIG		= 0x0e
DYNPD/FEATURE	= 0x00 0x00
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_LOW
ARC		= 15
```
Receiver:
```
SPI Speedz	= 10 Mhz
STATUS		= 0x0e RX_DR=0 TX_DS=0 TX_DF=0 RX_PIPE=7 TX_FULL=0
RX_ADDR_P0-1	= 0x65646f4e31 0x65646f4e32
RX_ADDR_P2-5	= 0x33 0xce 0x3e 0xe3
TX_ADDR		= 0x65646f4e31
RX_PW_P0-6	= 0x04 0x04 0x04 0x04 0x04 0x04
EN_AA		= 0x3f
EN_RXADDR	= 0x02
RF_CH		= 0x4c
RF_SETUP	= 0x03
CONFIG		= 0x0f
DYNPD/FEATURE	= 0x00 0x00
Data Rate	= 1 MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_LOW
ARC		= 15
```

This is all you should need to get up and running.

The-Futurist, 2025-05-05:
Much appreciate @TMRh20 .

I'm assuming that for repeated one way transmissions (sender sends over and over to a single address and receiver receives over and over - in my test) then SENDER.RX_PW_Px can all be zero because the sender is never receiving any payloads.

I'm also using a single address set as PIPE 0 on both sender and receiver so I make no use of PIPE1.

I will take your setup though and work with that, thanks a lot for posting that.

The-Futurist, 2025-05-05:
Wow, guys.

I ordered some alternative devices the other and they JUST got delivered a day early:

[E01-ML01DP5](https://www.amazon.com/dp/B0CDH1JJZG?ref=ppx_yo2ov_dt_b_fed_asin_title)

So I just swapped them with the older ones (pinouts are identical for the SPI etc) and hey presto - success !!!

The sender no longer gets MAX_RT it gets TX_DS and if I power down the receiver module it does get MAX_RT as expected.

Insane, but thank you guys for your help and I can personally recommend that module - but have yet to test its range...

2bndy5, 2025-05-05:
That ebyte module is a pretty popular and rather proven PA/LNA. We now keep a copy of [it's specifications (in English)](https://github.com/nRF24/RF24/blob/master/datasheets/E01-ML01DP5_Usermanual_EN_V1.3.pdf) in this repo.

Closing this as there is nothing to do with the RF24 library and your initial query seems to be resolved.

closed 2025-05-05 by 2bndy5