nRF24/RF24 #1042: [Question]: Must ackpayload be the same length as the sent payload?

xhy2008, 2025-08-07:
### What would you like to know?

The official ackpayload demo worked.
When I changed the ackpayload length(e.g. 8bytes sent and 16bytes ackpayload) ,the radios cannot communicate anymore.

2bndy5, 2025-08-07:
No ACK payload size is not dependent on incoming payload size.

ACK payloads are dynamically sized. This implies that all payloads are dynamically sized when using ACK payloads. 

> When I changed the ackpayload length(e.g. 8bytes sent and 16bytes ackpayload) ,the radios cannot communicate anymore.

I suspect you changes more than just the payload size. We cannot help you diagnose a problem if you only tell us what _you think_ caused it.

Also have a look at our docs:
- [`enableAckPayload()`](https://nrf24.github.io/RF24/classRF24.html#abf8efced2ee9edbcc6510878b20edc1b)
- [`writeAckPayload()`](https://nrf24.github.io/RF24/classRF24.html#a36b9fc363d3560358fe430a600a6f385)
- [`setRetries()`](https://nrf24.github.io/RF24/classRF24.html#a4c6d3959c8320e64568395f4ef507aef) This is important because a short delay between TX attempts will ultimately break use of ACK payloads.

2bndy5, 2025-08-07:
I'm not sure how we can help without seeing the code you are using (and it's output if any).

> Besides,only RF24_PA_MIN could work for me,and an extra power supply did't solve the problem.

This is confusing. The examples use `RF24_PA_MIN` as a hack to save on required TX power. If "an extra power supply did't solve the problem", then either <sup>A</sup>the power supply's current wasn't consistent/high enough or <sup>B</sup>you somehow wired the circuit wrong.

Remember, power is defined as $`P=I*V`$. It is not enough to only ensure that $`V`$ is around 3.3 volts. PA/LNA modules can require $`I`$ to be at least 150 mA to transmit (maybe more depending on the radio's model/manufacturer).

TMRh20, 2025-08-07:
I suspect this issue may just be related to:

>d. If ACK packet payload is activated, ACK packets have dynamic payload lengths and the Dynamic Pay-
load Length feature should be enabled for pipe 0 on the PTX and PRX. This is to ensure that they receive
the ACK packets with payloads. If the payload in ACK is more than 15 byte in 2Mbps mode the ARD must
be 500μS or more, and if the payload is more than 5byte in 1Mbps mode the ARD must be 500μS or more.

@xhy2008 You may just need to increase values in `radio.setRetries(15,15);`

xhy2008, 2025-08-07:
@2bndy5 
Is it the same incident (is @xhy2008 also noname-114514 on Arduino forums)?
Yes,that's me. You are so careful.😉
I would try reinstalling an old version.

@TMRh20 
I don't think it's because the retry time limit,as my rx node received nothing.But I would try that.

xhy2008, 2025-08-07:
@TMRh20 
I tried to increase retry delay,'half' of it worked.
I need to call write() lots of times to receive an ackpayload.
Now I would try an old version.

xhy2008, 2025-08-07:
I'm using esp32-s3 with a RF module produced by Ebyte.
Here is my code:
TX:
`#include "RF24.h"
#define RF_CE 48
#define RF_CSN 47
#define SCK 21
#define RF_MOSI 37
#define RF_MISO 36
const uint8_t WRITEADDRESS[6]={"12345"};
const uint8_t READADDRESS[6]={"45678"};
struct controldata{
    uint8_t throttle;//1byte 0-255
    int8_t roll;//1byte -127-128
    int8_t pitch;//1byte -127-128 
    uint8_t command;
};//total 4 byte
struct statdata{
    float roll;
    float pitch;
    uint8_t heading;
    float V;
    //above costs 16byte
};
class RF{
    private:
    SPIClass spi=SPIClass(FSPI);
    public:
    RF24 radio;
    bool init(){
        spi.begin(SCK,RF_MISO,RF_MOSI,RF_CSN);
        if (!radio.begin(&spi,RF_CE,RF_CSN)) {
            Serial.println("rf init failed");
            return false;
        }
        radio.setDataRate(RF24_250KBPS);
        radio.setAddressWidth(5);
        radio.setPALevel(RF24_PA_MIN);
        radio.setRetries(15,15);
        radio.setCRCLength(RF24_CRC_16);
        radio.enableDynamicPayloads();
        radio.enableAckPayload();
        radio.openWritingPipe(WRITEADDRESS);
        radio.openReadingPipe(1, READADDRESS);
        radio.stopListening();
        return true;
    }
    statdata sendData(controldata data){
        while(!radio.write(&data,sizeof(controldata)))
        {
            Serial.println("send data failed");
            delay(5);
        }
        statdata rec;
        if(radio.available())
          radio.read(&rec,sizeof(statdata));
        else
            Serial.println("No ack payload received");
        return rec;
    }
};
`

RX:
`
#include "RF24.h"
#define RF_CE 48
#define RF_CSN 47
#define SCK 21
#define RF_MOSI 20
#define RF_MISO 19
const uint8_t READADDRESS[6]={"12345"};
const uint8_t WRITEADDRESS[6]={"45678"};
struct controldata{
    uint8_t throttle;//1byte 0-255
    int8_t roll;//1byte -127-128
    int8_t pitch;//1byte -127-128    
    uint8_t command;
};//4 byte
struct statdata{
    float roll;
    float pitch;
    uint8_t heading;
    float V;
    //above costs 16byte
};
class RF{
    private:
    SPIClass spi=SPIClass(HSPI);
    public:
    RF24 radio;
    bool init(){
        spi.begin(SCK,RF_MISO,RF_MOSI,RF_CSN);
        if (!radio.begin(&spi,RF_CE,RF_CSN)) {
            Serial.println("rf init failed");
            return false;
        }
        radio.setDataRate(RF24_250KBPS);
        radio.setPALevel(RF24_PA_MIN);
        radio.setAddressWidth(5);
        radio.setRetries(15,15);
        radio.setCRCLength(RF24_CRC_16);
        radio.enableDynamicPayloads();
        radio.enableAckPayload();
        radio.openReadingPipe(1,READADDRESS);
        radio.openWritingPipe(WRITEADDRESS);
        radio.startListening();
        return true;
    }
    controldata recv(statdata data){
        radio.writeAckPayload(1, &data, sizeof(statdata));
        //Serial.print("Write ack payload ,size:");
        //Serial.println(sizeof(statdata));
        while(!radio.available())
        {
            delayMicroseconds(128);
        }
        controldata ret;
        radio.read(&ret,sizeof(controldata));
        return ret;
    }
};
`

closed 2026-10-23 by TMRh20